openwrt/openwrt #12278: mwan3: First four pings dropped/lost on mvebu/cortexa9 21.02.05

sscheib, 2023-03-28:
### Describe the bug

When using mwan3 some pings are "lost/dropped/never seen" in tcpdump on wanB (which is a repurposed switch port, lan4). This *only* happens when mwan3 is running. With mwan3 stopped, the behavior is as expected.
The pings are not really the issue (for me at least), but as soon as wanA goes down, wanB stops working entirely and no traffic flows through. This issue prevents a failover configuration to work properly.

This seems to be an issue on more than one device, so I would guess my device is fully functioning.
The **exact same** configuration used to work on 19.02 on the **exact same device**. The installation and configuration of my devices is automated, which leads me to believe that devices have been setup almost the same way.

There is one big difference, however. With 21.02 mvebu/cortexa9 switched to DSA, hence repurposing a switch port to a LAN port is different than with swconfig.
I have followed the tutorial/howto on the OpenWrt wiki included in the [mwan3 article](https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3#routers_using_distributed_switch_architecture_dsa).

For reference, please see the following forum threads:
- [Mwan3 - wanb stops working as soon as primary wan interfaces goes down](https://forum.openwrt.org/t/mwan3-wanb-stops-working-as-soon-as-primary-wan-interfaces-goes-down/152730)

- [Mwan3 first 4 pings drop on wlanb](https://forum.openwrt.org/t/mwan3-first-4-pings-drop-on-wlanb/154536)

I have opened this issue to draw the attention of the developers to the issue, as I don't know how to troubleshoot this any further.
I can test whatever is needed, as long as I get a helping hand in troubleshooting.

### OpenWrt version

r16688-fa9a932fdb

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Linksys WRT3200ACM

### Image kind

Official downloaded image

### Steps to reproduce

#### Preparation

1. Install mwan3: `opkg update && opkg install mwan3`
2. Closely follow the tutorial/howto to setup an additional WAN interface in the [mwan3 article](https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3#routers_using_distributed_switch_architecture_dsa)
3. Capture traffic in tcpdump in a seperate shell window/tmux/screen
```
tcpdump -enni any dst host heise.de -vvv
```
5. Ensure pings work on wanA (aka pppoe-wan in this case)
```
root@openwrt.dmz.int.example.com:~# ping -I pppoe-wan heise.de -4 -c 5
PING heise.de (193.99.144.80): 56 data bytes
64 bytes from 193.99.144.80: seq=0 ttl=248 time=15.008 ms
64 bytes from 193.99.144.80: seq=1 ttl=248 time=14.940 ms
64 bytes from 193.99.144.80: seq=2 ttl=248 time=15.073 ms
64 bytes from 193.99.144.80: seq=3 ttl=248 time=14.691 ms
64 bytes from 193.99.144.80: seq=4 ttl=248 time=15.061 ms

--- heise.de ping statistics ---
5 packets transmitted, 5 packets received, 0% packet loss
round-trip min/avg/max = 14.691/14.954/15.073 ms
root@openwrt.dmz.int.example.com:~# 

# tcpdump of the pings in seperate SSH session
root@openwrt.dmz.int.example.com:~# tcpdump -enni any dst host heise.de -vvv
tcpdump: listening on any, link-type LINUX_SLL (Linux cooked v1), capture size 262144 bytes
18:07:19.640195 Out ethertype IPv4 (0x0800), length 100: (tos 0x0, ttl 64, id 51366, offset 0, flags [DF], proto ICMP (1), length 84)
    10.2.174.182 > 193.99.144.80: ICMP echo request, id 9550, seq 0, length 64
18:07:20.640436 Out ethertype IPv4 (0x0800), length 100: (tos 0x0, ttl 64, id 51443, offset 0, flags [DF], proto ICMP (1), length 84)
    10.2.174.182 > 193.99.144.80: ICMP echo request, id 9550, seq 1, length 64
18:07:21.640573 Out ethertype IPv4 (0x0800), length 100: (tos 0x0, ttl 64, id 51518, offset 0, flags [DF], proto ICMP (1), length 84)
    10.2.174.182 > 193.99.144.80: ICMP echo request, id 9550, seq 2, length 64
18:07:22.640707 Out ethertype IPv4 (0x0800), length 100: (tos 0x0, ttl 64, id 51576, offset 0, flags [DF], proto ICMP (1), length 84)
[... 123 lines cut ...]
64 bytes from 193.99.144.80: seq=2 ttl=245 time=11.961 ms
64 bytes from 193.99.144.80: seq=3 ttl=245 time=13.294 ms
64 bytes from 193.99.144.80: seq=4 ttl=245 time=10.490 ms

--- heise.de ping statistics ---
5 packets transmitted, 5 packets received, 0% packet loss
round-trip min/avg/max = 10.490/13.371/17.473 ms
root@openwrt.dmz.int.example.com:~# mwan3 use wan ping heise.de -4 -c 5
Running 'ping heise.de -4 -c 5' with DEVICE=pppoe-wan SRCIP=10.2.174.182 FWMARK=0x3f00 FAMILY=ipv4
PING heise.de (193.99.144.80): 56 data bytes
64 bytes from 193.99.144.80: seq=0 ttl=248 time=15.124 ms
64 bytes from 193.99.144.80: seq=1 ttl=248 time=14.895 ms
64 bytes from 193.99.144.80: seq=2 ttl=248 time=15.044 ms
64 bytes from 193.99.144.80: seq=3 ttl=248 time=14.717 ms
64 bytes from 193.99.144.80: seq=4 ttl=248 time=14.880 ms

--- heise.de ping statistics ---
5 packets transmitted, 5 packets received, 0% packet loss
round-trip min/avg/max = 14.717/14.932/15.124 ms
root@openwrt.dmz.int.example.com:~# 
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-03-28:
There is no ARP on ppp. Zero means 'other end' i.e route was balanced there.

sscheib, 2023-03-28:
Okay, I see.
I tried to force `NOARP` on the lan4 device with `ip link set arp off lan4`, which then shows the same `NOARP` flag in `ip a s` as the pppoe-wan interface, but that just cut the whole interface from connecting.
As I said above, I am not really deep into networking, but from my point of view, this shouldn't happen. If I disable mwan3 I cannot see a single ARP request and the first ping already goes through.

sscheib, 2023-04-10:
Is there anyone that is willing to give me a helping hand on this?

farnwomt, 2023-04-10:
Definitely a problem and I completely agree this is the default behaviour.  I have had exactly the same problem for a long time and I am using the latest snapshot, so it definitely hasn't been fixed.

farnwomt, 2023-04-10:
Just digging around a bit if you were to do something like:

mwan3 use lan4 exec ping heise.de

Then that would work, assuming you called your interface lan4, but to get a list of interfaces just use:

mwan3 interfaces

So the issue is definitely somehow mwan3 related.

brada4, 2023-04-12:
Please repeat tcdumps s/dst/host/g so that we see if response comes later than 1s ping utility timeout.

sscheib, 2023-04-12:
@brada4
```
root@openwrt.example.com:~# ping -I lan4 heise.de -4 -c10
PING heise.de (193.99.144.80): 56 data bytes
64 bytes from 193.99.144.80: seq=4 ttl=245 time=14.123 ms
64 bytes from 193.99.144.80: seq=5 ttl=245 time=12.464 ms
64 bytes from 193.99.144.80: seq=6 ttl=245 time=13.419 ms
64 bytes from 193.99.144.80: seq=7 ttl=245 time=12.966 ms
64 bytes from 193.99.144.80: seq=8 ttl=245 time=11.692 ms
64 bytes from 193.99.144.80: seq=9 ttl=245 time=21.022 ms

--- heise.de ping statistics ---
10 packets transmitted, 6 packets received, 40% packet loss
round-trip min/avg/max = 11.692/14.281/21.022 ms
root@openwrt.example.com:~# 
```

```
root@openwrt.example.com:~# tcpdump -enni any host heise.de -vv
tcpdump: listening on any, link-type LINUX_SLL (Linux cooked v1), capture size 262144 bytes
18:46:11.478769 Out 30:23:03:e0:52:20 ethertype ARP (0x0806), length 44: Ethernet (len 6), IPv4 (len 4), Request who-has 193.99.144.80 tell 10.1.217.230, length 28
18:46:11.478789 Out 30:23:03:e0:52:20 ethertype ARP (0x0806), length 44: Ethernet (len 6), IPv4 (len 4), Request who-has 193.99.144.80 tell 10.1.217.230, length 28
18:46:12.553189 Out 30:23:03:e0:52:20 ethertype ARP (0x0806), length 44: Ethernet (len 6), IPv4 (len 4), Request who-has 193.99.144.80 tell 10.1.217.230, length 28
18:46:12.553199 Out 30:23:03:e0:52:20 ethertype ARP (0x0806), length 44: Ethernet (len 6), IPv4 (len 4), Request who-has 193.99.144.80 tell 10.1.217.230, length 28
18:46:13.593188 Out 30:23:03:e0:52:20 ethertype ARP (0x0806), length 44: Ethernet (len 6), IPv4 (len 4), Request who-has 193.99.144.80 tell 10.1.217.230, length 28
18:46:13.593197 Out 30:23:03:e0:52:20 ethertype ARP (0x0806), length 44: Ethernet (len 6), IPv4 (len 4), Request who-has 193.99.144.80 tell 10.1.217.230, length 28
18:46:15.479226 Out 30:23:03:e0:52:20 ethertype IPv4 (0x0800), length 100: (tos 0x0, ttl 64, id 26404, offset 0, flags [DF], proto ICMP (1), length 84)
    10.1.217.230 > 193.99.144.80: ICMP echo request, id 17600, seq 4, length 64
18:46:15.479240 Out 30:23:03:e0:52:20 ethertype IPv4 (0x0800), length 100: (tos 0x0, ttl 64, id 26404, offset 0, flags [DF], proto ICMP (1), length 84)
    10.1.217.230 > 193.99.144.80: ICMP echo request, id 17600, seq 4, length 64
18:46:15.493235  In dc:15:c8:30:8f:a8 ethertype IPv4 (0x0800), length 100: (tos 0x0, ttl 245, id 9744, offset 0, flags [none], proto ICMP (1), length 84)
    193.99.144.80 > 10.1.217.230: ICMP echo reply, id 17600, seq 4, length 64
18:46:15.493249 Out dc:15:c8:30:8f:a8 ethertype IPv4 (0x0800), length 100: (tos 0x0, ttl 245, id 9744, offset 0, flags [none], proto ICMP (1), length 84)
    193.99.144.80 > 10.1.217.230: ICMP echo reply, id 17600, seq 4, length 64
18:46:16.479298 Out 30:23:03:e0:52:20 ethertype IPv4 (0x0800), length 100: (tos 0x0, ttl 64, id 26411, offset 0, flags [DF], proto ICMP (1), length 84)
    10.1.217.230 > 193.99.144.80: ICMP echo request, id 17600, seq 5, length 64
18:46:16.479313 Out 30:23:03:e0:52:20 ethertype IPv4 (0x0800), length 100: (tos 0x0, ttl 64, id 26411, offset 0, flags [DF], proto ICMP (1), length 84)
    10.1.217.230 > 193.99.144.80: ICMP echo request, id 17600, seq 5, length 64
18:46:16.491648  In dc:15:c8:30:8f:a8 ethertype IPv4 (0x0800), length 100: (tos 0x0, ttl 245, id 28373, offset 0, flags [none], proto ICMP (1), length 84)
    193.99.144.80 > 10.1.217.230: ICMP echo reply, id 17600, seq 5, length 64
18:46:16.491663 Out dc:15:c8:30:8f:a8 ethertype IPv4 (0x0800), length 100: (tos 0x0, ttl 245, id 28373, offset 0, flags [none], proto ICMP (1), length 84)
    193.99.144.80 > 10.1.217.230: ICMP echo reply, id 17600, seq 5, length 64
18:46:17.479369 Out 30:23:03:e0:52:20 ethertype IPv4 (0x0800), length 100: (tos 0x0, ttl 64, id 26441, offset 0, flags [DF], proto ICMP (1), length 84)
    10.1.217.230 > 193.99.144.80: ICMP echo request, id 17600, seq 6, length 64
18:46:17.479381 Out 30:23:03:e0:52:20 ethertype IPv4 (0x0800), length 100: (tos 0x0, ttl 64, id 26441, offset 0, flags [DF], proto ICMP (1), length 84)
    10.1.217.230 > 193.99.144.80: ICMP echo request, id 17600, seq 6, length 64
18:46:17.492687  In dc:15:c8:30:8f:a8 ethertype IPv4 (0x0800), length 100: (tos 0x0, ttl 245, id 53089, offset 0, flags [none], proto ICMP (1), length 84)
    193.99.144.80 > 10.1.217.230: ICMP echo reply, id 17600, seq 6, length 64
18:46:17.492701 Out dc:15:c8:30:8f:a8 ethertype IPv4 (0x0800), length 100: (tos 0x0, ttl 245, id 53089, offset 0, flags [none], proto ICMP (1), length 84)
    193.99.144.80 > 10.1.217.230: ICMP echo reply, id 17600, seq 6, length 64
18:46:18.479430 Out 30:23:03:e0:52:20 ethertype IPv4 (0x0800), length 100: (tos 0x0, ttl 64, id 26502, offset 0, flags [DF], proto ICMP (1), length 84)
    10.1.217.230 > 193.99.144.80: ICMP echo request, id 17600, seq 7, length 64
18:46:18.479440 Out 30:23:03:e0:52:20 ethertype IPv4 (0x0800), length 100: (tos 0x0, ttl 64, id 26502, offset 0, flags [DF], proto ICMP (1), length 84)
    10.1.217.230 > 193.99.144.80: ICMP echo request, id 17600, seq 7, length 64
18:46:18.492302  In dc:15:c8:30:8f:a8 ethertype IPv4 (0x0800), length 100: (tos 0x0, ttl 245, id 19438, offset 0, flags [none], proto ICMP (1), length 84)
    193.99.144.80 > 10.1.217.230: ICMP echo reply, id 17600, seq 7, length 64
18:46:18.492317 Out dc:15:c8:30:8f:a8 ethertype IPv4 (0x0800), length 100: (tos 0x0, ttl 245, id 19438, offset 0, flags [none], proto ICMP (1), length 84)
[... cut ...]
18:46:20.500479  In dc:15:c8:30:8f:a8 ethertype IPv4 (0x0800), length 100: (tos 0x0, ttl 245, id 61969, offset 0, flags [none], proto ICMP (1), length 84)
    193.99.144.80 > 10.1.217.230: ICMP echo reply, id 17600, seq 9, length 64
18:46:20.500493 Out dc:15:c8:30:8f:a8 ethertype IPv4 (0x0800), length 100: (tos 0x0, ttl 245, id 61969, offset 0, flags [none], proto ICMP (1), length 84)
    193.99.144.80 > 10.1.217.230: ICMP echo reply, id 17600, seq 9, length 64
```

@farnwomt
I see a similarity here: Your affected interface is configured with a static address; Exactly the same my wanB. wanA is pppoe on my end (so the other way round)

brada4, 2023-04-12:
You got them in first dump? I'd suspect something wires up slowly in far end of pppoe. Im wondering if there is some way to add pinging whatever to pppoe bringup process.

brada4, 2023-04-12:
It is somewhat inconsistent, in original post the outgoing pings exit, now they dont.
My suspicion is that other end of your tunnel completely sets up in 3 seconds once it receives traffic from you. In this case identical loss should be right after interface is brought up and ping is first IP traffic there without mwan

sscheib, 2023-04-12:
I have to disagree. The initial post describes a proper setup of mwan with two interfaces, and thus showcasing that both interfaces work properly without mwan started.
Right after the *preparation* step , the *Actual reproducer* follows. which shows the exact same symptoms as described in my previous answer.
There is a difference, however.
As you asked to replaced `dst host` with `host` it will naturally show both the *outgoing* and *incoming/reply* ICMP packets. I left that out initially, to have a more dense output (as the bug description itself is pretty verbose).
Another difference is that in my last post I ping 10 times instead of 5 (as 10 times takes more time), to show case that the ping replies are not delayed in any sort.
Does that clear up things?

farnwomt, 2023-04-12:
I have two different router setups, one with a pppoe device and one without any pppoe devices, the only consistency I have found is that if I ping on the interface that is currently active it works immediately, whereas the secondary device that isn't active always starts working after the 4th packet.

root@KVM-Router:~# ping -I eth2 212.58.244.56
PING 212.58.244.56 (212.58.244.56): 56 data bytes
64 bytes from 212.58.244.56: seq=4 ttl=54 time=15.655 ms
64 bytes from 212.58.244.56: seq=5 ttl=54 time=16.155 ms
64 bytes from 212.58.244.56: seq=6 ttl=54 time=15.780 ms
^C
--- 212.58.244.56 ping statistics ---
7 packets transmitted, 3 packets received, 57% packet loss
round-trip min/avg/max = 15.655/15.863/16.155 ms

and the tcpdump shows:

root@KVM-Router:~# tcpdump -n -i eth2 'host 212.58.244.56'
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on eth2, link-type EN10MB (Ethernet), capture size 262144 bytes
21:24:38.198891 ARP, Request who-has 212.58.244.56 tell 192.168.254.44, length 28
21:24:39.221691 ARP, Request who-has 212.58.244.56 tell 192.168.254.44, length 28
21:24:40.245689 ARP, Request who-has 212.58.244.56 tell 192.168.254.44, length 28
21:24:42.199603 IP 192.168.254.44 > 212.58.244.56: ICMP echo request, id 7535, seq 4, length 64
21:24:42.214965 IP 212.58.244.56 > 192.168.254.44: ICMP echo reply, id 7535, seq 4, length 64
21:24:43.200045 IP 192.168.254.44 > 212.58.244.56: ICMP echo request, id 7535, seq 5, length 64
21:24:43.215976 IP 212.58.244.56 > 192.168.254.44: ICMP echo reply, id 7535, seq 5, length 64
21:24:44.200246 IP 192.168.254.44 > 212.58.244.56: ICMP echo request, id 7535, seq 6, length 64
21:24:44.215758 IP 212.58.244.56 > 192.168.254.44: ICMP echo reply, id 7535, seq 6, length 64
^C

It does appear to be time related though because if you do:

root@KVM-Router:~# ping -I eth2 -i 2 news.bbc.co.uk
PING news.bbc.co.uk (212.58.244.56): 56 data bytes
64 bytes from 212.58.244.56: seq=2 ttl=54 time=21.230 ms
64 bytes from 212.58.244.56: seq=3 ttl=54 time=16.473 ms
64 bytes from 212.58.244.56: seq=4 ttl=54 time=19.466 ms
^C
--- news.bbc.co.uk ping statistics ---
5 packets transmitted, 3 packets received, 40% packet loss
round-trip min/avg/max = 16.473/19.056/21.230 ms

It only takes 2 packets instead, which is also 4 seconds later ...

farnwomt, 2023-04-12:
Another interesting point, if you set a ping running in one window, wait a couple of seconds and then set another ping running to the same IP address using another window you won't lose the first few ping packets in the second window.

However if you set a ping running in one window and wait long enough for the first ping to start getting packets through, but leave it running, and then start your second ping, the second ping still loses the first 4 packets.

brada4, 2023-04-12:
pppoe should not use arp at all. Does it also use arp without mwan?

metaphys, 2023-04-13:
I made some further test in a virtualized environment:
- this time only static ip config
- the gateway is the default kvm gateway to which I added a second IP (192.168.122.2)

```config
config interface 'wan1'
	option device 'eth1'
	option proto 'static'
	option ipaddr '192.168.122.30'
	option netmask '255.255.255.0'
	option gateway '192.168.122.1'
	option dns '9.9.9.9'
	option metric '10'

config interface 'wan2'
	option device 'eth2'
	option proto 'static'
	option ipaddr '192.168.122.40'
	option netmask '255.255.255.0'
	option gateway '192.168.122.2'
	option dns '9.9.9.9'
	option metric '20'
```
```config
config globals 'globals'
	option mmx_mask '0x3F00'
	option local_source 'lan'
	option logging '1'

config interface 'wan1'
	option enabled '1'
	list track_ip '1.0.0.1'
	list track_ip '1.1.1.1'
	list track_ip '208.67.222.222'
	list track_ip '208.67.220.220'
	option family 'ipv4'
	option reliability '2'

config interface 'wan2'
	option enabled '1'
	list track_ip '1.0.0.1'
	list track_ip '1.1.1.1'
	list track_ip '208.67.222.222'
	list track_ip '208.67.220.220'
	option family 'ipv4'
	option reliability '1'
```
- same issue on wanB only
```sh
root@test_openwrt:/# ping -I eth1 1.1.1.1
PING 1.1.1.1 (1.1.1.1): 56 data bytes
64 bytes from 1.1.1.1: seq=0 ttl=56 time=27.006 ms
64 bytes from 1.1.1.1: seq=1 ttl=56 time=28.536 ms
64 bytes from 1.1.1.1: seq=2 ttl=56 time=25.329 ms
64 bytes from 1.1.1.1: seq=3 ttl=56 time=25.141 ms
64 bytes from 1.1.1.1: seq=4 ttl=56 time=25.846 ms
^C
--- 1.1.1.1 ping statistics ---
6 packets transmitted, 5 packets received, 16% packet loss
round-trip min/avg/max = 25.141/26.371/28.536 ms
root@test_openwrt:/# ping -I eth2 1.1.1.1
PING 1.1.1.1 (1.1.1.1): 56 data bytes
64 bytes from 1.1.1.1: seq=4 ttl=56 time=28.560 ms
64 bytes from 1.1.1.1: seq=5 ttl=56 time=25.348 ms
64 bytes from 1.1.1.1: seq=6 ttl=56 time=24.442 ms
^C
--- 1.1.1.1 ping statistics ---
9 packets transmitted, 5 packets received, 44% packet loss
round-trip min/avg/max = 23.026/25.172/28.560 ms
```
- if I stop mwan3 the issue is gone
```sh
root@test_openwrt:/# /etc/init.d/mwan3 stop
root@test_openwrt:/# ping -I eth2 1.1.1.1
PING 1.1.1.1 (1.1.1.1): 56 data bytes
64 bytes from 1.1.1.1: seq=0 ttl=56 time=27.527 ms
64 bytes from 1.1.1.1: seq=1 ttl=56 time=27.734 ms
64 bytes from 1.1.1.1: seq=2 ttl=56 time=35.066 ms
--- 1.1.1.1 ping statistics ---
3 packets transmitted, 3 packets received, 0% packet loss
round-trip min/avg/max = 27.527/30.109/35.066 ms
root@test_openwrt:/# ping -I eth1 1.1.1.1
PING 1.1.1.1 (1.1.1.1): 56 data bytes
64 bytes from 1.1.1.1: seq=0 ttl=56 time=26.515 ms
64 bytes from 1.1.1.1: seq=1 ttl=56 time=28.499 ms
64 bytes from 1.1.1.1: seq=2 ttl=56 time=23.308 ms
3 packets transmitted, 3 packets received, 0% packet loss
round-trip min/avg/max = 23.308/26.107/28.499 ms
```
- tcpdump
```sh
root@test_openwrt:/# /etc/init.d/mwan3 start
root@test_openwrt:~# tcpdump -n -i eth2 'host 212.58.244.56'
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on eth2, link-type EN10MB (Ethernet), capture size 262144 bytes
08:36:34.980319 ARP, Request who-has 212.58.244.56 tell 192.168.122.40, length 28
08:36:36.061749 ARP, Request who-has 212.58.244.56 tell 192.168.122.40, length 28
08:36:37.091743 ARP, Request who-has 212.58.244.56 tell 192.168.122.40, length 28
08:36:38.981351 IP 192.168.122.40 > 212.58.244.56: ICMP echo request, id 6041, seq 4, length 64
08:36:39.042377 IP 212.58.244.56 > 192.168.122.40: ICMP echo reply, id 6041, seq 4, length 64
08:36:39.981590 IP 192.168.122.40 > 212.58.244.56: ICMP echo request, id 6041, seq 5, length 64
08:36:40.040093 IP 212.58.244.56 > 192.168.122.40: ICMP echo reply, id 6041, seq 5, length 64
08:36:40.981727 IP 192.168.122.40 > 212.58.244.56: ICMP echo request, id 6041, seq 6, length 64
08:36:41.040228 IP 212.58.244.56 > 192.168.122.40: ICMP echo reply, id 6041, seq 6, length 64
08:36:41.981909 IP 192.168.122.40 > 212.58.244.56: ICMP echo request, id 6041, seq 7, length 64
08:36:42.037469 IP 212.58.244.56 > 192.168.122.40: ICMP echo reply, id 6041, seq 7, length 64
08:36:42.982050 IP 192.168.122.40 > 212.58.244.56: ICMP echo request, id 6041, seq 8, length 64
08:36:43.037572 IP 212.58.244.56 > 192.168.122.40: ICMP echo reply, id 6041, seq 8, length 64
08:36:43.982434 IP 192.168.122.40 > 212.58.244.56: ICMP echo request, id 6041, seq 9, length 64
08:36:44.038392 IP 212.58.244.56 > 192.168.122.40: ICMP echo reply, id 6041, seq 9, length 64
08:36:44.982673 IP 192.168.122.40 > 212.58.244.56: ICMP echo request, id 6041, seq 10, length 64
08:36:45.041882 IP 212.58.244.56 > 192.168.122.40: ICMP echo reply, id 6041, seq 10, length 64
08:36:45.982935 IP 192.168.122.40 > 212.58.244.56: ICMP echo request, id 6041, seq 11, length 64
08:36:46.037068 IP 212.58.244.56 > 192.168.122.40: ICMP echo reply, id 6041, seq 11, length 64
```
- same thing fishy with arp
```sh
root@test_openwrt:~# arp
IP address       HW type     Flags       HW address            Mask     Device
212.58.244.56    0x1         0x0         00:00:00:00:00:00     *        eth2
192.168.122.1    0x1         0x2         52:54:00:80:23:cb     *        eth2
192.168.122.2    0x1         0x2         52:54:00:80:23:cb     *        eth2
192.168.122.1    0x1         0x2         52:54:00:80:23:cb     *        eth1
1.1.1.1          0x1         0x0         00:00:00:00:00:00     *        eth2
```
So this seems to confirm:
- on second interface only
- indifferent of the type of link
- mwan3 related

brada4, 2023-04-13:
The deficiency I see that point  to point link gets ARP requests sent. There is certainly nobody to respond at other end.

sscheib, 2023-04-13:
@metaphys the option `local_source` should only be present up until 18.06 (according to https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3) and should only be used for a routeable `lo`. In my case (21.02), it is not present and setting it to `lan` does not make any difference.

metaphys, 2023-04-13:
@sscheib ah ok, thank you! I wasn’t sure about this
 but my test about this preventing the issue turns out to be wrong: because of the interval value if you don’t wait 10 sec before testing, it is not valid. So I deleted the option, restarted the router. The issue is still present :/

sscheib, 2023-04-13:
@metaphys Have you actually tried a failover?
I make the call that the failover will work if the interface with the higher metric will fail, but not the other way around.
Say you have the following situation:
- wanA: metric 10
- wanB: metric 20

I assume the following will happen:
If wanB goes down, failover will work.
If wanA goes down, wanB will stop to work until wanA is back up.

At least that's the situation for me. I can almost certainly rule out a misconfiguration of mwan3, as I have automated the setup of my routers, and the exact same configuration worked on the exact same device prior to 21.02.

The only symptom I could figure out where the missing pings, as well as the zero-MAC ARP table.

metaphys, 2023-04-14:
I have a VM OpenWrt 22.03.3, r20028-43d71ad93e in production dealing with failover. We have 300 employees behind that box and a legal obligation to be reachable by phone (Voip) 24/7. So yes, we did test it thoroughly in both directions.
That being said in ours case the failover is made in regards to traffic going through the box as to oppose to traffic originating from it.

> At least that's the situation for me. I can almost certainly rule out a misconfiguration of mwan3, as I have automated the setup of my routers, and the exact same configuration worked on the exact same device prior to 21.02.
I wouldn't be so sure of that one, the configuration can evolve quickly with openwrt.

sscheib, 2023-04-14:
@metaphys 
Sorry, I didn't mean to offend you. More often than not, people set up something that isn't regularly used (such as restoring of a backup, a fail over, etc.) and simply assume it's working without any issue. That was precisely the reason I asked whether you have tested it.
Again, it wasn't meant as an offense, sorry!

> I wouldn't be so sure of that one, the configuration can evolve quickly with openwrt.

I have compared the previous configuration on 19.07 to the one from 21.02, specifically for mwan3 and couldn't find a difference.
With that being said, there is a major change between 19.07 compared to 21.02: swconfig vs DSA (for mvebu/cortexa9), which might be the real issue.

As I said in my initial post:
>  I am not an expert in networking at all, so it might be only a symptom of a totally different issue.

I can say with absolute certainty that fail over from wanA to wanB does not work and the ping issue I found was the only symptom I could find. Again, this might be totally unrelated but in lack of further evidence/symptoms, I started the [thread](https://forum.openwrt.org/t/mwan3-first-4-pings-drop-on-wlanb/154536) in the forum where I was asked to open this bug report to draw the attention of the developers.
We might be going down the complete wrong road and the ping issue is totally unrelated, but that's something I can neither confirm nor deny - I was hoping to get someone's attention that can either help diagnose the problem properly or help with troubleshooting.

metaphys, 2023-04-14:
Don't worry I didn't felt offended, i just wanted to be very clear from the same reason you mentionned. Still the issue remains with the ping and arp. And maybe with failover for traffic originating from the routeur itself

metaphys, 2023-04-14:
What I find strange in your forum post, is that you wrote that the issue only appears after a while. In my case, once mwan3 starts and is active, the ping problem will always be there.
Is your default policy of type balanced or wanA_wanB?

metaphys, 2023-04-14:
also can you share your mwan3 / network configuration?

brada4, 2023-04-14:
Enable keepalive option for PPPoE so that other end does not shut down yielding another 5s startup.
https://openwrt.org/docs/guide-user/network/wan/wan_interface_protocols#protocol_pppoe_ppp_over_ethernet

metaphys, 2023-04-14:
@brada4 the problem is not related to pppoe. It appears also on ip static only configuration.

metaphys, 2023-04-14:
@sscheib 
 can you config that by changing the default policy (wan_wanb vs wanb_wan) you are moving the problem from one interface to the other (to be sure we are actually having the same issue)

sscheib, 2023-04-14:
@metaphys
Good point, I have never actually tried it, thanks for pointing that out!
I'll try to reproduce it this weekend and update you here

sscheib, 2023-04-15:
@metaphys 
I have verified your assumption and it is correct.
As it turn out my test methodology was wrong in the first place. Fail over does *work* for all clients, but not the router itself. I tested from the router whether a fail over works, but that is unfortunately the only test that will fail. This, however, used to work with 19.07 and started failing on 21.02.

So to summarize: The issue is moved from one interface to another when switching the default policy (e.g. switch the policy from wan_wanb to wanb_wan). This issue will only prevent traffic originated from the router itself to fail over, not the clients connected to the router.

metaphys, 2023-04-18:
@sscheib
I could check that the failover is not working for traffic originating from the openwrt itself. I do have that issue also.
In my case it's not a problem of connectivity but of route. 
I could turnaround the issue by changing the default route when wan is going down with the default route of wanb (and back when it goes back up) in `/etc/mwan3.user` with (I have wan static ip and wanb pppoe):
```sh
if [ "$INTERFACE" = "wan" ] && [ "$ACTION" = "connected" ]; then
    ip route replace $(uci get network.@wireguard_cdr[0].endpoint_host) via $(uci get network.pub1.gateway) dev $(uci get network.pub1.device) proto static metric 10
fi
if [ "$INTERFACE" = "wanb" ] && [ "$ACTION" = "disconnected" ]; then
    ip route replace $(uci get network.@wireguard_cdr[0].endpoint_host) via «pppoe gateway» dev pppoe-wanb proto static metric 10
fi

I still see no evidence of the ping issue being related to this so we should probably create a new bug report for this?
```

sscheib, 2023-04-19:
I don't know if that's a separate issue, but we could surely try.
Like I said in my initial report: I have no clue if it's related to the fail over failure for router originated traffic.

marianomd, 2023-09-01:
I have the exact same problem here, on ramips/mt7621 version 22.03.5

erik-hamburg, 2023-09-05:
Seems I have same issue on GS308T (realtek/rtl838x) on 22.03.5
MWAN3 shown both connections available, but DDNS does not work on second line: ping lost the first 4 pings and DDNS can't check the actual IP, nor set the IP on ddns provider (no answer after 3 sec. because packet silently lost).
I didn't check if it works when MWAN3 is stopped because the router provides OPENVPN server for remote access. 
Connecting from outside over the second line works, so seems related to traffic from router itself. 
May be related to DSA because before I tried to upgrade an R1 from 19.0 to 22.3 but failed getting second line to work reliable and gave up

erik-hamburg, 2023-09-06:
Today morning, the AVM box on the first line was updated.
Afterwards:
- ddns service was stopped (no process running), so IP addresses were not updated. Restart of ddns processes failed
- openvpn on the second line could be successfully connected, but the network behind the openwrt box was not accessable
Thefore, I decided to use the old 19.x box instead

cucucucu2, 2023-09-12:
Same problem here, I'm using OpenWRT 22.03.3 x86.

I have two ISPs, one of which obtains a DHCP connection and the other via PPPoE, my configuration is ISP DHCP on WAN_A and ISP PPPoE on WAN_B.

When activating MWAN3 and placing both in loadbalance, only WAN_B is able to PING correctly, while WAN_A is left with lost packets.

tsian077, 2023-11-14:
HI!
I have the same problem with version 22.03.5 with two interface, does this issue have any updated?

Thanks.

sahilister, 2024-01-16:
Observing the same issue on RB750Gr3 running OpenWrt 21.02.5

brada4, 2024-01-16:
Does the issue persist in v23 ie fw4 and nftables?

sahilister, 2024-01-17:
Issue persists with fresh install of OpenWrt 23.05.0 (including fw4 and nftables).

See logs:
```
root@OpenWrt:/etc/config# ping -I phy0-sta0 9.9.9.9
PING 9.9.9.9 (9.9.9.9): 56 data bytes
64 bytes from 9.9.9.9: seq=4 ttl=53 time=67.675 ms
64 bytes from 9.9.9.9: seq=5 ttl=53 time=47.877 ms
64 bytes from 9.9.9.9: seq=6 ttl=53 time=72.222 ms
64 bytes from 9.9.9.9: seq=7 ttl=53 time=151.380 ms
64 bytes from 9.9.9.9: seq=8 ttl=53 time=51.482 ms
64 bytes from 9.9.9.9: seq=9 ttl=53 time=41.084 ms
64 bytes from 9.9.9.9: seq=10 ttl=53 time=52.128 ms
64 bytes from 9.9.9.9: seq=11 ttl=53 time=41.646 ms
64 bytes from 9.9.9.9: seq=12 ttl=53 time=59.680 ms
64 bytes from 9.9.9.9: seq=13 ttl=53 time=69.524 ms
64 bytes from 9.9.9.9: seq=14 ttl=53 time=49.704 ms
64 bytes from 9.9.9.9: seq=15 ttl=53 time=44.573 ms
```
 `tcpdump` during above `ping`:
```
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on phy0-sta0, link-type EN10MB (Ethernet), snapshot length 262144 bytes
13:16:54.327826 ARP, Request who-has 9.9.9.9 tell 192.168.61.85, length 28
13:16:55.367233 ARP, Request who-has 9.9.9.9 tell 192.168.61.85, length 28
13:16:56.407238 ARP, Request who-has 9.9.9.9 tell 192.168.61.85, length 28
13:16:58.328949 IP 192.168.61.85 > 9.9.9.9: ICMP echo request, id 12091, seq 4, length 64
13:16:58.396228 IP 9.9.9.9 > 192.168.61.85: ICMP echo reply, id 12091, seq 4, length 64
13:16:59.329274 IP 192.168.61.85 > 9.9.9.9: ICMP echo request, id 12091, seq 5, length 64
13:16:59.376807 IP 9.9.9.9 > 192.168.61.85: ICMP echo reply, id 12091, seq 5, length 64
13:17:00.329597 IP 192.168.61.85 > 9.9.9.9: ICMP echo request, id 12091, seq 6, length 64
13:17:00.401476 IP 9.9.9.9 > 192.168.61.85: ICMP echo reply, id 12091, seq 6, length 64
13:17:01.329927 IP 192.168.61.85 > 9.9.9.9: ICMP echo request, id 12091, seq 7, length 64
13:17:01.480948 IP 9.9.9.9 > 192.168.61.85: ICMP echo reply, id 12091, seq 7, length 64
13:17:02.330250 IP 192.168.61.85 > 9.9.9.9: ICMP echo request, id 12091, seq 8, length 64
13:17:02.381381 IP 9.9.9.9 > 192.168.61.85: ICMP echo reply, id 12091, seq 8, length 64
```

Though weird thing is no packet loss observed while doing pings via mwan3 diagnostics in luci. Logs from luci:
```
Running 'ping -c 5 -W 1 1.0.0.1' with DEVICE=phy0-sta0 SRCIP=192.168.61.85 FWMARK=0x3f00 FAMILY=ipv4
PING 1.0.0.1 (1.0.0.1): 56 data bytes
64 bytes from 1.0.0.1: seq=0 ttl=52 time=71.992 ms
64 bytes from 1.0.0.1: seq=1 ttl=52 time=55.227 ms
64 bytes from 1.0.0.1: seq=2 ttl=52 time=70.025 ms
64 bytes from 1.0.0.1: seq=3 ttl=52 time=50.851 ms
64 bytes from 1.0.0.1: seq=4 ttl=52 time=69.584 ms

--- 1.0.0.1 ping statistics ---
5 packets transmitted, 5 packets received, 0% packet loss
round-trip min/avg/max = 50.851/63.535/71.992 ms
Running 'ping -c 5 -W 1 1.1.1.1' with DEVICE=phy0-sta0 SRCIP=192.168.61.85 FWMARK=0x3f00 FAMILY=ipv4
PING 1.1.1.1 (1.1.1.1): 56 data bytes
64 bytes from 1.1.1.1: seq=0 ttl=52 time=68.370 ms
64 bytes from 1.1.1.1: seq=1 ttl=52 time=64.883 ms
64 bytes from 1.1.1.1: seq=2 ttl=52 time=67.105 ms
64 bytes from 1.1.1.1: seq=3 ttl=52 time=64.456 ms
64 bytes from 1.1.1.1: seq=4 ttl=52 time=65.298 ms

--- 1.1.1.1 ping statistics ---
5 packets transmitted, 5 packets received, 0% packet loss
round-trip min/avg/max = 64.456/66.022/68.370 ms
Running 'ping -c 5 -W 1 208.67.222.222' with DEVICE=phy0-sta0 SRCIP=192.168.61.85 FWMARK=0x3f00 FAMILY=ipv4
PING 208.67.222.222 (208.67.222.222): 56 data bytes
64 bytes from 208.67.222.222: seq=0 ttl=53 time=115.768 ms
64 bytes from 208.67.222.222: seq=1 ttl=53 time=125.900 ms
64 bytes from 208.67.222.222: seq=2 ttl=53 time=78.044 ms
64 bytes from 208.67.222.222: seq=3 ttl=53 time=94.989 ms
64 bytes from 208.67.222.222: seq=4 ttl=53 time=96.092 ms

--- 208.67.222.222 ping statistics ---
5 packets transmitted, 5 packets received, 0% packet loss
round-trip min/avg/max = 78.044/102.158/125.900 ms
Running 'ping -c 5 -W 1 208.67.220.220' with DEVICE=phy0-sta0 SRCIP=192.168.61.85 FWMARK=0x3f00 FAMILY=ipv4
PING 208.67.220.220 (208.67.220.220): 56 data bytes
64 bytes from 208.67.220.220: seq=0 ttl=53 time=74.503 ms
64 bytes from 208.67.220.220: seq=1 ttl=53 time=73.409 ms
64 bytes from 208.67.220.220: seq=2 ttl=53 time=71.657 ms
64 bytes from 208.67.220.220: seq=3 ttl=53 time=70.849 ms
64 bytes from 208.67.220.220: seq=4 ttl=53 time=91.641 ms

--- 208.67.220.220 ping statistics ---
5 packets transmitted, 5 packets received, 0% packet loss
round-trip min/avg/max = 70.849/76.411/91.641 ms
```

but running ping with same arguments in terminal shows packet loss again:
```
root@OpenWrt:/etc/config# ping -c 5 -W 1 1.0.0.1 -I phy0-sta0
PING 1.0.0.1 (1.0.0.1): 56 data bytes
64 bytes from 1.0.0.1: seq=4 ttl=52 time=199.566 ms

--- 1.0.0.1 ping statistics ---
5 packets transmitted, 1 packets received, 80% packet loss
round-trip min/avg/max = 199.566/199.566/199.566 ms
```

permarcoX, 2024-03-03:
> @permarcoX do you lose WAN connection from LAN or from router only?

Both, ping timeouts from inside the lan and from the router too, if I understood correctly.

If I stop the wan interface from Luci everything works fine, wanb takes place of wan and everything works fine
[image]

yurtesen, 2024-03-17:
I have the same issue on BananaPI R3. The pings get lost because system first tries to do an ARP lookup for no reason. Disabling MWAN3 solves the problem.

https://forum.openwrt.org/t/weird-arp-issue-on-with-multiwan-manager/191795

vpecinka, 2024-03-23:
The same here, Turris Omnia, `mwan3 2.10.13-1`:
interfaces `eth2` and `lan4`
2 default routes
```
default via 10.126.11.1 dev eth2 proto static metric 1 
default via 192.168.1.1 dev lan4 proto static metric 2 
```

when started mwan3, first 4 ping packets are lost on the lan4 interface.
```
root@turris:~# ping -I lan4 www.google.com
PING www.google.com (142.251.36.68) from 192.168.1.12 lan4: 56(84) bytes of data.
64 bytes from prg03s10-in-f4.1e100.net (142.251.36.68): icmp_seq=5 ttl=118 time=37.6 ms
64 bytes from prg03s10-in-f4.1e100.net (142.251.36.68): icmp_seq=6 ttl=118 time=35.2 ms
64 bytes from prg03s10-in-f4.1e100.net (142.251.36.68): icmp_seq=7 ttl=118 time=34.3 ms
```

However, when ping already running, another ping in different windows works fine. When both ping are stopped, the first works fine when run again in few seconds... otherwise again 4 packets drop...

after `/etc/init.d/mwan3 stop` ping works as usual.

Note to say it happens only on `lan4` not `eth2` interface. Both interfaces are configured in the same way (static IP)

balvik, 2024-05-14:
Turris Omnia `mwan3 2.11.7-1`

I just came to confirm the issue. I won't add any logs since they add nothing new and show pretty much the same as everyone else (identical similar to vpecinka above, just different names). Changing the interval between pings affects the number of pings lost of course (bigger interval gives more time for the ARP resolution so fewer pings lost, and vice versa). Both primary and backup WANs are plain old static IP networks, nothing unusual. 

Strange thing is that I saw this issue once when I first set up mwan3 a couple of months ago. The next day it mysteriously disappeared. Today the issue mysterious reappeared. I have an always-running process pinging both WANs every 10s and with the result showing on HW LEDs in my office so I notice the change immediately.

No SW update happened recently and I made no config changes.

yurtesen, 2024-07-28:
So nobody knows why pinging an IP which is not on the LAN still cause ARP requests to be sent? what causes this behavior?

metaphys, 2024-08-01:
from `https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3`

> mwan3 use was added in version 2.10. This additional option is designed to allow you test network commands like ping, iperf3 etc by binding the command to a specific interface reliably. A common side effect with mwan3 is it can skew the output of commands that rely on binding to specific interfaces, as traffic will be routed according to the rules defined in /etc/config/mwan3 and essentially override the desired scenario in some cases.

sscheib, 2024-08-04:
@metaphys Thanks for pointing that out.

As I said in my issue description, I wasn't sure whether that's a symptom of another issue.

This clears up that pings are lost - I am fine with that.

However, the real issue I pointed out was that on a failover the router originated traffic wouldn't go through - **any** traffic originating from the router itself will not work

This means that DNS isn't working (when OpenWrt is used as DNS server, which should be quite common) as long as the primary interface is down.

As smartphones in particular (but also Windows and some Linux desktop distributions, e.g. Fedora) try to reach a specific DNS to verify whether the Wi-fi/wired network connected to is providing internet access, this will cause all sorts of issues in the home network.

This could surely be worked around by not using OpenWrt as a DNS resolver (or not serving any service that requires a working internet connection for that matter), but IMHO that's not a sustainable workaround.

Of course this impacts other services such as NTP, etc. as well.

filippocarletti, 2024-10-11:
@sscheib could you please check if https://github.com/openwrt/packages/pull/25121 fixes the `ping -I` issue?

Neustradamus, 2024-12-28:
@ all, have you seen the latest @filippocarletti comment?
Have you tested?

@erik-hamburg: Have you tried on your GS308T (realtek/rtl838x)?

marianomd, 2024-12-28:
It's not happening anymore on OpenWrt 24.10.0-rc2 r28161-ea17e958b9.

Target: ramips/mt7621
Kernel Version: 6.6.63
mwan3 2.11.16-r5

Didn't apply aforementioned fix.

```
root@OpenWrt:~# ping -I pppoe-wan 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
64 bytes from 8.8.8.8: seq=0 ttl=119 time=15.321 ms
64 bytes from 8.8.8.8: seq=1 ttl=119 time=10.813 ms
64 bytes from 8.8.8.8: seq=2 ttl=119 time=11.380 ms
64 bytes from 8.8.8.8: seq=3 ttl=119 time=13.525 ms
64 bytes from 8.8.8.8: seq=4 ttl=119 time=12.771 ms
^C
--- 8.8.8.8 ping statistics ---
5 packets transmitted, 5 packets received, 0% packet loss
round-trip min/avg/max = 10.813/12.762/15.321 ms

root@OpenWrt:~# ping -I lan1 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
64 bytes from 8.8.8.8: seq=0 ttl=118 time=6.175 ms
64 bytes from 8.8.8.8: seq=1 ttl=118 time=6.323 ms
64 bytes from 8.8.8.8: seq=2 ttl=118 time=6.500 ms
64 bytes from 8.8.8.8: seq=3 ttl=118 time=5.557 ms
^C
--- 8.8.8.8 ping statistics ---
4 packets transmitted, 4 packets received, 0% packet loss
round-trip min/avg/max = 5.557/6.138/6.500 ms
```

Happy to provide any required additional information.

yiannis88, 2025-01-08:
> @sscheib could you please check if [openwrt/packages#25121](https://github.com/openwrt/packages/pull/25121) fixes the `ping -I` issue?

Not using the latest version but It fixes the issue in **mwan3-2.10.12**.

@filippocarletti, shouldn't we be deleting the 1500s rules in `mwan3_delete_iface_rules()`?

filippocarletti, 2025-01-10:
AFAIK, rules are deleted if their ID is between 1000 and 4000.
See https://github.com/openwrt/packages/blob/master/net/mwan3/files/lib/mwan3/mwan3.sh#L552

yiannis88, 2025-01-10:
Unless I misread the condition, they will need to satisfy this one as well `'$1 % 1000 == '$id'`, which will fail for the 1500s IDs?

filippocarletti, 2025-01-14:
You are right, the patch needs more work.
Let's continue on https://github.com/openwrt/packages/pull/25121

fathonix, 2025-09-13:
The same thing happens as well on MikroTik hAP ac2 (ipq40xx/mikrotik) running 24.10.2, kernel 6.6.93, mwan3 2.11.16-r5.