openwrt/openwrt #12288: Wrong timezone - /etc/TZ should not end with the new-line

perexg, 2023-03-29:
### Describe the bug

`/etc/init.d/system` script initializes /tmp/TZ file with `echo "$timezone" > /tmp/TZ` command which adds a new line to the end.

The library does not expect that:

```
# cat /etc/TZ
Europe/Prague
# date
Wed Mar 29 20:24:01 UTC 2023
# strace -e trace=open date 2>&1 | grep zoneinfo
open("/usr/share/zoneinfo/Europe/Prague\n", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/share/zoneinfo/Europe/Prague\n", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/etc/zoneinfo/Europe/Prague\n", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
```

### OpenWrt version

22.03.3

### OpenWrt target/subtarget

not relevant

### Device

not relevant

### Image kind

Official downloaded image

### Steps to reproduce

```
# cat /etc/TZ
Europe/Prague
# date
Wed Mar 29 20:24:01 UTC 2023
# strace -e trace=open date 2>&1 | grep zoneinfo
open("/usr/share/zoneinfo/Europe/Prague\n", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/share/zoneinfo/Europe/Prague\n", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/etc/zoneinfo/Europe/Prague\n", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
```

### Expected behaviour

```
# cat /etc/TZ
Europe/Pragueroot@openwrt:/# date
Wed Mar 29 22:27:56 CEST 2023
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

neheb, 2023-03-29:
I tested this. Seems to be an issue when zoneinfo is present. Adding -n does not fix nor break /etc/TZ usage.

perexg, 2023-03-30:
Nope, musl:

```
open("/etc/ld-musl-armhf.path", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libgcc_s.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
open("/etc/TZ", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = 3
open("/usr/share/zoneinfo/Europe/Prague", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = 3
open("/etc/TZ", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = 3
```

With '\n':

```
pen("/etc/ld-musl-armhf.path", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libgcc_s.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
open("/etc/TZ", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = 3
open("/usr/share/zoneinfo/Europe/Prague\n", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/share/zoneinfo/Europe/Prague\n", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/etc/zoneinfo/Europe/Prague\n", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/etc/TZ", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_CLOEXEC) = 3
```

ynezz, 2023-03-30:
This is on stock 22.03.3 image:

```
root@OpenWrt:/# ls -al /usr/share/zoneinfo/
ls: /usr/share/zoneinfo/: No such file or directory

root@OpenWrt:/# uci set system.@system[0].timezone='CET-1CEST,M3.5.0,M10.5.0/3'
root@OpenWrt:/# uci commit system

root@OpenWrt:/# date
Thu Mar 30 09:03:13 UTC 2023

root@OpenWrt:/# service system restart

root@OpenWrt:/# date
Thu Mar 30 11:03:19 CEST 2023

root@OpenWrt:/# cat /etc/TZ
CET-1CEST,M3.5.0,M10.5.0/3
```

Can you perhaps provide complete reproducer? What additional packages have you installed?

perexg, 2023-03-30:
You set TZ to value which is interpreted in libc directly in a simple format. I'm using zoneinfo string (see the issue description - `Europe/Prague` in my case) to use the definitions in `/usr/share/zoneinfo` tree (`zoneinfo-europe` package is installed).

Zoneinfo contains more data than DST (keeps history of DST changes for example).