openwrt/openwrt #12290: [mikrotik rbm33g mt7621] Easily bricked and OpenWRT corrupted *BOUNTY OFFERED*

benletchford, 2023-03-30:
### Describe the bug

I am utilizing the rbm33s in a manner where sudden power loss may occur periodically and connecting it to a UPS is not a practical solution. Unfortunately, it seems that the JFFS2 file-system is vulnerable to corruption if a write operation is in progress and the power is abruptly cut. As a result, the boot process is halted at "jumping to kernel code," and the device becomes inoperable (i.e., bricked).

[image]

From my preliminary investigations this is happening on at least 21.03 and 22.03 (haven't tested anything else at this stage).

### OpenWrt version

21.03, 22.03

### OpenWrt target/subtarget

ramips mt7621

### Device

rbm33g

### Image kind

Official downloaded image

### Steps to reproduce

1. Run `wget https://downloads.openwrt.org/releases/22.03.3/targets/ramips/mt7621/openwrt-22.03.3-ramips-mt7621-mikrotik_routerboard-m33g-squashfs-sysupgrade.bin -O /sysupgrade.bin`
2. Unplug the board mid-download. (This will trigger the bug ~50% of the time. I try to unplug after 600k has been downloaded.)

### Actual behaviour

The boot process gets stuck at "jumping to kernel code" and the device is bricked ~50% of the time.

### Expected behaviour

The system is able to successfully recover even if some file is corrupted.

### Additional info

You can fix a bricked board by loading it into an initramfs image and either erasing the corrupted partition with the command `mtd erase rootfs_data` or by mounting the partition and writing data to it until the bad blocks are removed through garbage collection.

This indicates that there should be a mechanism in the firmware to enhance the device's ability to boot, even if a single file becomes corrupted.

My short-term strategy is to minimize the number of writes to decrease the likelihood of triggering the bug. For example, we won't store persistent logs.

**Bounty Offer**

I'm offering a $1,500 USD reward for a software/firmware-only solution that significantly increases the firmware's resistance to this type of bricking (it's acceptable if some files are lost). We can collaborate here or you can contact me privately at me(at)benletchford.com

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-03-30:
You can netboot the device to recover (and use 64MB /tmp for temp files in future)

benletchford, 2023-03-30:
@brada4, recovering a device isn't the issue. Making it resistant to this kind of corruption while still supporting flash writes is.

I'm definitely using tmpfs where it makes sense. There's some cases where I am forced to perform flash writes.

gerber4, 2023-03-30:
The kernel+rootfs+rootfs_data blocks do not overlap. After using a netboot we checked the contents of the kernel and rootfs. They were identical to a working board. Also, it's still possible to mount rootfs_data while running the netboot image. Running `head -c 100k /dev/urandom > sysupgrade.bin ; sync` a couple times can fix a "corrupt" rootfs_data.

Since the kernel and rootfs partitions aren't corrupted and the rootfs_data is mountable, we think there's a problem in the kernel.

brada4, 2023-03-30:
Your flash chip can erase in 4K / 32K / 64K blocks, can you check alignment matches /sys/class/mtd/mtd0/erasesize ?

john-tho, 2023-03-31:
Would like to see / know the differences on NOR between a device that boots, and one that no longer boots. If the kernel bits are the same (NOR offsets 0x40000, around 0x30000 long), the kernel should boot. Writing to the OpenWrt rootfs (overlay) should not ever touch this kernel on NOR. Even if the rootfs squashfs, and/or overlay are damaged, the kernel should still boot and tell you it cannot load a root.

OpenWrt is currently using the vmlinuz self-extracting kernel ELF for these devices: https://elixir.bootlin.com/linux/latest/source/arch/mips/boot/compressed/decompress.c
There is no output configured for it in OpenWrt, but it would be worth enabling it to see if it is being starting, then getting stuck. You should be able to adapt this commit to your kernel version: https://github.com/john-tho/openwrt/commit/d31ddccde4ae29dd7ce225118387c16657fd1dea
Then, if the vmlinuz is being started, you should see: `zimage at:`, `Uncompressing Linux at load address`, `Copy device tree to address`, finally `Now, booting the kernel...`. Then you would normally see standard Linux console start: `Linux version`.
Let me know what you see there?

We switched to this self-extracting kernel from the OpenWrt LZMA decompressor after seeing users getting stuck with the LZMA decompressor. You could try reverting this: https://github.com/openwrt/openwrt/commit/254f51443ddaee873267d41993b5bbc4b5ef7959

I know that on my 760igs, my RouterBOOT is temperamental. It will regularly fail to download a BOOTP image, and often takes a few resets to successfully BOOTP load, so I am suspicious that RouterBOOT on these mt7621 devices has more bugs than usual.
What RouterBOOT versions are you seeing this issue? Have you tried changing the primary RouterBOOT version?

Also like to know if removing power, and reapplying power when stuck after `jumping to kernel code` ever gets to a booted kernel. If so, starting the watchdog in vmlinuz might get you through the bad boots.

---

Some extra background:
On NOR RouterBOOT boots from a YAFFS file system. This YAFFS fs starts from a set point after the various bootloaders and config: ARM hardware needs initial bootloader in addition, everything has; RouterBOOT initial loader, RouterBOOT backup, hard, soft, keys, primary RouterBOOT in various orders.
Historically, it boots the ELF file at `kernel`. This no longer works for RouterBOOT v7 (which boots the NPK at `bootimage`, using the ELF packed into the NPK filecontainer).
In OpenWrt, we use [kernel2minor](https://github.com/adron-s/kernel2minor) to pack the kernel ELF into a fake YAFFS image: https://github.com/openwrt/openwrt/blob/75e78bcaab847557ce1782eb2dea9dff9a029171/target/linux/ramips/image/mt7621.mk#L1520, and an mtd driver to find the point where kernel stops, and rootfs starts: https://github.com/openwrt/openwrt/blob/master/target/linux/generic/files/drivers/mtd/mtdsplit/mtdsplit_minor.c
RouterBOOT does have some tests to check if the `kernel` is an ELF file, but it cannot tell, or do anything if it is a valid ELF, but a section in has been clobbered.

benletchford, 2023-03-31:
Everything is aligned.

```
$ cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00040000 00010000 "RouterBoot"
mtd1: 0000f000 00010000 "bootloader1"
mtd2: 00001000 00010000 "hard_config"
mtd3: 0000f000 00010000 "bootloader2"
mtd4: 00001000 00001000 "soft_config"
mtd5: 00001000 00010000 "bios"
mtd6: 00f40000 00010000 "firmware"
mtd7: 00280000 00010000 "kernel"
mtd8: 00cc0000 00010000 "rootfs"
mtd9: 00360000 00010000 "rootfs_data"
mtd10: 00080000 00010000 "persistent"
```

```
root@OpenWrt:~# cat /sys/class/mtd/mtd0/erasesize
65536
root@OpenWrt:~# cat /sys/class/mtd/mtd1/erasesize
65536
root@OpenWrt:~# cat /sys/class/mtd/mtd2/erasesize
65536
root@OpenWrt:~# cat /sys/class/mtd/mtd3/erasesize
65536
root@OpenWrt:~# cat /sys/class/mtd/mtd4/erasesize
4096
root@OpenWrt:~# cat /sys/class/mtd/mtd5/erasesize
65536
root@OpenWrt:~# cat /sys/class/mtd/mtd6/erasesize
65536
root@OpenWrt:~# cat /sys/class/mtd/mtd7/erasesize
65536
root@OpenWrt:~# cat /sys/class/mtd/mtd8/erasesize
65536
root@OpenWrt:~# cat /sys/class/mtd/mtd9/erasesize
65536
root@OpenWrt:~# cat /sys/class/mtd/mtd10/erasesize
65536
```

brada4, 2023-03-31:
Offsets seem OK 64k=0x10000. Kernel and squashfs do not get written separately, so the bent border does not hurt them.

brada4, 2023-03-31:
In principle with broken jffs2 (i.e one block with erase+write lock) mtd should be able to boot to OpenWRT safe mode to permit you to erase overlay. But injustice happens.

gerber4, 2023-03-31:
Reverting [254f514](https://github.com/openwrt/openwrt/commit/254f51443ddaee873267d41993b5bbc4b5ef7959) doesn't resolve the issue. I tried downgrading to 19.07.9 and the bug is still there.
```
jumping to kernel code

OpenWrt kernel loader for MIPS based SoC
Copyright (C) 2011 Gabor Juhos <juhosg@openwrt.org>
Decompressing kernel... failed, data error!

System halted!
```

john-tho, 2023-03-31:
> @john-tho I got good results adding [john-tho@d31ddcc](https://github.com/john-tho/openwrt/commit/d31ddccde4ae29dd7ce225118387c16657fd1dea).
> 
> ```
> Press any key within 2 seconds to enter setup..
> m33 boot from CS1
> 
> loading kernel... OK
> setting up elf image... OK
> jumping to kernel code
> zimage at:     80793E90 809FA9D4
> Uncompressing Linux at load address 80001000
> 
> 
> XZ-compressed data is corrupt
> 
>  -- System halted
> ```
> 
> 
> The content of the kernel partition is unchanged so I don't know why it's considered corrupt.
> 
> ```
> Fresh Install
> sha256sum /dev/mtdblock7
> 1efa5a7f9b65b1e622da60b0bdcdc49b5723f672f5845dcafc92a56221190b94  /dev/mtdblock7
> head -c 3008k /dev/mtdblock8 | sha256sum
> b0ba74534e49fa08c860e82b82c5cf72ce883b7145ad17fc16facfa62cde7c2b  -
> 
> Bricked
> sha256sum /dev/mtdblock7
> 1efa5a7f9b65b1e622da60b0bdcdc49b5723f672f5845dcafc92a56221190b94  /dev/mtdblock7
> head -c 3008k /dev/mtdblock8 | sha256sum
> b0ba74534e49fa08c860e82b82c5cf72ce883b7145ad17fc16facfa62cde7c2b  -
> ```
> 
> Restarting the board doesn't resolve the issue. A bricked board stays bricked.

Great information. Very strange.

If you just want a workaround, you could try to switch to a different compressor, but this will make the kernel on flash slightly bigger:
https://elixir.bootlin.com/linux/v5.10.176/source/init/Kconfig#L209
through `make kernel_menuconfig CONFIG_TARGET=subtarget`
Some of the other decompressors look more chatty, and might give a more descriptive error.

If you do change compressors, I _maybe_ remember that for one of the compressors, OpenWrt sets the compression number too high for how much memory these devices have. Might be the `lzma` `-d20` here: https://github.com/openwrt/openwrt/blob/master/target/linux/generic/hack-5.10/230-openwrt_lzma_options.patch

---

I would really like to know _why_ this is happening, but you would need to add a few more print strings to `decompress.c`:
Very rough suggestions, probably won't work as-is:

We will want to make the `error` function non-blocking
We will to record the result of `__decompress` in `decompress.c`, so we can avoid `error` blocking, but still stop the vmlinuz https://elixir.bootlin.com/linux/v5.10.176/source/arch/mips/boot/compressed/head.S from continuing with bad data.
Also, at least for xz: https://elixir.bootlin.com/linux/v5.10.176/source/lib/decompress_unxz.c, it looks like we could provide a `long *pos` as the parameter before error (instead of 0). This would get incremented as the function works, so we could tell _where_ in the input the decompressor broke.
```diff
void error(char *x)
{
	puts("\n\n");
	puts(x);
	puts("\n\n -- System halted");
-
-	while (1)
-		;	/* Halt */
}
```

```diff
+	int ret = 0;
+	long *pos;
	/* Decompress the kernel with according algorithm */
-	__decompress((char *)zimage_start, zimage_size, 0, 0,
+	ret = __decompress((char *)zimage_start, zimage_size, 0, 0,
-		   (void *)VMLINUX_LOAD_ADDRESS_ULL, 0, 0, error);
+		   (void *)VMLINUX_LOAD_ADDRESS_ULL, 0, pos, error);
+
+	puts("pos where decompressor stopped:     ");
+	puthex(*pos);
+	if (ret < 0) {
+		puts("\nstall\n");
+		while (1)
+			;	/* Halt */
+	}
```

If none of that finds anything interesting, it might be worth modifying the `error` function more, so that you can print a message, and also print a number. Then, within the (configured) decompressor file make calls to error where the decompressor is setting an error code, and show numbers from the decompressor.
Even further, it might be worthwhile printing from the main vmlinuz, but it is assembly: https://elixir.bootlin.com/linux/v5.10.176/source/arch/mips/boot/compressed/head.S

---

Please let me know the RouterBOOT versions (also should be able to get/set from RouterBOOT cli).
From OpenWrt, this is:
```
cat /sys/firmware/mikrotik/soft_config/booter
cat /sys/firmware/mikrotik/soft_config/bios_version
cat /sys/firmware/mikrotik/hard_config/booter_version
```
If the backup booter is forced, maybe try to set it to regular, so we can try some different versions.
```
echo "regular">/sys/firmware/mikrotik/soft_config/booter
echo "y">/sys/firmware/mikrotik/soft_config/commit
```

gerber4, 2023-04-03:
Here is what I got.

Working 
```
loading kernel... OK
setting up elf image... OK
jumping to kernel code
zimage at:     80793F30 809FAA58
Uncompressing Linux at load address 80001000
pos where decompressor stopped:     88890003Now, booting the kernel...
[    0.000000] Linux version 5.4.188 (build@cdd45a104449) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r16554-1d4dea6d4f)) #0 SMP Sat Apr 16 12:59:34 2022
```

v.s

Bricked
```
loading kernel... OK
setting up elf image... OK
jumping to kernel code
zimage at:     80793F30 809FAA58
Uncompressing Linux at load address 80001000

XZ-compressed data is corrupt

 -- System haltedpos where decompressor stopped:     88890003
stall
```

They're reporting the same stop point. 

```
/sys/firmware/mikrotik/soft_config/booter = regular
/sys/firmware/mikrotik/soft_config/bios_version = 6.47.1
/sys/firmware/mikrotik/hard_config/booter_version = 6.46.3
```

This bug occurs on all of the boards we've tested and that includes different bootloader versions.

Can you think of any checksum registers in NOR flash that Linux could be checking? The bug occurs if we write to the persistent partition as well so it seems to be any write that can cause the issue.

john-tho, 2023-04-04:
> Here is what I got.
>
> Working
>
> ```
> loading kernel... OK
> setting up elf image... OK
> jumping to kernel code
> zimage at:     80793F30 809FAA58
> Uncompressing Linux at load address 80001000
> pos where [...]

Thank you.
I am suspicious that that is a memory address, not the dereferenced value at it, given zimage is `hex(0x809FAA58-0x80793F30) = '0x266b28'` long. Likely I got my demo code wrong somewhere as well, sorry.
But looking through https://elixir.bootlin.com/linux/v5.10.176/source/lib/xz/xz_dec_stream.c#L750, it seems as though xz will not give us the byte position anyway, as it sometimes resets it to zero in fail, so we may not get anything from it, sorry.

> This bug occurs on all of the boards we've tested and that includes different bootloader versions.

Thanks

> Can you think of any checksum registers in NOR flash that Linux could be checking? The bug occurs if we write to the persistent partition as well so it seems to be any write that can cause the issue.

Could you please tell me where the `persistent` partition is on NOR? That part of your DTS?

At this point here, RouterBOOT has loaded the vmlinuz ELF to memory (from NOR),
then the vmlinuz decompressor decompresses the bytes from memory. vmlinuz does not touch NOR.
Linux (vmlinux Linux, after vmlinuz decompressor finishes and jumps to it) does not access NOR until well after it is up and running, past the console line `Linux version 5.4.188`, when the NOR is recognised, and later the rootfs is loaded from NOR.

Some thoughts:
Given that we know the kernel bytes on NOR are okay (same kernel checksum on a netbooted broken board as on a working board):
- It is possible that RouterBOOT is reading data _somewhere_ from NOR, and identifying some condition (or it might see the power failure from watchdog), then setting bytes at a fixed position in memory after it has loaded the ELF. RouterOS kernel ELF is much smaller than our kernel, so this could clobber our xz data. Or RouterBOOT is reading the ELF wrong. I vaguely remember RouterBOOT can load the initrd ELF section to memory, one page after the end of the kernel, with `INRD` initrd_size in the bytes before this page boundary.
- Less likely: NOR chip, or SoC gets into a state that is not reset by RouterBOOT.
  Given:
  > You can fix a bricked board by loading it into an initramfs image and either erasing the corrupted partition with the command mtd erase rootfs_data or by mounting the partition and writing data to it until the bad blocks are removed through garbage collection.
  
  and
  > Also, it's still possible to mount rootfs_data while running the netboot image. Running `head -c 100k /dev/urandom > sysupgrade.bin ; sync` a couple times can fix a "corrupt" rootfs_data.

  This makes it seem like RouterBOOT is checking a byte on NOR that we are not aware of.

I think what I would like to do, is find the difference on NOR that causes this boot failure.
- Activate kernel config [`MTD_PARTITIONED_MASTER`](https://elixir.bootlin.com/linux/latest/K/ident/CONFIG_MTD_PARTITIONED_MASTER). Note this will change your mtd numbering . This would allow us to see and check all the NOR bytes: OpenWrt option here: https://github.com/john-tho/openwrt/commit/c349fb4a223020658d3280ecbf8f9cbdfc46e2d4 allows you to set it at the OpenWrt level, so it is not lost each OpenWrt rebuild. That way we can take full before and after images of the master of both NOR chips, and check if bytes are being changed outside of the `rootfs_data` or `persistent` partitions bounds.
- See if you can trigger this error when writing consistent data (easier to compare against). Example would be netboot, then erase rootfs_data and persistent, then start writing `\x00`'s, direct to NOR within those partitions. The roofts_data erased device  should still boot Linux, but then complain it cannot find a rootfs. This is good, netboot and continue writing untill kernel decompress fails to boot.
- Might not get anywhere, but would like to see the parameters (linux cmdline) that RouterBOOT is supplying the decompressor, but would need to modify the decompressor head.S for that: https://elixir.bootlin.com/linux/latest/source/arch/mips/ralink/prom.c#L36

Other options:
See if you can get this same fail-to-decompress error using one of the alternate vmlinuz decompressors. May need this commit to be able to netboot the initramfs, if the kernel gets too big: https://github.com/john-tho/openwrt/commit/b2067c792923a966773b6383862e921b0ca1899b

Fair bit of poking around to get it to work, but you could load u-boot, load the bytes from kernel partition on NOR, and `bootelf` them. Could either netboot the u-boot ELF, or write it to NOR in the reserved part of spi1 and swap the board boot jumpers (would probably also need to change the spi numbers in DTS then), then within U-Boot. u-boot has some mt7621 devices in tree, or there was a u-boot built in original m33g PR: https://github.com/TobleMiner/u-boot-ralink/tree/master/MikroTik-RBM33G

Very unlikely long shot (the ELF processing for within the NPK is likely the same as classic ELF processing), try the NPK packed kernel: https://github.com/openwrt/openwrt/compare/master...john-tho:openwrt:routerboot-v7

gerber4, 2023-04-04:
I'm encountering some odd behavior. Here's what I've done so far:

I began with our base image, combined it with the verbose decompress.c, and enabled MTD_PARTITIONED_MASTER. The board booted as expected.

```
loading kernel... OK
setting up elf image... OK
jumping to kernel code
zimage at:     80793FA0 809FAB28

Uncompressing Linux at load address 80001000

pos where decompressor stopped:     88890003Now, booting the kernel...

[    0.000000] Linux version 5.4.188 (build@3f52adc6ee79) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r16554-1d4dea6d4f)) #0 SMP Sat Apr 16 12:59:34 2022
```

Two new partitions, spi0.0 and spi0.1, appeared. I then proceeded with testing.

I executed wget --no-check-certificate https://downloads.openwrt.org/releases/22.03.3/targets/ramips/mt7621/openwrt-22.03.3-ramips-mt7621-mikrotik_routerboard-m33g-squashfs-sysupgrade.bin -O /dev/mtdblock11 and let it finish.

After rebooting the board, things got strange. The new log messages were absent.

```
loading kernel... OK
setting up elf image... OK
jumping to kernel code
[    0.000000] Linux version 5.10.161 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r20028-43d71ad93e) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Tue Jan 3 00:24:21 2023
```

Both the spi0.0 and spi0.1 partitions were now missing.

However, when I run mtd erase rootfs_data and reboot, the original messages and partitions are restored.

It seems like RouterBOOT is selecting the kernel in rootfs_data instead of the correct one. This could be why the board gets bricked – RouterBOOT finds the corrupted kernel and attempts to boot from it.

gerber4, 2023-04-04:
Additionally, I tried writing only a portion of the image to see if I could reproduce the original issue. I ran `head -c 800k image > /dev/mtdblock11` to write just the first 800 KB of the image to the partition. After doing this, I encountered the same problem as before, where the board got bricked.

john-tho, 2023-04-04:
Very nice digging.
Because the sysupgrade image contains the kernel packed into the fake YAFFS partition, it has that magic: YAFFS->file named `kernel`->ELF file. Is this a jffs2 partition? I will need to look into how they store data.
It might be that RouterBOOT is searching from the end (or at least further along) the NOR than our kernel is first, or has different priorities.
To check this, maybe try to move your persistent part before kernel, write the sysupgrade there, then see if it gets selected.

john-tho, 2023-04-05:
Okay.
I will need to try to reverse RouerBOOT, or do a lot of guessing and testing to find why RouterBOOT is not stopping at the first YAFFS `kernel` ELF on NOR.

Let me know if:
- you can make this happen with anything other than the start of a sysupgrade?
- you can work around this, or you would prefer not to need to?

Michał Kępień just released a tool that can write to a YAFFS filesystem without needing to have YAFFS compiled into the kernel. It is designed and tested for an ath79 NAND device, but with some adaptation, [could be made to work for NOR as well](https://github.com/kempniu/yafut/issues/4). Using that, the kernel would not need to be YAFFS packed within the sysupgrade file, so RouterBOOT _should not_ find it.

xback, 2023-04-11:
I experienced the same issue with corrupted files on powercuts on imx6 target,  especially when using UBI (NAND)
We heavily reduced this issue over here by simply adding following to the kernel cmdline:

`rootflags=sync`

dmesg will report:  "parsed rootflags" or so when this option is encountered during driver loading.

The problem seems to be that UBI does not flush all writes immediately to nand to reduced wear-tear.
Above cmdline option will flush all changes directly to nand, reducing the issue hugely.

As long as you don't write gigabytes of data continuously, the wear-tear will not be noticeable.

Hope this helps :-)

niseva, 2023-07-22:
We also have number of devices in field and get the same issue. The difference is that we are not writing anything intentionally. We have LTE modem installed on RBM33G. We also have some RBM11G as well and experience similar issues. @gerber4 were you able to solve this issue?

niseva, 2023-07-22:
@john-tho Any pointers on how to install U-Boot bootloader on RBM33G? I have seen https://github.com/TobleMiner/u-boot-ralink but don't know how to compile it and flash it on the RBM33G device.

john-tho, 2023-07-22:
> We also have number of devices in field and get the same issue. The difference is that we are not writing anything intentionally. We have LTE modem installed on RBM33G. We also have some RBM11G as well and experience similar issues. @gerber4 were you able to solve this issue?

Unless this is specifically the same issue in: RouterBoot loading kernel from an OpenWrt sysupgrade image later in NOR than installed kernel, open a new issue so that we can look into what is happening.

---

> @john-tho Any pointers on how to install U-Boot bootloader on RBM33G? I have seen https://github.com/TobleMiner/u-boot-ralink but don't know how to compile it and flash it on the RBM33G device.

From the looks of this device, it should be much easier than usual. Guessing the two SPI NOR chips, and the boot jumper headers on the board _should_ allow you to switch to booting from the other NOR chip. If I had one of these, I would have built u-boot from upstream https://source.denx.de/u-boot/u-boot/-/commit/0eaee027f7f352666d583d67d85cdd64d79ef121, and put this on the other NOR chip (at the same offset as RouterBOOT, `0`).

May be worth reaching out to your distributor (or Mikrotik) to see if they can describe what the boot selector header options are.
See console output:
```
m33 boot from CS1
```
and _Boot configuration jumpers_ part of manual: https://help.mikrotik.com/docs/display/UM/RBM33G

niseva-openwrt, 2023-07-22:
@john-tho It is the same issue I am facing as original post. 
I had asked Mikrotik support about this jumper and they replied that that the default jumper setup is what the board support and did not tell how to boot from other flash chip.
The description of the board is availble at https://help.mikrotik.com/docs/display/UM/RBM33G
and jumpers are mentioned in "Boot configuration jumpers"

There are 2 bootloaders in the first flash chip. One is called main and other backup. When using RouterOS, one can flash the main bootloader using their WinBox software. In case main bootloader becomes currupt, one can boot the board using backup bootloader by pressing reset switch for 10 seconds.

I have compiled OpenWRT and have ported few devices looking at architecture and dts files. However, I have never ventured into compiling bootloader. 
Are there any instructions to compile u-Boot? How do I flash it? can we use mtd utility to write the bootloader file into the mtd partition directly? As RBM33G has backup bootloader, I can always netboot device and recover it, so me making mistakes won't be a big trouble.

The reason why I feel bootloader trying to load from rootfs_data is the root cause because erasing rootfs_data always reverts the board to working condition. So, trying u-Boot bootloader seems to be the way forward.

john-tho, 2023-07-22:
> @john-tho It is the same issue I am facing as original post.

Then do one of:
- don't store sysupgrade on flash
- build image with upgrade process that uses `yafut`

> There are 2 bootloaders in the first flash chip. One is called main and other backup. When using RouterOS, one can flash the main bootloader using their WinBox software. In case main bootloader becomes currupt, one can boot the board using backup bootloader by pressing reset switch for 10 seconds.

This is not accurate. You need to understand the difference between the bootloader that the SoC starts, and a program subsequently loaded by that bootloader.

> I have compiled OpenWRT and have ported few devices looking at architecture and dts files. However, I have never ventured into compiling bootloader. Are there any instructions to compile u-Boot? How [...]

No, if you are modifying the bootloader, you will not have a recovery option. Modifying the RouterBoot primary payload is different. Best use a SOIC8 clip, and a method of flashing the NOR chip.

niseva-openwrt, 2023-07-22:
@john-tho 
On System-> Backup/ Flash Firmware menu (http://192.168.1.1/cgi-bin/luci/admin/system/flash), in "Save MTD contents" dropdown, I can see bootloader1 and bootloader2 options. The mikrotik WinBox software also updates bootloader1 mtd. Will this partition (mtd) holds U-Boot? Or my understanding is wrong?

Regarding Yafut, is there any reference I can use to generate sysupgrade to see if Yafut work?

danpawlik, 2023-07-24:
> @john-tho On System-> Backup/ Flash Firmware menu (http://192.168.1.1/cgi-bin/luci/admin/system/flash), in "Save MTD contents" dropdown, I can see bootloader1 and bootloader2 options. The mikrotik [...]

Maybe @kempniu will know

kempniu, 2023-07-27:
Sorry, while the Yafut code in its `main` branch technically supports reading/writing from/to NOR flash, I have not yet tackled the challenge of wiring it into the OpenWRT sysupgrade machinery.  I plan to do so in the foreseeable future as I have some ideas, but I have not yet implemented any of them (not even as a proof-of-concept), so - in the worst case - this may not happen at all.

Cuciyabgu, 2026-09-10:
@benletchford I opened #25121 with a software-only mitigation for the RouterBOOT/sysupgrade-copy mechanism documented in this issue. It changes the downloadable MikroTik NOR sysupgrade representation to deterministic gzip while preserving the decompressed YAFFS+rootfs flash layout.

Before I represent this as bounty-qualified, could you confirm whether the original $1,500 bounty offer is still active and whether this mitigation would qualify if physical RBM33G testing confirms that interrupted copies no longer cause RouterBOOT to select a truncated kernel?

I do not have a physical RBM33G attached here, so I am not claiming hardware validation. I have preserved an exact A/B software evidence set and a hardware test protocol; device-owner testing is still the empirical gate.