openwrt/openwrt #12294: [22.03] Kernel panic due to mac80211, mt76: add fixes for recently discovered security issues patch

db260179, 2023-03-30:
### Describe the bug

Recent patch added in the 22.03 snapshot branch - https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=4ae854d05568bc36a4df2cb6dd8fb023b5ef9944

Testing on Xiaomi 4a gigabit router model - other ramips most likely affected

This patch will produce the following kernel crash
`26.260018] CPU 2 Unable to handle kernel paging request at virtual address 0000000c, epc == 82ca0504, ra == 82ca04b0
[   26.270739] Oops[#1]:
[   26.273042] CPU: 2 PID: 1656 Comm: hostapd Not tainted 5.10.176 #0
[   26.279213] $ 0   : 00000000 00000001 00000000 00800004
[   26.284473] $ 4   : 82cdbec0 000000fe 00000000 0000005b
[   26.289723] $ 8   : 00000030 803af28c 00000001 00000006
[   26.294967] $12   : 81024d00 00000400 00000000 00000000
[   26.300211] $16   : 82c8ce70 00000000 82cda040 82c8d130
[   26.305456] $20   : 00000000 00000000 82c8c4e0 00000000
[   26.310706] $24   : 00000001 82640178                  
[   26.315955] $28   : 82d9a000 82d9bca0 82589900 82ca04b0
[   26.321201] Hi    : 51eb6307
[   26.324082] Lo    : 51eb674a
[   26.326991] epc   : 82ca0504 0x82ca0504 [mt76x02_lib@74d79ea0+0xa020]
[   26.333434] ra    : 82ca04b0 0x82ca04b0 [mt76x02_lib@74d79ea0+0xa020]
[   26.339860] Status: 1100fc03	KERNEL EXL IE 
[   26.344069] Cause : 4080000c (ExcCode 03)
[   26.348076] BadVA : 0000000c
[   26.350964] PrId  : 0001992f (MIPS 1004Kc)
[   26.355053] Modules linked in: pppoe ppp_async pppox ppp_generic nft_redir nft_nat nft_masq nft_flow_offload nft_fib_inet nft_ct nft_chain_nat nf_nat nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet nf_flow_table nf_conntrack_netlink nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY slhc sch_cake nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_quota nft_objref nft_numgen nft_log nft_limit nft_hash nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_counter nf_tables nf_reject_ipv6 nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact xt_set x_tables
[   26.355645]  ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ifb tun sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac leds_gpio gpio_button_hotplug crc32c_generic
[   26.482145] Process hostapd (pid: 1656, threadinfo=5e19ff5d, task=03e9485c, tls=77e72df0)
[   26.490315] Stack : 00000001 00000000 82cda040 8258bd28 82c8c4e0 82c8c000 82cd8da0 00000001
[   26.498703]         ffffffff 82c03f5c 00000000 00000000 8258990c 00000000 82c8c4e8 82c1d240
[   26.507092]         8258990c 80e98600 82c80fd0 ffffffe6 82c8c000 00000000 82c73640 00000000
[   26.515480]         82c8c18c 00000000 8258990c 82c1d574 00003a84 80187504 82d9bd24 82c8c000
[   26.523868]         00000000 82c8c000 00000001 80528e24 00000002 00000001 00000002 80725044
[   26.532254]         ...
[   26.534718] Call Trace:
[   26.534841] [<82c03f5c>] 0x82c03f5c [mac80211@b8756d94+0x81320]
[   26.543225] [<82c1d240>] 0x82c1d240 [mac80211@b8756d94+0x81320]
[   26.549194] [<82c1d574>] 0x82c1d574 [mac80211@b8756d94+0x81320]
[   26.555108] [<80187504>] 0x80187504
[   26.558633] [<80528e24>] 0x80528e24
[   26.562148] [<8052932c>] 0x8052932c
[   26.565662] [<805293c8>] 0x805293c8
[   26.569154] [<8051ae94>] 0x8051ae94
[   26.572642] [<8060d194>] 0x8060d194
[   26.576160] [<8061195c>] 0x8061195c
[   26.579664] [<8051b02c>] 0x8051b02c
[   26.583154] [<80556c4c>] 0x80556c4c
[   26.586647] [<80175c18>] 0x80175c18
[   26.590141] [<804f580c>] 0x804f580c
[   26.593628] [<8017608c>] 0x8017608c
[   26.597156] [<801c8d6c>] 0x801c8d6c
[   26.600649] [<80014578>] 0x80014578
[   26.604152] 
[   26.605643] Code: 02442021  00001025  ac930000 <a4c5000c> 8fbf0024  8fb40020  8fb3001c  8fb20018  8fb10014 
[   26.615424] 
[   26.617495] ---[ end trace 8aeffa40a27352d5 ]---
[   26.622227] Kernel panic - not syncing: Fatal exception
[   26.627493] Rebooting in 3 seconds..`

Thus bricking the router, until you turn off the wireless from failsafe mode, will stay booted once this is done

### OpenWrt version

r20124+69-e58ecccfac

### OpenWrt target/subtarget

ramips/mt7621

### Device

Xiaomi Mi Router 4A Gigabit Edition

### Image kind

Self-built image

### Steps to reproduce

Build latest 22.03 master branch containing patch - 4ae854d05568bc36a4df2cb6dd8fb023b5ef9944

### Actual behaviour

Enabling wifi will produce the kernel oops message as reported

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

hayzamjs, 2023-03-30:
Similar crash for me too on banana pi r3

```
[   14.357586] urngd: v1.0.2 started.
[   14.435459] kmodloader: loading kernel modules from /etc/modules.d/*
[   14.486685] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000038
[   14.495531] Mem abort info:
[   14.498328]   ESR = 0x0000000096000006
[   14.502060]   EC = 0x25: DABT (current EL), IL = 32 bits
[   14.507352]   SET = 0, FnV = 0
[   14.510400]   EA = 0, S1PTW = 0
[   14.513531]   FSC = 0x06: level 2 translation fault
[   14.518405] Data abort info:
[   14.521283]   ISV = 0, ISS = 0x00000006
[   14.525100]   CM = 0, WnR = 0
[   14.528060] user pgtable: 4k pages, 39-bit VAs, pgdp=0000000042578000
[   14.534487] [0000000000000038] pgd=0800000043d62003, p4d=0800000043d62003, pud=0800000043d62003, pmd=0000000000000000
[   14.545101] Internal error: Oops: 96000006 [#1] SMP
[   14.549966] Modules linked in: x_tables(+) nf_reject_ipv6 ifb ip6_udp_tunnel udp_tunnel snd_rawmidi snd_seq_device snd_pcm_oss snd_mixer_oss snd_hwdep snd_compress snd_pcm snd_timer snd souni
[   14.586612] CPU: 1 PID: 1073 Comm: kmodloader Not tainted 5.15.104 #0
[   14.593048] Hardware name: Banana-Pi R3 (DT)
[   14.597046] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   14.604000] pc : xt_find_table+0x154/0x18c [x_tables]
[   14.609049] lr : ops_init+0x3c/0x140
[   14.612616] sp : ffffffc00a7cba20
[   14.615914] x29: ffffffc00a7cba20 x28: ffffffc0009d6250 x27: 0000000000000024
[   14.623035] x26: 0000000000000001 x25: ffffffc0080bb430 x24: ffffffc008b56200
[   14.630166] x23: ffffffc008b56260 x22: ffffffc008bacdc0 x21: 0000000000000007
[   14.637288] x20: ffffff8000f94700 x19: ffffffc0009d61a0 x18: 0000000000000000
[   14.644419] x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000403711
[   14.651540] x14: 0000000000000001 x13: 0000000000000001 x12: 0000000000000006
[   14.658679] x11: ffffffc008a83f80 x10: 0000000000000000 x9 : 0000000000000000
[   14.665808] x8 : ffffff8000f94800 x7 : 0000000000000000 x6 : 000000000000003f
[   14.672930] x5 : 0000000000000040 x4 : 0000000000000000 x3 : 0000000000000004
[   14.680061] x2 : 0000000000000000 x1 : 0000000000000007 x0 : 0000000000000000
[   14.687183] Call trace:
[   14.689629]  xt_find_table+0x154/0x18c [x_tables]
[   14.694332]  ops_init+0x3c/0x140
[   14.697550]  register_pernet_operations+0xd8/0x230
[   14.702324]  register_pernet_subsys+0x2c/0x50
[   14.706669]  init_module+0xe8/0x1000 [x_tables]
[   14.711193]  do_one_initcall+0x4c/0x1b0
[   14.715015]  do_init_module+0x40/0x1cc
[   14.718756]  load_module+0x1bfc/0x2414
[   14.722506]  __do_sys_init_module+0x118/0x1b0
[   14.726853]  __arm64_sys_init_module+0x18/0x20
[   14.731295]  invoke_syscall.constprop.0+0x4c/0xe0
[   14.735991]  do_el0_svc+0x40/0xd0
[   14.739308]  el0_svc+0x14/0x50
[   14.742351]  el0t_64_sync_handler+0xe0/0x110
[   14.746605]  el0t_64_sync+0x158/0x15c
[   14.750255] Code: 910003fd b941e821 f9000bf3 f944f400 (f8615813) 
[   14.756335] ---[ end trace 1e27fa035003ea4c ]---
[   14.764460] Kernel panic - not syncing: Oops: Fatal exception
[   14.770195] SMP: stopping secondary CPUs
[   14.774128] Kernel Offset: disabled
[   14.777610] CPU features: 0x00000000,20000802
[   14.781951] Memory Limit: none
[   14.788503] Rebooting in 1 seconds..
```

nbd168, 2023-03-30:
Could you please send me your .config? I can't reproduce these issues

hayzamjs, 2023-03-30:
I have reverted to 9186da4027bc8d23912621b5471e98e9526f2878 for now and was able to get my device working again, here's the full config: https://pastebin.com/Fmyc3EUS

nbd168, 2023-03-30:
@hayzamjs, I made a fresh build from master with your exact config and couldn't reproduce any crashes on my banana pi r3

nbd168, 2023-03-30:
@db260179, based on your reported openwrt version, you seem to have some local commits in your tree. Anything that could be related?

hayzamjs, 2023-03-30:
> @hayzamjs, I made a fresh build from master with your exact config and couldn't reproduce any crashes on my banana pi r3

Weird, does it matter that I did a sysupgrade?

nbd168, 2023-03-30:
I also flashed using sysupgrade. Can you reproduce the problem by making a new build and flashing it again?

nbd168, 2023-03-30:
One other potential reason I can think of for this crash is that maybe your sysupgrade went wrong and only flashed the kernel. Commit 75e78bcaab847557ce1782eb2dea9dff9a029171 changed kernel headers in a way that old kernel modules would be incompatible and crashy when loaded with a new kernel.

hayzamjs, 2023-03-30:
Flashing instead of sysupgrading worked fine

db260179, 2023-03-31:
Have rebuilt and rerun the imagebuilder - now that the openwrt repo kmod are up to date with this change no more kernel oops!

Closing issue

closed 2023-03-31 by db260179