openwrt/openwrt #12296: Remote DoS Attack over Networks NATed with OpenWrt

brafumSec, 2023-03-31:
### Describe the bug

The session mappings in OpenWrt maintained for TCP connections can be removed by off-path attackers on the Internet via crafted TCP RST packets. After that, the attacker crafts and injects TCP SYN-ACK packets into the target TCP connection to trigger the challenge ACK mechanism on the connection, i.e., manipulating the state of the connection to cause an inconsistency between the server and the client. Consequently TCP connections issued from NATed clients will be terminated, leading to a DoS attack.

### OpenWrt version

r19803-9a599fee93

### OpenWrt target/subtarget

x86/generic

### Device

Xiaomi Mi Router AC1200

### Image kind

Official downloaded image

### Steps to reproduce

[steps.pdf](https://github.com/openwrt/openwrt/files/11117205/steps.pdf)
1. The attacker impersonates the victim server to craft multiple TCP RST packets to the public IP address of the NAT device. The sequence number of the crafted RST packets is arbitrary, and the acknowledgment number is not required.
2. The crafted TCP RST packets with correct port number will trigger the NAT device to remove session mappings for the remote server.
3. The attacker impersonates the NAT device (i.e., using the public IP address of the NAT device) to craft TCP SYN-ACK packets to the server. The source port number is specified within the range that TCP usually selects. The sequence number and the acknowledgment number are arbitrary.
4. For SYN-ACK packets with source port equal to the prior established TCP connections’ source port, the server will issue a challenge ACK packet to the public IP address of the NAT device. The acknowledgment number of the challenge ACK packet will be specified as the client’s exact sequence number, i.e., the lower boundary of the server’s receive window (rcv.nxt).
5. The NAT device will issue a TCP RST packet to the server for each received challenge ACK packet. This is because the session mappings for the TCP connections have been removed previously by the attacker.
6. The RST packet that carries the exact sequence number of the client will deceive the server into tearing down the corresponding socket.
7. As a result, the server will discard normal TCP data from the client. Meanwhile, the server will issue RST packets to the client to tear down the TCP connection, leading to a DoS attack.

### Actual behaviour

TCP connections between the internal NATed clients and the remote server are terminated.

### Additional info

Enabling OpenWrt’s validation by default in the kernel to verify the sequence number of the received RST packets is one possible countermeasure, i.e., setting the parameter of “nf_conntrack_tcp_no_windows_check” to 0 to prevent the remote attacker's removal of the session mappings. 

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brafumSec, 2023-03-31:
In our measurement study, we found that this attack affects many NAT products based on OpenWrt and a great deal of real-world NAT networks. For example, many Wi-Fi networks using OpenWrt are vulnerable.

brada4, 2023-03-31:
To my understanding you say that RST can be sent to firewall and ctstate gets dropped with any sequence number based only on port match i.e knowing service ip+port and source IP <64k packets to kill connection? In principle ICMP unreachable can be injected to kill route in 2000s style knowing just endpoint IP-s, but this exhibit adds unnecessary exposure.

Does it affect  current, nftables based revision
 OpenWrt 22.03.3, r20028-43d71ad93e
And does it affect last pure iptables version
 OpenWrt 21.02.5, r16688-fa9a932fdb
?

brafumSec, 2023-03-31:
In our observation, even if a crafted ICMP unreachable message traverses the NAT device and reaches the internal host, the host will still perform a check on the payload of at least 28 octets in the message (e.g., checking the sequence number) to verify the legitimacy of the message. By contrast, the crafted RST packets will not be checked by the NAT device. 

Yes, we observe that 22.03.3 and 21.02.5 are vulnerable.

brafumSec, 2023-03-31:
> To my understanding you say that RST can be sent to firewall and ctstate gets dropped with any sequence number based only on port match i.e knowing service ip+port and source IP <64k packets to kill [...]

In our observation, even if a crafted ICMP unreachable message traverses the NAT device and reaches the internal host, the host will still perform a check on the payload of at least 28 octets in the message (e.g., checking the sequence number) to verify the legitimacy of the message. By contrast, the crafted RST packets will not be checked by the NAT device, thus removing the NAT's mappings.

Yes, we observe that 22.03.3 and 21.02.5 are vulnerable.

jow-, 2023-03-31:
Disabling of TCP window checking has been removed in cbe73ea33d027dbb4b2cf1eca947ae746119e7d2 for openwrt-21.02, c2331038b2577cfeb6d34b3b6a812c6c1cf7ba5f for openwrt-22.03 and 75e78bcaab847557ce1782eb2dea9dff9a029171 for master.

brafumSec, 2023-03-31:
> Disabling of TCP window checking has been removed in [cbe73ea](https://github.com/openwrt/openwrt/commit/cbe73ea33d027dbb4b2cf1eca947ae746119e7d2) for openwrt-21.02, [c233103](https://github.com/openwrt/openwrt/commit/c2331038b2577cfeb6d34b3b6a812c6c1cf7ba5f) for openwrt-22.03 and [75e78bc](https://github.com/openwrt/openwrt/commit/75e78bcaab847557ce1782eb2dea9dff9a029171) for master.

Great job!

brada4, 2023-03-31:
ICMP is out of scope. I just mentioned other source of similar "disconnect" effect. things like network unreachabe dont have IP identifiers to match against, no socket at routing table level etc. They are usually not acted upon (but not always)

So what is the effect?
a) ctstate gets removed by wrong-sequence RST
b) misguided RST reaches NAT user and discards there
???

brafumSec, 2023-03-31:
The effect is that TCP connections between the internal hosts and a remote server will be cut off, causing a DoS attack launched at the external realm.

brada4, 2023-03-31:
At which point "cut off" happens?

a/ firewall state disappears from OOB rst
b/ OOB rst reaches internal host
B/ that RST is actually tcp-seq corrected? both happen?

What is affected (four code paths, you use some snapshot no idea on which side of things it is):
iptables-legacy (with or without offload)
nftables (...)

Without offload - quite verbatim linux kernel netfilter (or their old ipt-ables)
With offload - openwrt

brafumSec, 2023-03-31:
After the attacker completes the attack steps, the server and the client will tear down their TCP sockets, so I mean the connection is cut off by the attacker.  The sessions (defined in RFC 5382)  preserved in the NAT device for TCP connections are directly affected by the crafted RST packet.

brada4, 2023-03-31:
What do hou mean by 'arbitrary' sequence number? Is it any like zero or you need to guess?

brafumSec, 2023-03-31:
The padding of the sequence number may vary depending on the versions and the TCP connection status. If you are interested, I will send you our research paper, which provides a detailed description of this issue.

brada4, 2023-04-01:
I have tried with zero did not work. 
It is linux iptables or netfilter, not really specific to OpenWRT. You dont say whats your server, so i used netcat on OpenBSD, guessing ports from netstat-s.

brafumSec, 2023-04-01:
My server is Ubuntu 22.04

brafumSec, 2023-04-01:
show me your network configurations in detail and your code.

brada4, 2023-04-01:
I used netcat on server - a kvm in public network in regard to openwrt.
A debian nat client connect to that service

from kale box in same public  subnet i sent syn+ack using nmap based on netstat to public service
no rst was generated and nothing happened
I helped with RST using same kale -nothing happened.
Openwrt uses quite straitforward netfilter in Linux kernel. 
Just RST with way off seq numbers does not work. 

You really need to send your paper to netfilter.

brafumSec, 2023-04-01:
Just use the  correct OpenWrt versions as the  contributor mentioned, and confirm that the attacker is spoof-able on the Internet.

brada4, 2023-04-01:
OpenWRT uses mainline linux kernel with its firewalls, probably more prevalent than ubuntu ufw, though same code paths enacted.
Really check with common upstream at netfilter.org.

brada4, 2023-04-14:
There is a local sysctl that disables window checking (for me does not drop state, but does not mark OOB RST as INVALID)
net.netfilter.nf_conntrack_tcp_no_window_check=0 -> run the upstream code.
Could you check?