openwrt/openwrt #12300: DHCP Offer message is not passed toward end client via relayd bridge

AlfaBravoX, 2023-03-31:
### Describe the bug

In relayd brideg setup for some reason LAN (eth0.1) connected clients does not receive DHCP offer message coming from 192.168.4.1 DHCP server. 

I did 2 PCAP on 2 different places: 
1st at client (MAC OSX) NIC card
2nd next to DHCP server

It is clearly shown that DHCP server received DHCP Discovery from Client and sent DHCP Offer, but this message is not received by client which keeps sending DHCP Discovery requests.

This is the setup:

ISP <-> Cisco ISP with DHCP server (192.168.4.1) <-> TP-Link TL-WDR4300 v1 as 5G wifi client (192.168.4.8) <-> LAN (MAC OSx) client (ETHER a0:ce:c8:c5:9f:37)

Below is the WDR4300 config.

```
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fdfa:a5b6:9bbf::/48'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth0.1'

config device
	option name 'eth0.2'
	option macaddr '90:f6:52:f2:54:9f'

config switch
	option name 'switch0'
	option reset '1'
	option enable_vlan '1'

config switch_vlan
	option device 'switch0'
	option vlan '1'
	option ports '0t 2 3 4'
	option vid '1'

config switch_vlan
	option device 'switch0'
	option vlan '2'
	option ports '0t 5 1'
	option vid '2'

config interface 'wwan'
	option proto 'static'
	option netmask '255.255.255.0'
	option gateway '192.168.4.1'
	list dns '192.168.4.1'
	option ipaddr '192.168.4.8'

config interface 'MANAGEMENT_LAN'
	option device 'eth0.2'
	option proto 'static'
	option ipaddr '192.168.1.1'
	option netmask '255.255.255.0'

config interface 'LAN'
	option proto 'static'
	option device 'eth0.1'

config interface 'BRIDGERELAYD'
	option proto 'relay'
	list network 'LAN'
	list network 'wwan'
```

### OpenWrt version

r20028-43d71ad93e

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link TL-WDR4300 v1

### Image kind

Official downloaded image

### Steps to reproduce

Use my config and try on same chip router 

### Expected behaviour

LAN connected clients will receive `DHCP offer message` from 192.168.4.1 server 

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

AlfaBravoX, 2023-03-31:
[PCAP_dhcp_relayd_issue_offer_only_captured_at_DHCP_server_side.pcapng.zip](https://github.com/openwrt/openwrt/files/11120881/PCAP_dhcp_relayd_issue_offer_only_captured_at_DHCP_server_side.pcapng.zip)

AlfaBravoX, 2023-03-31:
[PCAP_dhcp_relayd_only_discover_seen_captured_at_client_NIC.pcapng.zip](https://github.com/openwrt/openwrt/files/11120892/PCAP_dhcp_relayd_only_discover_seen_captured_at_client_NIC.pcapng.zip)

brada4, 2023-03-31:
Does firewall permit anything on that interface?

AlfaBravoX, 2023-04-01:
There are no connectivity issues if clients are set with static IP. Firewall service is disabled. The only issue is client`s DHCP association due to above mentioned problem with broadcast `DHCP Offer` message

AlfaBravoX, 2023-04-03:
Issue is solved. the resolution is to add ip helper-address pointing to DHCP server at 192.168.4.1.

INET ISP DOCSIS <-> Cisco ISP with DHCP server (192.168.4.1) <-> Cisco Autonomous AP (BVI with static IP) <<-- add ip helper-address here <-> TP-Link TL-WDR4300 v1 as 5G wifi client (192.168.4.8) <-> LAN (MAC OSx) client (ETHER a0:ce:c8:c5:9f:37)

closed 2023-04-03 by AlfaBravoX