openwrt/openwrt #12301: [22.03] Archer C7 V2, WAN LED is missing.

DittelHome, 2023-03-31:
### Describe the bug

WLAN LED is missing:
root@Repeater_WZ:~# ls /sys/class/leds
ath10k-phy0   green:qss     green:usb1    green:wlan2g
ath9k-phy1    green:system  green:usb2    green:wlan5g

### OpenWrt version

 r20028-43d71ad93e

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer C7 v2

### Image kind

Official downloaded image

### Additional info

In older AR71xx  days we have had a WLAN LED.
Since ATH79 WLAN LED is missing.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-04-01:
There is v4/v5 conf in leds scrkpt. Seems ledtrig-switc kmod  at work

DittelHome, 2023-04-05:
> There is v4/v5 conf in leds scrkpt. Seems ledtrig-switc kmod at work

OK, what does it mean ? Do i need to install any package in addition ?
Can i contribute to a solution for that ?

DittelHome, 2023-04-06:
`after kmod v5 config in /etc/config/system is:`

Sorry, but i dont understand.
How can i install kmod v5 ?

I miss the leds mentioned above in my /etc/config/system..

brada4, 2023-04-06:
Mine is LED config from C7 v5 device generated by script in /etc/board.d .
If it does not work if added to /etc/config/system lets dig deeper to pre-ath79 config.

DittelHome, 2023-04-06:
OK, i have tried your (V5) LED config.
No chance, the WAN LED keeps off.

DittelHome, 2023-04-07:
For a test i have installed this modules:
kmod-leds-gpio
kmod-ledtrig-gpio
luci-app-ledtrig-switch

Unfortunately without success.

brada4, 2023-04-07:
I hoped its more similar hardware. 
I dont see anything for specific leds in ar71xx https://github.com/openwrt/openwrt/blob/openwrt-18.06/target/linux/ar71xx/image/generic-tp-link.mk that could be lost.
Do you have an option to downgrade that far to see how leds look vs nowadays?

DittelHome, 2023-04-07:
OK, i will downgrade to ar71xx and report  the output from" ls /sys/class/leds/", right ?

brada4, 2023-04-07:
Explore anything like /etc/config/* swconfig kmods /sys/module , luci picture, and try to summarize urself, or post all if you cant get the idea.

DittelHome, 2023-04-07:
It was a nightmare..
After trying downgrade to 19.07.10, my device was bricked, no tftp restore, no telnet, nothing. 
Downgrade to such a old version is not a good idea.
Ok, i flashed the device again via TTL adapter !!
Result: All LED's are present, including WAN-LED !!!

Here are the results:

ls /sys/class/leds

 ```
-----------------------------------------------------
 OpenWrt 19.07.10, r11427-9ce6aa9d8d
 -----------------------------------------------------
root@OpenWrt:~#  ls /sys/class/leds/
ath10k-phy0           tp-link:green:lan4    tp-link:green:wan
ath9k-phy1            tp-link:green:qss     tp-link:green:wlan2g
tp-link:green:lan1    tp-link:green:system  tp-link:green:wlan5g
tp-link:green:lan2    tp-link:green:usb1
tp-link:green:lan3    tp-link:green:usb2
root@OpenWrt:~#
```
/etc/config/system

```
config system
	option hostname 'OpenWrt'
	option timezone 'UTC'
	option ttylogin '0'
	option log_size '64'
	option urandom_seed '0'

config timeserver 'ntp'
	option enabled '1'
	option enable_server '0'
	list server '0.openwrt.pool.ntp.org'
	list server '1.openwrt.pool.ntp.org'
	list server '2.openwrt.pool.ntp.org'
	list server '3.openwrt.pool.ntp.org'

config led 'led_usb1'
	option name 'USB1'
	option sysfs 'tp-link:green:usb1'
	option trigger 'usbport'
	list port 'usb1-port1'

config led 'led_usb2'
	option name 'USB2'
	option sysfs 'tp-link:green:usb2'
	option trigger 'usbport'
	list port 'usb2-port1'

config led 'led_wlan2g'
	option name 'WLAN2G'
	option sysfs 'tp-link:green:wlan2g'
	option trigger 'phy1tpt'

config led 'led_wlan5g'
	option name 'WLAN5G'
	option sysfs 'tp-link:green:wlan5g'
	option trigger 'phy0tpt'
```

Bootlog:

```

U-Boot 1.1.4 (Dec 17 2013 - 16:37:27)

ap135 - Scorpion 1.0DRAM:  
[... 225 lines cut ...]
[   14.295553] ath10k_pci 0000:01:00.0: board_file api 1 bmi_id N/A crc32 bebc7c08
[   15.565744] ath10k_pci 0000:01:00.0: htt-ver 2.1 wmi-op 5 htt-op 2 cal file max-sta 128 raw 0 hwcrypto 1
[   15.781231] ieee80211 phy1: Atheros AR9550 Rev:0 mem=0xb8100000, irq=47
[   15.845973] kmodloader: done loading kernel modules from /etc/modules.d/*
[   26.595313] eth1: link up (1000Mbps/Full duplex)
[   26.600076] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
[   26.638381] br-lan: port 1(eth1.1) entered blocking state
[   26.643906] br-lan: port 1(eth1.1) entered disabled state
[   26.649633] device eth1.1 entered promiscuous mode
[   26.654533] device eth1 entered promiscuous mode
[   26.730544] br-lan: port 1(eth1.1) entered blocking state
[   26.736070] br-lan: port 1(eth1.1) entered forwarding state
[   26.741951] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
[   26.817960] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   26.856346] IPv6: ADDRCONF(NETDEV_UP): eth0.2: link is not ready
[   27.641312] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   27.648183] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   27.882730] eth0: link up (1000Mbps/Full duplex)
[   27.887560] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   27.968206] IPv6: ADDRCONF(NETDEV_CHANGE): eth0.2: link becomes ready
```

Screenshot of LED Config:

[image]

brada4, 2023-04-08:
Try to find `devalias` or `driver` where led symlinks point to.
some specialties with v5
I get default-on trigger saved if luci-app-ledtrig-switch is not installed.
Switching on green and amber blacks all leds till poweroff of device.

DittelHome, 2023-10-21:
With 23.05 still no WAN and LAN LED's ..