openwrt/openwrt #12306: Totolink X5000R bootlooping due to unsupport SPI-NOR in current kernel

ThranduilII, 2023-04-01:
### Describe the bug

Totolink stealth changed the spi-nor in their router. New spi-nor is ZB25VQ128ASIG with JEDEC id bytes: 5e 40 18 5e 40 18. 
I believe support has been added per this thread: https://lore.kernel.org/lkml/20210918072256.3505379-1-daniel@0x0f.com/T/ but has not been integrated into OpenWRT as of now.  

### OpenWrt version

23.03.3

### OpenWrt target/subtarget

mt762

### Device

Totolink X5000R V 1.0

### Image kind

Official downloaded image

### Steps to reproduce

Downgrade to OEM first release firmware.  Upgrade to OpenWRT using OEM webupgrade GUI.

### Actual behaviour

Bootloop after installing OpenWRT.

### Expected behaviour

OpenWRT to run without bootloop

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dfateyev, 2023-04-04:
@981213 I believe, you may be interested in this issue

ThranduilII, 2023-04-14:
I have requested pull requests for 5.10 and 5.15 patches that would allow the router to be flashed without any issues of bootlooping.

carlosfilhoadm, 2023-04-28:
Hi @ThranduilII

I bought a X5000R in August 2022 and intended to finally install OpenWRT on it in the next few days, but thankfully found this issue report first.

Will the next release (22.03.5) include your patches in order to fix this issue?

Thank you.

dfateyev, 2023-04-28:
I see https://github.com/openwrt/openwrt/pull/12475 was closed without merge

carlosfilhoadm, 2023-05-17:
Thank you @dfateyev

carlosfilhoadm, 2023-05-17:
I haven't seen any commits for this issue in git.openwrt.org. Is it still unsolved, then?

Does it also affect 21.02.7?

d687r02j8g, 2023-05-17:
Is there a way to recover from the bootloop?

@ThranduilII the below user tried using your 23.x image but doesn't seem to work. Is there anything special the user needs to do? They are stuck flashing openwrt

https://forum.openwrt.org/t/got-stuck-on-system-running-in-recovery-initramfs-mode-totolink-x5000r/160442?u=d687r02j8g

ThranduilII, 2023-05-23:
> Is there a way to recover from the bootloop?
>
> @ThranduilII the below user tried using your 23.x image but doesn't seem to work. Is there anything special the user needs to do? They are stuck [...]

That user is using an image that does not have the merge that would solve the boot loops issue. I never ended up merging my commits as I am a novice here on GitHub.

I have compiled an image for the current 22.03.5 on my own fork and I have been running it for the past 19 days. [image]

I'll head over to that thread and link my compiled image.

carlosfilhoadm, 2023-05-25:
I installed the latest release (22.03.5) on the X5000R.

OpenWRT is working normally, even though the device was purchased on August 2022.

I assumed that if the installation had failed, I would have used the emergency recovery page to revert it.

ThranduilII, 2023-05-25:
> I installed the latest release (22.03.5) on the X5000R.
> 
> OpenWRT is working normally, even though the device was purchased on August 2022.
> 
> I assumed that if the installation had failed, I would have used the emergency recovery page to revert it.

If the firmware doesn't support your device in any way, it will boot loop like what happened to my x5000r before I patched in the support for the Zbits SPI-NOR.  Glad to see it works for you.

jchiutwn, 2023-07-02:
Hi I'd like to report a potential alternative spi-nor on my newly purchased X5000R, it seems to come with BoHong bh25q128as.

OpenWrt release builds have bootloop. No issues with @ThranduilII Openwrt-22.03.5-X5000R-Zbit-support so far.

No sure if this is something to do with 405-mtd-spi-nor-Add-support-for-BoHong-bh25q128as.patch though.

[    0.641583] spi-mt7621 1e000b00.spi: sys_freq: 220000000
[    0.648635] spi-nor spi0.0: bh25q128as (16384 Kbytes)
[    0.653743] 4 fixed-partitions partitions found on MTD device spi0.0
[    0.660242] OF: Bad cell count for /palmbus@1e000000/spi@b00/flash@0/partitions
[    0.667541] OF: Bad cell count for /palmbus@1e000000/spi@b00/flash@0/partitions
[    0.675542] OF: Bad cell count for /palmbus@1e000000/spi@b00/flash@0/partitions
[    0.682926] OF: Bad cell count for /palmbus@1e000000/spi@b00/flash@0/partitions
[    0.690480] Creating 4 MTD partitions on "spi0.0":

lnxbil, 2023-08-30:
> I installed the latest release (22.03.5) on the X5000R.
> 
> OpenWRT is working normally, even though the device was purchased on August 2022.
> 
> I assumed that if the installation had failed, I would have used the emergency recovery page to revert it.

Works for me too. 22.03.5 added a fix for another device that may have solved the problem with the X5000R too.

Smankusors, 2024-04-09:
btw, the 22.03 security support will be ended in 11 April 2024...

If anyone wanted to upgrade to the 23.05 and wondering if it's safe to do so, you can determine the SPI flash chip manufacturer by following these steps:
1. login to your router via SSH
2. run command `dmesg | grep spi`
   you should see an output similar to the following:
   ```
   [    0.648548] spi-nor spi0.0: en25qh128 (16384 Kbytes)
   ```
3. in quick Google search, I found out that it's manufactured by Eon, and I verified that it exist on the kernel source code (v5.15)

I gave the upgrade a shot with the sysupgrade image and it worked like a charm! OpenWRT is up and running on the latest release build. 😄

Menelmacar, 2024-10-23:
Hi, I'm new to this all and got an X5000R and want to install openwrt the first time. I've read the device page with a big warning. The linked support journey of a user, and this issue.
By the last comments here, this issue seems to be fixed? Or not? I can't follow I must confess. (also don't know how to open the device and identify the chip 😅)

Smankusors, 2024-10-24:
no, it's not, I'm just explaining how to check if you have a problematic flash chip, and if it's safe to upgrade the OpenWRT, not the first install.

It seems there's no progress on the mainline:
https://lore.kernel.org/lkml/20210918072256.3505379-1-daniel@0x0f.com/T/

---

On OpenWRT, @ThranduilII opened and closed the PRs without any reason, so it's stuck in limbo:
- #12396
- #12404
- #12475

I don't have the flash chip in question, so I can't verify if the ThranduilII's patch is working or not.

BeJoNe, 2025-01-03:
Thoman is selling its "own branded" 19" Wifi Router https://www.thomann.de/intl/swissonic_professional_router_2_mkii.htm which is internally the TOTOLINK X5000R.
Are there any news about integrating the patch into the baseline image?
Thanks

sunsky131221, 2025-01-13:
> no, it's not, I'm just explaining how to check if you have a problematic flash chip, and if it's safe to upgrade the OpenWRT, not the first install.
> 
> It seems there's no progress on the mainline: https://lore.kernel.org/lkml/20210918072256.3505379-1-daniel@0x0f.com/T/
> 
> On OpenWRT, @ThranduilII opened and closed the PRs without any reason, so it's stuck in limbo:
> 
> * [ ramips: add linux 5.15 and 5.10 support for Zbits ZB25VQ128 SPI-NOR on Totolink X5000R #12396](https://github.com/openwrt/openwrt/pull/12396)
> * [ramips: add linux 5.15 and 5.10 support for Zbits ZB25VQ128 SPI-NOR on Totolink X5000R #12404](https://github.com/openwrt/openwrt/pull/12404)
> * [ramips: add linux 5.15 and 5.10 support for Zbits ZB25VQ128 SPI-NOR on Totolink X5000R #12475](https://github.com/openwrt/openwrt/pull/12475)
> 
> I don't have the flash chip in question, so I can't verify if the ThranduilII's patch is working or not.

Work for me with 23.05.x but 24.10 won't boot.
Need to upgrade this patch with kernel version 6.6.x.

Here the patch for 6.6.x

# vi target/linux/ramips/patches-6.6/412-mtd-spi-nor-add-support-for-zbit-zb25vq128.patch
`
--- a/drivers/mtd/spi-nor/core.c	2025-01-13 00:03:29.956254613 +0800
+++ b/drivers/mtd/spi-nor/core.c	2025-01-13 00:09:47.061829016 +0800
@@ -2019,6 +2019,7 @@
 	&spi_nor_xilinx,
 	&spi_nor_xmc,
 	&spi_nor_xtx,
+	&spi_nor_zbit,
 };
 
 static const struct flash_info spi_nor_generic_flash = {
--- a/drivers/mtd/spi-nor/core.h	2025-01-13 00:03:29.955254595 +0800
+++ b/drivers/mtd/spi-nor/core.h	2025-01-13 00:10:27.872540504 +0800
@@ -649,6 +649,7 @@
 extern const struct spi_nor_manufacturer spi_nor_xilinx;
 extern const struct spi_nor_manufacturer spi_nor_xmc;
 extern const struct spi_nor_manufacturer spi_nor_xtx;
+extern const struct spi_nor_manufacturer spi_nor_zbit;
 
 extern const struct attribute_group *spi_nor_sysfs_groups[];
 
--- a/drivers/mtd/spi-nor/Makefile	2025-01-13 00:03:29.953254560 +0800
+++ b/drivers/mtd/spi-nor/Makefile	2025-01-13 00:10:04.332130103 +0800
@@ -19,6 +19,7 @@
 spi-nor-objs			+= xilinx.o
 spi-nor-objs			+= xmc.o
 spi-nor-objs			+= xtx.o
+spi-nor-objs			+= zbit.o
 spi-nor-$(CONFIG_DEBUG_FS)	+= debugfs.o
 obj-$(CONFIG_MTD_SPI_NOR)	+= spi-nor.o
 
--- a/drivers/mtd/spi-nor/zbit.c	1970-01-01 08:00:00.000000000 +0800
+++ b/drivers/mtd/spi-nor/zbit.c	2025-01-13 00:03:36.512368910 +0800
@@ -0,0 +1,21 @@
+// SPDX-License-Identifier: GPL-2.0
+/*
+ * Copyright (C) 2021, Daniel Palmer<daniel@thingy.jp>
+ */
+
+#include <linux/mtd/spi-nor.h>
+
+#include "core.h"
+
+static const struct flash_info zbit_parts[] = {
+	/* zbit */
+	{ "zb25vq128", INFO(0x5e4018, 0, 64 * 1024, 256)
+		       FLAGS(SPI_NOR_HAS_LOCK | SPI_NOR_HAS_TB)
+		       NO_SFDP_FLAGS(SECT_4K | SPI_NOR_DUAL_READ | SPI_NOR_QUAD_READ)},
+};
+
+const struct spi_nor_manufacturer spi_nor_zbit = {
+	.name = "zbit",
+	.parts = zbit_parts,
+	.nparts = ARRAY_SIZE(zbit_parts),
+};
`

sunsky131221, 2025-01-13:
> no, it's not, I'm just explaining how to check if you have a problematic flash chip, and if it's safe to upgrade the OpenWRT, not the first install.
>
> It seems there's no progress on the mainline: [...]

Work for me with 23.05.x but 24.10 won't boot.
Need to upgrade this patch with kernel version 6.6.x.

Here the patch for 6.6.x, drop it into <openwrt>/target/linux/ramips/patches-6.6
[412-mtd-spi-nor-add-support-for-zbit-zb25vq128.patch](https://github.com/user-attachments/files/18398214/412-mtd-spi-nor-add-support-for-zbit-zb25vq128.patch)

BeJoNe, 2025-01-15:
there is also a repository providing builds with the patch..
https://github.com/akm-04/Openwrt-23.05_Totolink-x5000r_Zbit-Support
see
https://forum.openwrt.org/t/totolink-x5000r-any-way-to-determine-the-flash-chip-used-on-software-level/178494/4?u=bejone

Smankusors, 2025-01-15:
if you guys have a working patch, I recommend you to open a new PR on this repo. So that you can get latest update without waiting someone build it for you...

sunsky131221, 2025-11-14:
> > if you guys have a working patch, I recommend you to open a new PR on this repo. So that you can get latest update without waiting someone build it for you...
> 
> I would create a PR, but I don't have this device any more — unfortunately cannot test the changes before submitting. Maybe, someone else with the device will do.

I have two X5000R units. One can run the official OpenWRT firmware normally, but the other encounters a this issue. I just submitted a PR, but I'm not sure if it's in the correct format.

Here the PR link: https://github.com/openwrt/openwrt/pull/20785

sunsky131221, 2025-11-16:
My X5000R with ZB25VQ128 install official released 24.10.4 firmware can boot normally.

Here the dmesg log below:
> [    1.542500] spi-mt7621 1e000b00.spi: sys_freq: 220000000
> [    1.549432] spi-nor spi0.0: spi-nor-generic (16384 Kbytes)
> [    1.555055] 4 fixed-partitions partitions found on MTD device spi0.0
> [    1.561498] OF: Bad cell count for /palmbus@1e000000/spi@b00/flash@0/partitions
> [    1.568863] OF: Bad cell count for /palmbus@1e000000/spi@b00/flash@0/partitions
> [    1.576765] Creating 4 MTD partitions on "spi0.0":
> [    1.581584] 0x000000000000-0x000000030000 : "u-boot"
> [    1.588701] 0x000000030000-0x000000040000 : "u-boot-env"
> [    1.595335] 0x000000040000-0x000000050000 : "factory"
> [    1.601829] OF: Bad cell count for /palmbus@1e000000/spi@b00/flash@0/partitions
> [    1.609556] 0x000000050000-0x000001000000 : "firmware"
> [    1.615992] 2 uimage-fw partitions found on MTD device firmware
> [    1.621959] Creating 2 MTD partitions on "firmware":
> [    1.626932] 0x000000000000-0x000000308b61 : "kernel"
> [    1.631883] mtd: partition "kernel" doesn't end on an erase/write block -- force read-only
> [    1.641782] 0x000000308b61-0x000000fb0000 : "rootfs"
> [    1.646800] mtd: partition "rootfs" doesn't start on an erase/write block boundary -- force read-only
> [    1.657121] mtd: setting mtd5 (rootfs) as root device
> [    1.662282] 1 squashfs-split partitions found on MTD device rootfs
> [    1.668489] 0x000000740000-0x000000fb0000 : "rootfs_data"

Seems be identified as spi-nor-generic.