openwrt/openwrt #12308: `/etc/config/fstab` only mounts the last of two btrfs subvolume entries on  the same partition

TheRealGramdalf, 2023-04-01:
### Describe the bug

When attempting to bind mount multiple btrfs subvolumes to an alternate location via `/etc/config/fstab`, only the last in the anonymous section is actually mounted.

### What I expected:
```
.
├── etc
│   └── klipper << Root btrfs volume mounted to '/etc/klipper'
│       ├── gcc << btrfs subvolume '/gcc' in its usual location
│       └── python3.10 << btrfs subvolume '/python3.10' in its usual location
└── usr
    └── lib
        ├── gcc << btrfs subvolume '/gcc' bind mounted to '/usr/lib/gcc'
        └── python3.10 << btrfs subvolume '/python3.10' bind mounted to '/usr/lib/python3.10'
```

### What acutally happened
```
.
├── etc
└── usr
    └── lib
        └── python3.10 << btrfs subvolume '/python3.10' mounted to '/usr/lib/python3.10'
```

More detailed explanation below (this is just a visual summary)

### OpenWrt version

r20028-43d71ad93e

### OpenWrt target/subtarget

ipq806x/generic

### Device

Linksys EA8500 WiFi Router

### Image kind

Self-built image

### Steps to reproduce

1. Create an image with the below packages (or install the necessary packages on a live system). 
    - Exact packages used can be found below, but the explicitly installed ones I believe to be relevant are:
    - ` btrfs-progs block-mount e2fsprogs parted kmod-usb-storage kmod-usb-storage-uas kmod-usb-ohci kmod-usb-uhci kmod-usb3`
2. Partition a 16gb USB drive with one (BTRFS) partition on a separate device (In this case: Laptop, Fedora 37, gparted: Device > Create Partition table > GPT; Partition > New > File System: BTRFS)
3. Plug USB drive into routers USB3 port
4. Enter CLI (In this case, SSH)
5. Execute commands below:
```bash
#!/bin/sh
# Bind mounts
DEVICE='sda1'
MOUNTPOINT='/etc/klipper'
# Name of the directory to mount under `/usr/lib/`. This may change between python versions, check the wiki if "Not enough space" errors are thrown
PYTHONNAME='python3.10'
GCCNAME='gcc'
KLIPPERNAME='klipper'
# Mountpoint paths
NGINX='/etc/nginx'
PYTHON="/usr/lib/$PYTHONNAME"
GCC="/usr/lib/$GCCNAME"
# Evaluate `block info $DEVICE` to obtain the uuid as a variable
eval "$(block info ${DEVICE} | grep -o -e "UUID=\S*")"

# Create parent directory
[... 497 lines cut ...]
uci del wireless.radio0.disabled='0'

uci commit wireless

wifi reload

# Delete the default listen rules for 
# BROKEN? uci del uhttpd.main.listen_http
# BROKEN? uci del uhttpd.main.listen_https
# BROKEN? uci add_list uhttpd.main.listen_http="$LAN_IP""$LISTEN_HTTP"
# BROKEN? uci add_list uhttpd.main.listen_https="$LAN_IP""$LISTEN_HTTPS"
uci set uhttpd.main.redirect_https='on'

uci commit uhttpd

# Make a backup of /etc/shadow to /etc/shadow- as per the busybox passwd convention
cp /etc/shadow /etc/shadow-
# Change root password. Note the escaped \$ character in the regex.
# The $pass variable must be defined as literal, as it contains special characters. Either escape said characters or use single quotes.
sed -i -e "s;^root:[\$A-z0-9]*:;root:$PASS:;" /etc/shadow
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

boretom, 2023-04-04:
I bave part of it a go in a VM with OpenWrt installed (for earlier testing of RAID).

But in point 5 the script is the variable NGINXNAME is not set and therefore won't succeed.

M95D, 2023-05-13:
I think it's a duplicate of this issue: https://github.com/openwrt/openwrt/issues/9576

ryjelsum, 2025-12-19:
> I think it's a duplicate of this issue: [#9576](https://github.com/openwrt/openwrt/issues/9576)

To me it seems like this is kind of a different (but related) behavior that also causes multiple btrfs subvolumes on the same partition to not really work with Openwrt's filesystem mounting facilities in general - so this might actually not be a duplicate per se, just related. Your issue describes `block mount` bailing out if it detects that a block device is already mounted, despite that assumption not working with btrfs. Here, as far as I can tell the mounting proceeds fine for both configuration entries, it just ends with only the final subvolume specified in `/etc/config/fstab` mounted.

Though I can say, the same thing triggers both issues either way. I have an /etc/config/fstab configuration as such: 

```
config mount
	option target '/opt/'
	option uuid '0bb9eed1-b9e8-479e-815c-05183af22148'
	option enabled '1'
	option options 'subvol=@opt,compress-force=zstd:1'
	option fstype 'btrfs'

config mount
	option enabled '1'
	option uuid '0bb9eed1-b9e8-479e-815c-05183af22148'
	option target '/srv/'
	option fstype 'btrfs'
	option options 'subvol=@srv,compress-force=zstd:1'
```

The end result after boot is that only the `@srv` subvolume is mounted on /srv, this issue. If I then run `block mount`, I get a `block: /dev/sda1 is already mounted on /srv` error, your issue. I would think if both issues were due to `block mount`'s behavior, that the end result here after the boot would be `@opt` mounted on /opt.