openwrt/openwrt #12318: uclient-fetch does not send Content-Length

axet, 2023-04-03:
### Describe the bug

using uclient-fetch with --post-data="string" causing server to crash due to Content-Length header is missing.

### OpenWrt version

22.03.3

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link TL-WR842N/ND v2

### Image kind

Self-built image

### Steps to reproduce

root@OpenWrt:~# wget -O- --post-data="kkk" http://10.10.5.4/apply.cgi?formPasswordAuth
Downloading 'http://10.10.5.4/apply.cgi?formPasswordAuth'
Connecting to 10.10.5.4:80
(null)                   0   - stalled -
Connection reset prematurely

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-04-03:
It sends zero-length chunk, meaning message ends there and no size needed to be specifid.

axet, 2023-04-03:
@brada4

> Applications SHOULD use this field to indicate the transfer-length of the message-body, unless this is prohibited by the rules in [section-4.4](https://www.rfc-editor.org/rfc/rfc2616#section-4.4)

Message body is a post data of course.

closed 2023-04-03 by axet