openwrt/openwrt #12321: Cannot negotiate 2500baseT/Full on Aquantia AQR113C

axelgenus, 2023-04-04:
### Describe the bug

I installed OpenWRT on my QNAP QHora 301W and I noticed an issue with the 10G ports: they do not advertise 2500baseT/Full or 5000baseT/Full speeds, even if they are listed among the supported link modes:

```
Settings for 10g-1:
	Supported ports: [ ]
	Supported link modes:   100baseT/Half 100baseT/Full
	                        1000baseT/Full
	                        10000baseT/Full
	                        1000baseKX/Full
	                        10000baseKX4/Full
	                        10000baseKR/Full
	                        2500baseT/Full
	                        5000baseT/Full
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  100baseT/Half 100baseT/Full
	                        1000baseT/Full
	                        10000baseT/Full
	                        1000baseKX/Full
	                        10000baseKX4/Full
	                        10000baseKR/Full
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full
	                                     100baseT/Half 100baseT/Full
	                                     1000baseT/Full
	                                     2500baseT/Full
	Link partner advertised pause frame use: Symmetric Receive-only
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 1000Mb/s
	Duplex: Full
	Port: Twisted Pair
	PHYAD: 8
	Transceiver: external
	Auto-negotiation: on
	MDI-X: Unknown
	Link detected: yes
```

Following the suggestion on the forum [1], I run the following commands to force advertising the correct/supported speeds:

```
ethtool -s 10g-1 advertise 18000000E102C
ethtool -s 10g-2 advertise 18000000E102C
```

[1] https://forum.openwrt.org/t/adding-openwrt-support-for-qnap-qhora-301w/96934

Unfortunately I has mixed success following this approach: when I connect one of the two 10G ports to my ISP's ONT (ZTE F6005v6) the router is unable to negotiate 2500baseT/Full speed, it reverts automatically to 1000baseT/Full and reset the advertised link modes. When I connect to the 2.5Gbps port on my Fritzbox 5530 instead it does switch automatically to 2500baseT/Full and works fine.

The stock firmware seems to negotiate the maximum common speed correctly.

I tried playing around with ethtool to force the link mode and turn off auto-negotiation to no avail.

### OpenWrt version

r22496-d98c8fc06d

### OpenWrt target/subtarget

ipq807x/generic

### Device

QNAP 301w

### Image kind

Official downloaded image

### Steps to reproduce

1. Boot OpenWRT up;
2. Connect by means of SSH to the router;
3. Check the negotiated speed on the 10G interfaces with `ethtool 10g-1` and `ethtool 10g-2` (it shows 1Gbps full duplex for both ports);
4. Run `ethtool -s 10g-1 advertise 18000000E102C` and `ethtool -s 10g-2 advertise 18000000E102C`.

### Actual behaviour

The port connected to the ONT negotiate a 1Gbps/Full link whilst the one connected to the Fritzbox negotiate 2.5Gbps/Full.

### Expected behaviour

Both ports should correctly negotiate 2.5Gbps/Full link speed on boot and after forcing the advertised link modes by means of ethtool.

### Additional info

dmesg: [messages.log](https://github.com/openwrt/openwrt/files/11147607/messages.log)

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

1715173329, 2023-04-04:
See #11812

If you can compile firmware by yourself, try to put the following patch into `package/network/config/netifd/patches/100-Revert-device-add-support-for-configuring-device-link-spe.patch`
<details>

```patch
--- a/device.c
+++ b/device.c
@@ -61,8 +61,6 @@ static const struct blobmsg_policy dev_a
 	[DEV_ATTR_DROP_UNSOLICITED_NA] = { .name = "drop_unsolicited_na", .type = BLOBMSG_TYPE_BOOL },
 	[DEV_ATTR_ARP_ACCEPT] = { .name = "arp_accept", .type = BLOBMSG_TYPE_BOOL },
 	[DEV_ATTR_AUTH] = { .name = "auth", .type = BLOBMSG_TYPE_BOOL },
-	[DEV_ATTR_SPEED] = { .name = "speed", .type = BLOBMSG_TYPE_INT32 },
-	[DEV_ATTR_DUPLEX] = { .name = "duplex", .type = BLOBMSG_TYPE_BOOL },
 };
 
 const struct uci_blob_param_list device_attr_list = {
@@ -278,8 +276,6 @@ device_merge_settings(struct device *dev
 	n->arp_accept = s->flags & DEV_OPT_ARP_ACCEPT ?
 		s->arp_accept : os->arp_accept;
 	n->auth = s->flags & DEV_OPT_AUTH ? s->auth : os->auth;
-	n->speed = s->flags & DEV_OPT_SPEED ? s->speed : os->speed;
-	n->duplex = s->flags & DEV_OPT_DUPLEX ? s->duplex : os->duplex;
 	n->flags = s->flags | os->flags | os->valid_flags;
 }
 
@@ -454,16 +450,6 @@ device_init_settings(struct device *dev,
 		s->flags |= DEV_OPT_AUTH;
 	}
 
-	if ((cur = tb[DEV_ATTR_SPEED])) {
-		s->speed = blobmsg_get_u32(cur);
-		s->flags |= DEV_OPT_SPEED;
-	}
-
-	if ((cur = tb[DEV_ATTR_DUPLEX])) {
-		s->duplex = blobmsg_get_bool(cur);
-		s->flags |= DEV_OPT_DUPLEX;
-	}
-
 	device_set_disabled(dev, disabled);
 }
 
--- a/device.h
+++ b/device.h
@@ -60,8 +60,6 @@ enum {
 	DEV_ATTR_DROP_UNSOLICITED_NA,
 	DEV_ATTR_ARP_ACCEPT,
 	DEV_ATTR_AUTH,
-	DEV_ATTR_SPEED,
-	DEV_ATTR_DUPLEX,
 	__DEV_ATTR_MAX,
 };
 
@@ -124,8 +122,6 @@ enum {
 	DEV_OPT_DROP_GRATUITOUS_ARP	= (1ULL << 27),
 	DEV_OPT_DROP_UNSOLICITED_NA	= (1ULL << 28),
 	DEV_OPT_ARP_ACCEPT		= (1ULL << 29),
-	DEV_OPT_SPEED			= (1ULL << 30),
-	DEV_OPT_DUPLEX			= (1ULL << 31),
 };
 
 /* events broadcasted to all users of a device */
@@ -201,8 +197,6 @@ struct device_settings {
 	bool drop_unsolicited_na;
 	bool arp_accept;
 	bool auth;
-	unsigned int speed;
-	bool duplex;
 };
 
 /*
--- a/system-linux.c
+++ b/system-linux.c
@@ -1691,57 +1691,6 @@ int system_vlandev_del(struct device *vl
 	return system_link_del(vlandev->ifname);
 }
 
-static void
-system_set_ethtool_settings(struct device *dev, struct device_settings *s)
-{
-	struct ethtool_cmd ecmd = {
-		.cmd = ETHTOOL_GSET,
-	};
-	struct ifreq ifr = {
-		.ifr_data = (caddr_t)&ecmd,
-	};
-	static const struct {
-		unsigned int speed;
-		uint8_t bit_half;
-		uint8_t bit_full;
-	} speed_mask[] = {
-		{ 10, ETHTOOL_LINK_MODE_10baseT_Half_BIT, ETHTOOL_LINK_MODE_10baseT_Full_BIT },
-		{ 100, ETHTOOL_LINK_MODE_100baseT_Half_BIT, ETHTOOL_LINK_MODE_100baseT_Full_BIT },
-		{ 1000, ETHTOOL_LINK_MODE_1000baseT_Half_BIT, ETHTOOL_LINK_MODE_1000baseT_Full_BIT },
-	};
-	uint32_t adv;
-	size_t i;
-
-	strncpy(ifr.ifr_name, dev->ifname, sizeof(ifr.ifr_name) - 1);
-
-	if (ioctl(sock_ioctl, SIOCETHTOOL, &ifr) != 0)
-		return;
-
-	adv = ecmd.supported;
-	for (i = 0; i < ARRAY_SIZE(speed_mask); i++) {
-		if (s->flags & DEV_OPT_DUPLEX) {
-			int bit = s->duplex ? speed_mask[i].bit_half : speed_mask[i].bit_full;
-			adv &= ~(1 << bit);
-		}
-
-		if (!(s->flags & DEV_OPT_SPEED) ||
-		    s->speed == speed_mask[i].speed)
-			continue;
-
-		adv &= ~(1 << speed_mask[i].bit_full);
-		adv &= ~(1 << speed_mask[i].bit_half);
-	}
-
-
-	if (ecmd.autoneg && ecmd.advertising == adv)
-		return;
-
-	ecmd.autoneg = 1;
-	ecmd.advertising = adv;
-	ecmd.cmd = ETHTOOL_SSET;
-	ioctl(sock_ioctl, SIOCETHTOOL, &ifr);
-}
-
 void
 system_if_get_settings(struct device *dev, struct device_settings *s)
 {
@@ -1965,7 +1914,6 @@ system_if_apply_settings(struct device *
 		system_set_drop_unsolicited_na(dev, s->drop_unsolicited_na ? "1" : "0");
 	if (apply_mask & DEV_OPT_ARP_ACCEPT)
 		system_set_arp_accept(dev, s->arp_accept ? "1" : "0");
-	system_set_ethtool_settings(dev, s);
 }
 
 int system_if_up(struct device *dev)
```

</details>

axelgenus, 2023-04-05:
With your patch, after boot the advertised link modes are indeed correct and the selected speed is the right one:

```
Settings for 10g-1:
	Supported ports: [  ]
	Supported link modes:   100baseT/Half 100baseT/Full
	                        1000baseT/Full
	                        10000baseT/Full
	                        1000baseKX/Full
	                        10000baseKX4/Full
	                        10000baseKR/Full
	                        2500baseT/Full
	                        5000baseT/Full
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  100baseT/Full
	                        1000baseT/Full
	                        10000baseT/Full
	                        1000baseKX/Full
	                        10000baseKX4/Full
	                        10000baseKR/Full
	                        2500baseT/Full
	                        5000baseT/Full
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full
	                                     100baseT/Half 100baseT/Full
	                                     1000baseT/Full
	                                     2500baseT/Full
	Link partner advertised pause frame use: Symmetric Receive-only
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 2500Mb/s
	Duplex: Full
	Auto-negotiation: on
	Port: Twisted Pair
	PHYAD: 8
	Transceiver: external
	MDI-X: Unknown
	Link detected: yes
```

Thank you!

a011528, 2023-04-05:
@axelgenus now that you are getting 2.5G is your throughput improving on the 301W   ?

sjkhsl, 2023-04-09:
You try the patch mentioned in # 11812

axelgenus, 2023-04-10:
> You try the patch mentioned in #11812

That patch made the router completely unusable. The 10G interface kept restarting.

crwnet, 2023-04-10:
> > You try the patch mentioned in #11812
> 
> That patch made the router completely unusable. The 10G interface kept restarting.

That patch only baseT mode only, you need add 1000baseKX/Full, 10000baseKX4/Full, 10000baseKR/Full to fixed your interface.

axelgenus, 2023-04-10:
> That patch fixed baseT mode only, you need add 1000baseKX/Full, 10000baseKX4/Full, 10000baseKR/Full to fixed your interface.

The two 10G interfaces are connected to 2500baseT/Full devices whilst the 1G ones are connected to 1000baseT/Full and 100baseT/Full devices.

soxrok2212, 2023-05-26:
Patch works for me as well 👍

mardock2009, 2023-08-22:
pernament fix after restart add to rc.local on startup

#################################################
# Put your custom commands here that should be executed once
# the system init finished. By default this file does nothing.
#################################################

ethtool -s 10g-1 advertise 18000000E102C
ethtool -s 10g-2 advertise 18000000E102C

exit 0

soxrok2212, 2023-08-23:
> pernament fix after restart add to rc.local on startup
>
> #################################################
>
> # Put your custom commands here that should be executed once
> # the system init finished. By [...]

This "fix" does not work. It _does_ enable 2.5G advertising, but the link rates are far worse than 1G. The patch proposed in https://github.com/openwrt/openwrt/issues/12321#issuecomment-1496429799 works beautifully.

mardock2009, 2023-08-24:
working mate
after restart my log 

root@QNAP301W:~# ethtool 10g-1
Settings for 10g-1:
        Supported ports: [ ]
        Supported link modes:   100baseT/Half 100baseT/Full
                                1000baseT/Full
                                10000baseT/Full
                                1000baseKX/Full
                                10000baseKX4/Full
                                10000baseKR/Full
                                2500baseT/Full
                                5000baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  100baseT/Half 100baseT/Full
                                1000baseT/Full
                                10000baseT/Full
                                1000baseKX/Full
                                10000baseKX4/Full
                                10000baseKR/Full
                                2500baseT/Full
                                5000baseT/Full
        Advertised pause frame use: No
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  10baseT/Half 10baseT/Full
                                             100baseT/Half 100baseT/Full
                                             1000baseT/Full
                                             2500baseT/Full
        Link partner advertised pause frame use: Symmetric Receive-only
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 2500Mb/s
        Duplex: Full
        Port: Twisted Pair
        PHYAD: 8
        Transceiver: external
        Auto-negotiation: on
        MDI-X: Unknown
        Link detected: yes

robimarko, 2023-09-02:
This was fixed by:
[b85646f32f0cc613e7d6f4b0b75c12cd0d1ca91b](https://github.com/openwrt/openwrt/commit/b85646f32f0cc613e7d6f4b0b75c12cd0d1ca91b)

lynxis, 2023-09-13:
@axelgenus can we close this issue?

closed 2023-09-13 by axelgenus