openwrt/openwrt #12322: TX power limited to 15dBm for PL or GB  country setting for tp-link wr1043nd v4

namiltd, 2023-04-04:
### Describe the bug

TX power limited to 15dBm for PL or GB country setting for TP-Link wr1043nd v4 instead of 20dbm

### OpenWrt version

r20028-43d71ad93e

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link TL-WR1043ND v4

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2023-04-05:
But it doesn't happen when I set e.g. TW. I can set 20dBm and it shows that it is 20dBm

brada4, 2023-04-06:
ETSI standard mentions 20 dBm power emission from fixed or typical antennas, regulatory domain is applied correctly, though visualisation of that is awkward. Should be told somewhere milliwat+antenna=20dBm.

namiltd, 2023-04-06:
The problem is that according to what you write 15dbm + antenna = 20dBm, but the observed power is much lower than on other routers, e.g. WR1043ND v1 with 20dBm. This indicates that the value is displayed together with the antenna gain.

brada4, 2023-04-06:
Once you set antenna gain to 5dBm via `iw` for v1 router you will get the real legally permissible (read - weaker) signal out of it.
At some point "typical detachable antennas" were added to ETSI norm (GB & PL) and likely tplink added them either via new hardware or software version.
Older version may or may not have packaged 5dBm attachable antennas considered (hardware v1 does not have them recorded)
Also remember that all regulatory fuses add together to make signal weaker, never stronger.

namiltd, 2023-04-07:
This means that for WR1043ND v1 the driver incorrectly collects the antenna gain and for v4 it is OK?

brada4, 2023-04-07:
Anything with antenna transmitting at 100mW is overly strong in the air. I suppose it cannot collect antenna gain, i doubt it can even detect 50ohm antenna presence as such.

closed 2023-04-13 by namiltd