openwrt/openwrt #12323: luks encryption

HTMofDreams, 2023-04-05:
### Describe the bug

Some time ago, I was trying openWRT 22.x and I was wondering that I cant use luks encryption. The solution was, to use openWRT 21.x ... There works luks encryption fine.
Now, I testing openWRT-snapshot and luks encryption is still broken. Formating with cryptsetup works fine and then formating the encrypted partition with ext4, works fine, too. No, problems with mounting it.
But when I copy/move/add/delete some files I get a lots of errors from dmesg. It does not matter, which filesystem I use. Every time I get errors. Do I use an ext4 partition without encryption, there are no errros, too.
Could it be a kernel-mod problem?

### OpenWrt version

r16554-1d4dea6d4f

### OpenWrt target/subtarget

lantiq/xrx200

### Device

AVM FRITZ!Box 7362 SL

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

neheb, 2023-04-05:
probably. I changed cryptsetup to use the kmods at some point. I haven't tested it other than cryptsetup --benchmark.

HTMofDreams, 2023-04-05:
it looks like, when luks was formated with '--type luks1', then I become filesystem errors. some of my old external HDDs are luks1 formated. convertig from luks1 to luks2 works, but the same filesystem errors when I copy/add/delete/move files. Like this:

root@OpenWrt:~# dmesg 
[  227.261300] EXT4-fs error (device dm-0): ext4_lookup:1751: inode #9962513: comm mc: iget: bad extra_isize 45481 (inode size 256)
[  227.311125] EXT4-fs error (device dm-0): ext4_lookup:1751: inode #9962514: comm mc: iget: bad extra_isize 47547 (inode size 256)
[  227.684214] EXT4-fs error (device dm-0): ext4_lookup:1751: inode #9961684: comm mc: iget: bad extra_isize 39773 (inode size 256)
[  227.712559] EXT4-fs error (device dm-0): ext4_lookup:1751: inode #9961683: comm mc: iget: bad extra_isize 54870 (inode size 256)
[  260.080776] EXT4-fs error (device dm-0): ext4_validate_block_bitmap:390: comm kworker/u4:4: bg 16: bad block bitmap checksum
[  260.135458] EXT4-fs (dm-0): Delayed block allocation failed for inode 26214405 at logical offset 24576 with max blocks 32 with error 77
[  260.146423] EXT4-fs (dm-0): This should not happen!! Data will be lost
[  260.146423] 
[  260.776914] EXT4-fs error (device dm-0): ext4_validate_block_bitmap:390: comm kworker/u4:1: bg 16: bad block bitmap checksum
[  260.813213] EXT4-fs (dm-0): Delayed block allocation failed for inode 26214405 at logical offset 24608 with max blocks 1024 with error 77
[  260.824237] EXT4-fs (dm-0): This should not happen!! Data will be lost
[  260.824237] 
[  261.358948] EXT4-fs error (device dm-0): ext4_validate_block_bitmap:390: comm kworker/u4:1: bg 16: bad block bitmap checksum
[  261.391005] EXT4-fs (dm-0): Delayed block allocation failed for inode 26214405 at logical offset 25632 with max blocks 1024 with error 77
[  261.402115] EXT4-fs (dm-0): This should not happen!! Data will be lost
[  261.402115] 
[  262.172780] EXT4-fs error (device dm-0): ext4_validate_block_bitmap:390: comm kworker/u4:4: bg 16: bad block bitmap checksum
[  262.191114] EXT4-fs error (device dm-0): ext4_validate_block_bitmap:390: comm kworker/u4:4: bg 16: bad block bitmap checksum

HTMofDreams, 2023-04-06:
With openWRT 19.x, luks works fine, too.