openwrt/openwrt #12337: BananaPi R3 - kernel crash in sta_set_sinfo

uzytkownik, 2023-04-06:
### Describe the bug

System frequently crashes 

### OpenWrt version

r22480-18d516a649

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R3

### Image kind

Official downloaded image

### Steps to reproduce

Unknown. It happens about once per day

### Actual behaviour

System crash and reboot into rescue mode

### Expected behaviour

System doesn't crash

### Additional info

I use auc to update

```
<1>[121583.307174] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000008
<1>[121583.316039] Mem abort info:
<1>[121583.318916]   ESR = 0x0000000096000005
<1>[121583.322734]   EC = 0x25: DABT (current EL), IL = 32 bits
<1>[121583.328122]   SET = 0, FnV = 0
<1>[121583.331252]   EA = 0, S1PTW = 0
<1>[121583.334463]   FSC = 0x05: level 1 translation fault
<1>[121583.339415] Data abort info:
<1>[121583.342366]   ISV = 0, ISS = 0x00000005
<1>[121583.346270]   CM = 0, WnR = 0
<1>[121583.349313] user pgtable: 4k pages, 39-bit VAs, pgdp=0000000044fb0000
<1>[121583.355819] [0000000000000008] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
<0>[121583.364590] Internal error: Oops: 96000005 [#1] SMP
<7>[121583.369535] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 cfg80211 slhc sfp nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 mdio_i2c libcrc32c crc_ccitt compat crypto_safexcel pwm_fan hwmon i2c_gpio i2c_algo_bit sha1_generic seqiv md5 des_generic libdes authencesn authenc leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug usbcore usb_common
<7>[121583.438630] CPU: 3 PID: 1592 Comm: hostapd Not tainted 5.15.105 #0
<7>[121583.444878] Hardware name: Bananapi BPI-R3 (DT)
<7>[121583.449476] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
<7>[121583.456503] pc : sta_set_sinfo+0xa18/0xc30 [mac80211]
<7>[121583.461660] lr : sta_set_sinfo+0x5b8/0xc30 [mac80211]
<7>[121583.466796] sp : ffffffc00b15b7a0
<7>[121583.470181] x29: ffffffc00b15b7a0 x28: 0000000000000001 x27: ffffffc00b15bdd0
<7>[121583.477384] x26: ffffff800007e880 x25: ffffff8004a34900 x24: ffffff8003dcc735
<7>[121583.484586] x23: 00000000009b95c1 x22: ffffff8004a34900 x21: 0000058d1783bf82
<7>[121583.491788] x20: ffffff8003dcc000 x19: ffffff800471ef00 x18: 0000000000000000
<7>[121583.498991] x17: 0000000000001e20 x16: ffffffc008f1d000 x15: 0000000000000f10
<7>[121583.506193] x14: ffffff8003dccdf8 x13: ffffff8003dccdf8 x12: 0000000000000000
<7>[121583.513395] x11: 0000000000000000 x10: ffffff8003dcce00 x9 : 0000000000000000
<7>[121583.520597] x8 : ffffff800471f000 x7 : 0000000000002010 x6 : ffffff8003dd03a0
<7>[121583.527799] x5 : ffffff8003dd0880 x4 : 0000000000000000 x3 : ffffff800471ef44
<7>[121583.535001] x2 : 0000000000000000 x1 : 0000000000000001 x0 : 0000000000000000
<7>[121583.542203] Call trace:
<7>[121583.544722]  sta_set_sinfo+0xa18/0xc30 [mac80211]
<7>[121583.549513]  sta_set_sinfo+0xb48/0xc30 [mac80211]
<7>[121583.554302]  sta_info_destroy_addr_bss+0x4c/0x70 [mac80211]
<7>[121583.559958]  ieee80211_color_change_finish+0x1bf8/0x1e80 [mac80211]
<7>[121583.566306]  cfg80211_check_station_change+0x1384/0x4720 [cfg80211]
<7>[121583.572652]  genl_family_rcv_msg_doit+0xb4/0x110
<7>[121583.577343]  genl_rcv_msg+0xd0/0x1c0
<7>[121583.580989]  netlink_rcv_skb+0x58/0x120
<7>[121583.584895]  genl_rcv+0x34/0x50
<7>[121583.588108]  netlink_unicast+0x1f0/0x2ec
<7>[121583.592100]  netlink_sendmsg+0x19c/0x3d0
<7>[121583.596094]  ____sys_sendmsg+0x21c/0x260
<7>[121583.600088]  ___sys_sendmsg+0x80/0xf0
<7>[121583.603821]  __sys_sendmsg+0x44/0xa0
<7>[121583.607469]  __arm64_sys_sendmsg+0x20/0x30
<7>[121583.611635]  invoke_syscall.constprop.0+0x4c/0xe0
<7>[121583.616410]  do_el0_svc+0x40/0xd0
<7>[121583.619796]  el0_svc+0x14/0x50
<7>[121583.622925]  el0t_64_sync_handler+0xe0/0x110
<7>[121583.627266]  el0t_64_sync+0x158/0x15c
<0>[121583.631002] Code: d3441c42 12000c00 8b020cc2 f9409c42 (f9400446) 
<4>[121583.637162] ---[ end trace 9e8d6e1f5ab75c87 ]---
```

### Diffconfig

```text
N/A
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

im-0, 2023-04-24:
I do not see this crash on older version - r22180-f57a0709e8.

Adminelix, 2023-05-02:
I saw this crash recently on `r22652-9a26669510`
```
Sat Apr 29 12:03:35 2023 kern.alert kernel: [73016.435149] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000008
Sat Apr 29 12:03:35 2023 kern.alert kernel: [73016.443928] Mem abort info:
Sat Apr 29 12:03:35 2023 kern.alert kernel: [73016.446722]   ESR = 0x0000000096000005
Sat Apr 29 12:03:35 2023 kern.alert kernel: [73016.450456]   EC = 0x25: DABT (current EL), IL = 32 bits
Sat Apr 29 12:03:35 2023 kern.alert kernel: [73016.455763]   SET = 0, FnV = 0
Sat Apr 29 12:03:35 2023 kern.alert kernel: [73016.458806]   EA = 0, S1PTW = 0
Sat Apr 29 12:03:35 2023 kern.alert kernel: [73016.461933]   FSC = 0x05: level 1 translation fault
Sat Apr 29 12:03:35 2023 kern.alert kernel: [73016.466807] Data abort info:
Sat Apr 29 12:03:35 2023 kern.alert kernel: [73016.469679]   ISV = 0, ISS = 0x00000005
Sat Apr 29 12:03:35 2023 kern.alert kernel: [73016.473499]   CM = 0, WnR = 0
Sat Apr 29 12:03:35 2023 kern.alert kernel: [73016.476465] user pgtable: 4k pages, 39-bit VAs, pgdp=0000000045c9c000
Sat Apr 29 12:03:35 2023 kern.alert kernel: [73016.482892] [0000000000000008] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
Sat Apr 29 12:03:35 2023 kern.emerg kernel: [73016.491582] Internal error: Oops: 96000005 [#1] SMP

```