openwrt/openwrt #12339: ipq40xx: Linksys MR8300: fix propagated error since 22.03.x snapshot r20071 still works with the 5.10.168 kernel and has been propagated to other merge

roblad, 2023-04-06:
### Describe the bug

I noticed error with update new version of snapshot 

Also people have been experienced the same issue

Initially it was discovered in the forum threat

https://forum.openwrt.org/t/openwrt-snapshot-r22302-on-mr8300/154878/37

and suggested that it may be a recent kernel modification. Once that modification has been merged, it may bugs both 5.10 and 5.15 kernels. We already know that 5.15 snapshots don't boot, so there is now a possibility that the next 22.03.4 won't boot either.

On Github, [@clauded](https://forum.openwrt.org/u/clauded) posted this link
[https://lore.kernel.org/lkml/20230306013308.3884777-1-chengzhihao1@huawei.com/t/ 5](https://lore.kernel.org/lkml/20230306013308.3884777-1-chengzhihao1@huawei.com/t/)

### OpenWrt version

OpenWrt 22.03-SNAPSHOT r20068-788a0cf15c

### OpenWrt target/subtarget

SNAPSHOT 

### Device

MR8300 v. 1.1

### Image kind

Self-built image

### Steps to reproduce

just install EKO1 build

https://dl.eko.one.pl/luci/openwrt-22.03/ipq40xx/generic/luci-22.03-snapshot-r20090-c2331038b2-ipq40xx-generic-linksys_mr8300-squashfs-factory.bin

### Actual behaviour

Stop boot after upgrade flashed factory, overwrite second partition after sysupgrade

### Expected behaviour

flash factory bin with success

### Additional info

there is EKO1 threat I wrote

https://eko.one.pl/forum/viewtopic.php?pid=283918#p283918

### Diffconfig

```text
N/A
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

badulesia, 2023-04-07:
Here is the serial log with official openwrt-22.03-snapshot-r20119-50d707e344-ipq40xx-generic-linksys_mr8300-squashfs-factory.bin
The boot hangs while trying to detect mtd11. Same result than in this thread about master snapshot.
https://github.com/openwrt/openwrt/issues/12232

```
Format: Log Type - Time(microsec) - Message - Optional Info
Log Type: B - Since Boot(Power On Reset),  D - Delta,  S - Statistic
S - QC_IMAGE_VERSION_STRING=BOOT.BF.3.1.1-00108
S - IMAGE_VARIANT_STRING=DAACANAZA
S - OEM_IMAGE_VERSION_STRING=CRM
S - Boot Config, 0x00000025
S - Reset status Config, 0x00000010
S - Core 0 Frequency, 0 MHz
B -       262 - PBL, Start
B -      1341 - bootable_media_detect_entry, Start
B -      2618 - bootable_media_detect_success, Start
B -      2633 - elf_loader_entry, Start
B -      4048 - auth_hash_seg_entry, Start
B -      6214 - auth_hash_seg_exit, Start
B -     71411 - elf_segs_hash_verify_entry, Start
B -    194208 - PBL, End
B -    194233 - SBL1, Start
B -    286416 - pm_device_init, Start
D -         7 - pm_device_init, Delta
B -    287914 - boot_flash_init, Start
D -     87543 - boot_flash_init, Delta
B -    379768 - boot_config_data_table_init, Start
D -     14027 - boot_config_data_table_init, Delta - (419 Bytes)
B -    396488 - clock_init, Start
D -      7553 - clock_init, Delta
B -    407481 - CDT version:2,Platform ID:8,Major ID:1,Minor ID:0,Subtype:6
B -    410894 - sbl1_ddr_set_params, Start
B -    415988 - cpr_init, Start
D -         2 - cpr_init, Delta
B -    420371 - Pre_DDR_clock_init, Start
D -         4 - Pre_DDR_clock_init, Delta
D -     13172 - sbl1_ddr_set_params, Delta
B -    433676 - pm_driver_init, Start
D -         2 - pm_driver_init, Delta
B -    504376 - sbl1_wait_for_ddr_training, Start
D -        28 - sbl1_wait_for_ddr_training, Delta
B -    520471 - Image Load, Start
D -    144013 - QSEE Image Loaded, Delta - (269176 Bytes)
B -    664913 - Image Load, Start
D -      2119 - SEC Image Loaded, Delta - (2048 Bytes)
B -    675094 - Image Load, Start
D -    187367 - APPSBL Image Loaded, Delta - (444135 Bytes)
B -    862855 - QSEE Execution, Start
D -        56 - QSEE Execution, Delta
B -    869088 - SBL1, End
D -    676857 - SBL1, Delta
S - Flash Throughput, 2085 KB/s  (715778 Bytes,  343243 us)
S - DDR Frequency, 672 MHz

U-Boot 2012.07 [Chaos Calmer 15.05.1,r35193] (Sep 27 2018 - 09:37:31)

CBT U-Boot ver: 1.3.3

smem ram ptable found: ver: 1 len: 3
DRAM:  512 MiB
machid : 0x8010006
NAND:  ID = 9590dac2
Vendor = c2
Device = da
ONFI device found
SF NAND unsupported id:ff:ff:ff:ffSF: Unsupported manufacturer ff
ipq_spi: SPI Flash not found (bus/cs/speed/mode) = (0/0/48000000/0)
256 MiB
MMC:   qca_mmc: 0
PCI0 Link Intialized
[... 193 lines cut ...]
[    1.363458] 0x000000a80000-0x000005f80000 : "rootfs"
[    1.444539] mtd: setting mtd11 (rootfs) as root device
[    1.444881] mtdsplit: no squashfs found in "rootfs"
[    1.448600] 0x000005f80000-0x00000b780000 : "alt_kernel"
[    1.537516] 0x000006280000-0x00000b780000 : "alt_rootfs"
[    1.618388] 0x00000b780000-0x00000b880000 : "sysdiag"
[    1.619864] 0x00000b880000-0x00000ff00000 : "syscfg"
[    1.729994] ESS reset ok!
[    1.804236] ESS reset ok!
[    2.481747] EDMA using MAC@ - using
[    2.481764] f6:7f:00:e7:e2:40
[    2.567061] i2c /dev entries driver
[    2.568979] sdhci: Secure Digital Host Controller Interface driver
[    2.569364] sdhci: Copyright(c) Pierre Ossman
[    2.575699] sdhci-pltfm: SDHCI platform and OF driver helper
[    2.584203] NET: Registered protocol family 10
[    2.587367] Segment Routing with IPv6
[    2.590124] NET: Registered protocol family 17
[    2.593867] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[    2.598112] 8021q: 802.1Q VLAN Support v1.8
[    2.611369] Registering SWP/SWPB emulation handler
[    2.626542] ubi0: attaching mtd11
[    2.626590] ubi0 error: 0xc06f4048: VID header offset 2048 too large.
[ ▒[    2.636662] Waiting for root device /dev/ubiblock0_0...
```

badulesia, 2023-04-07:
And here's serial log with EKO1 build, very same issue.

```
Format: Log Type - Time(microsec) - Message - Optional Info
Log Type: B - Since Boot(Power On Reset),  D - Delta,  S - Statistic
S - QC_IMAGE_VERSION_STRING=BOOT.BF.3.1.1-00108
S - IMAGE_VARIANT_STRING=DAACANAZA
S - OEM_IMAGE_VERSION_STRING=CRM
S - Boot Config, 0x00000025
S - Reset status Config, 0x00000010
S - Core 0 Frequency, 0 MHz
B -       262 - PBL, Start
B -      1341 - bootable_media_detect_entry, Start
B -      2619 - bootable_media_detect_success, Start
B -      2633 - elf_loader_entry, Start
B -      4048 - auth_hash_seg_entry, Start
B -      6215 - auth_hash_seg_exit, Start
B -     71405 - elf_segs_hash_verify_entry, Start
B -    194176 - PBL, End
B -    194200 - SBL1, Start
B -    286362 - pm_device_init, Start
D -         7 - pm_device_init, Delta
B -    287860 - boot_flash_init, Start
D -     87494 - boot_flash_init, Delta
B -    379666 - boot_config_data_table_init, Start
D -     14023 - boot_config_data_table_init, Delta - (419 Bytes)
B -    396384 - clock_init, Start
D -      7580 - clock_init, Delta
B -    407401 - CDT version:2,Platform ID:8,Major ID:1,Minor ID:0,Subtype:6
B -    410815 - sbl1_ddr_set_params, Start
B -    415909 - cpr_init, Start
D -         2 - cpr_init, Delta
B -    420292 - Pre_DDR_clock_init, Start
D -         4 - Pre_DDR_clock_init, Delta
D -     13172 - sbl1_ddr_set_params, Delta
B -    433598 - pm_driver_init, Start
D -         2 - pm_driver_init, Delta
B -    504276 - sbl1_wait_for_ddr_training, Start
D -        28 - sbl1_wait_for_ddr_training, Delta
B -    520368 - Image Load, Start
D -    143991 - QSEE Image Loaded, Delta - (269176 Bytes)
B -    664789 - Image Load, Start
D -      2122 - SEC Image Loaded, Delta - (2048 Bytes)
B -    674975 - Image Load, Start
D -    187318 - APPSBL Image Loaded, Delta - (444135 Bytes)
B -    862689 - QSEE Execution, Start
D -        56 - QSEE Execution, Delta
B -    868925 - SBL1, End
D -    676727 - SBL1, Delta
S - Flash Throughput, 2085 KB/s  (715778 Bytes,  343172 us)
S - DDR Frequency, 672 MHz

U-Boot 2012.07 [Chaos Calmer 15.05.1,r35193] (Sep 27 2018 - 09:37:31)

CBT U-Boot ver: 1.3.3

smem ram ptable found: ver: 1 len: 3
DRAM:  512 MiB
machid : 0x8010006
NAND:  ID = 9590dac2
Vendor = c2
Device = da
ONFI device found
SF NAND unsupported id:ff:ff:ff:ffSF: Unsupported manufacturer ff
ipq_spi: SPI Flash not found (bus/cs/speed/mode) = (0/0/48000000/0)
256 MiB
MMC:   qca_mmc: 0
PCI0 Link Intialized
In:    serial
Out:   serial
[... 192 lines cut ...]
[    1.443169] mtd: setting mtd11 (rootfs) as root device
[    1.443518] mtdsplit: no squashfs found in "rootfs"
[    1.447227] 0x000005f80000-0x00000b780000 : "alt_kernel"
[    1.535450] 0x000006280000-0x00000b780000 : "alt_rootfs"
[    1.615734] 0x00000b780000-0x00000b880000 : "sysdiag"
[    1.617229] 0x00000b880000-0x00000ff00000 : "syscfg"
[    1.726570] ESS reset ok!
[    1.800794] ESS reset ok!
[    2.477805] EDMA using MAC@ - using
[    2.477819] 0a:c0:9b:d5:70:6d
[    2.567202] i2c /dev entries driver
[    2.569118] sdhci: Secure Digital Host Controller Interface driver
[    2.569506] sdhci: Copyright(c) Pierre Ossman
[    2.575840] sdhci-pltfm: SDHCI platform and OF driver helper
[    2.584479] NET: Registered protocol family 10
[    2.587529] Segment Routing with IPv6
[    2.590265] NET: Registered protocol family 17
[    2.594027] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[    2.598252] 8021q: 802.1Q VLAN Support v1.8
[    2.611516] Registering SWP/SWPB emulation handler
[    2.626694] ubi0: attaching mtd11
[    2.626739] ubi0 error: 0xc06f4048: VID header offset 2048 too large.
[ ▒[    2.636824] Waiting for root device /dev/ubiblock0_0...

```

roblad, 2023-04-07:
Could you try that too:

https://dl.eko.one.pl/luci/openwrt-22.03/ipq40xx/generic/luci-22.03-snapshot-r20090-c2331038b2-ipq40xx-generic-linksys_mr8300-squashfs-factory.bin

roblad, 2023-04-07:
thnx

roblad, 2023-04-10:
I suggest close the https://github.com/openwrt/openwrt/issues/12232 here are console logs, the problem is the same

closed 2023-04-15 by openwrt-bot

hhasert, 2023-04-16:
Question: Will the 22.03.4 images for Linksys be recompiled and shared on the download links or do we have to wait for 22.03.5 ?

roblad, 2023-04-16:
> Question: Will the 22.03.4 images for Linksys MR8300 & EA8300 be recompiled and shared on the download links or do we have to wait for 22.03.5 ?

You can use that

OpenWrt 22.03-SNAPSHOT r20127-074db57936 / LuCI openwrt-22.03 branch git-23.093.57104-ce20b4a

[https://dl.eko.one.pl/test/openwrt-22.03-snapshot-r20127-074db57936-ipq40xx-generic-linksys_mr8300-squashfs-factory.bin 2](https://dl.eko.one.pl/test/openwrt-22.03-snapshot-r20127-074db57936-ipq40xx-generic-linksys_mr8300-squashfs-factory.bin)

https://dl.eko.one.pl/test/openwrt-22.03-snapshot-r20127-074db57936-ipq40xx-generic-linksys_mr8300-squashfs-sysupgrade.bin

hhasert, 2023-04-16:
Thanks but I need the EA8300 images (or are they identical since the 2 types are similar but on memory size ) ?

commit aad34818 (2023-04-15): kernel: backport fix for recently introduced UBI bug

Import commit "ubi: Fix failure attaching when vid_hdr offset equals to
(sub)page size" which did not yet make it to stable upstream Linux trees.

Fixes: #12232
Fixes: #12339
PR #12398 (2023-04-15): kernel: backport fix for recently introduced UBI bug
Import commit "ubi: Fix failure attaching when vid_hdr offset equals to (sub)page size" which did not yet make it to stable upstream Linux trees.

Fixes: #12232
Fixes: #12339
commit 074db579 (2023-04-15): kernel: backport fix for recently introduced UBI bug

Import commit "ubi: Fix failure attaching when vid_hdr offset equals to
(sub)page size" which did not yet make it to stable upstream Linux trees.

Fixes: #12232
Fixes: #12339

(cherry picked from commit aad34818b50029e07ed9221ae46f9770d6e29785)
PR #12399 (2023-04-15): [22.03] kernel: backport fix for recently introduced UBI bug
Import commit "ubi: Fix failure attaching when vid_hdr offset equals to (sub)page size" which did not yet make it to stable upstream Linux trees.

Fixes: #12232
Fixes: #12339

(cherry picked from commit aad34818b50029e07ed9221ae46f9770d6e29785)