openwrt/openwrt #12350: Problem with usbip and libudev-zero

dob46, 2023-04-08:
### Describe the bug

I have a problem with libudev-zero and usbip. usbip port command show 
> libusbip: error: no available usb controllers
usbip: error: open vhci_driver
usbip: error: list imported devices
This command works with libudev from LEDE 17.01.9

part of strace output with libudev
>close(3)                                = 0
munmap(0xb6f1c000, 4096)                = 0
open("/etc/udev/udev.conf", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
statx(AT_FDCWD, "/sys/subsystem/platform/devices/vhci_hcd.0", AT_STATX_SYNC_AS_STAT, STATX_BASIC_STATS, 0xbed743b0) = -1 ENOENT (No such file or directory)
statx(AT_FDCWD, "/sys/bus/platform/devices/vhci_hcd.0", AT_STATX_SYNC_AS_STAT, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFDIR|0755, stx_size=0, ...}) = 0
readlink("/sys/bus/platform/devices/vhci_hcd.0", "../../../devices/platform/vhci_h"..., 1024) = 36
statx(AT_FDCWD, "/sys/devices/platform/vhci_hcd.0/uevent", AT_STATX_SYNC_AS_STAT, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=4096, ...}) = 0
statx(AT_FDCWD, "/sys/devices/platform/vhci_hcd.0/nports", AT_STATX_SYNC_AS_STAT|AT_SYMLINK_NOFOLLOW, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0444, stx_size=4096, ...}) = 0
open("/sys/devices/platform/vhci_hcd.0/nports", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
read(3, "16\n", 4096)                   = 3
close(3)                                = 0
mmap2(NULL, 16384, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6cb2000
readlink("/sys/devices/platform", 0xbed73b7c, 1024) = -1 EINVAL (Invalid argument)
statx(AT_FDCWD, "/sys/devices/platform/uevent", AT_STATX_SYNC_AS_STAT, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=4096, ...}) = 0
open("/sys/devices/platform", O_RDONLY|O_LARGEFILE|O_CLOEXEC|O_DIRECTORY) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6cb0000
getdents64(3, 0xb6cb01a8 /* 20 entries */, 2048) = 624
getdents64(3, 0xb6cb01a8 /* 0 entries */, 2048) = 0
close(3)                                = 0
munmap(0xb6cb0000, 8192)                = 0
statx(AT_FDCWD, "/sys/devices/platform/vhci_hcd.0/status", AT_STATX_SYNC_AS_STAT|AT_SYMLINK_NOFOLLOW, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0444, stx_size=4096, ...}) = 0
open("/sys/devices/platform/vhci_hcd.0/status", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
read(3, "hub port sta spd dev      sockfd"..., 4096) = 637
close(3)                                = 0
ioctl(1, TIOCGWINSZ, {ws_row=27, ws_col=110, ws_xpixel=882, ws_ypixel=475}) = 0
writev(1, [{iov_base="Imported USB devices", iov_len=20}, {iov_base="\n", iov_len=1}], 2Imported USB devices
) = 21
writev(1, [{iov_base="====================", iov_len=20}, {iov_base="\n", iov_len=1}], 2====================
) = 21
munmap(0xb6cb2000, 16384)               = 0
munmap(0xb6cb6000, 16384)               = 0
munmap(0xb6cba000, 32768)               = 0
munmap(0xb6cc2000, 16384)               = 0
munmap(0xb6cc6000, 32768)               = 0
munmap(0xb6cce000, 16384)               = 0
munmap(0xb6cd2000, 32768)               = 0
munmap(0xb6cda000, 32768)               = 0
munmap(0xb6ce2000, 16384)               = 0
munmap(0xb6ce6000, 32768)               = 0
munmap(0xb6cee000, 16384)               = 0
munmap(0xb6cf2000, 32768)               = 0
munmap(0xb6cfa000, 32768)               = 0
munmap(0xb6d02000, 16384)               = 0
munmap(0xb6d06000, 32768)               = 0
munmap(0xb6d0e000, 16384)               = 0
munmap(0xb6d12000, 32768)               = 0
munmap(0xb6d1a000, 16384)               = 0
munmap(0xb6d1e000, 32768)               = 0
munmap(0xb6d26000, 32768)               = 0
munmap(0xb6d2e000, 16384)               = 0
munmap(0xb6d32000, 32768)               = 0
munmap(0xb6d3a000, 16384)               = 0
munmap(0xb6d3e000, 16384)               = 0
munmap(0xb6d42000, 32768)               = 0
munmap(0xb6d4a000, 16384)               = 0
munmap(0xb6d4e000, 32768)               = 0
munmap(0xb6d56000, 16384)               = 0
munmap(0xb6d5a000, 16384)               = 0
[... 184 lines cut ...]
+++ exited with 1 +++

### OpenWrt version

r20060-f7541aecdc

### OpenWrt target/subtarget

ipq806x/generic

### Device

Netgear Nighthawk X4S R7800

### Image kind

Self-built image

### Steps to reproduce

Used `usbip port` command with libudev-zero library

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dangowrt, 2023-04-17:
@illiliti Any ideas?

illiliti, 2023-04-18:
I think I know what's wrong. A fix is on the way.

illiliti, 2023-04-18:
Pushed a fix to libudev-zero master. I'll cut a release once I get confirmation that issue is fixed.

dob46, 2023-04-22:
@illiliti 
It works
```
root@OpenWrt:~# usbip list -l
 - busid 3-1 (04e8:6860)
   Samsung Electronics Co., Ltd : Galaxy A5 (MTP) (04e8:6860)

root@OpenWrt:~# usbip port
Imported USB devices
====================
root@OpenWrt:~# ldd /usr/sbin/usbip
        /lib/ld-musl-armhf.so.1 (0xb6f5b000)
        libusbip.so.0 => /usr/lib/libusbip.so.0 (0xb6f3a000)
        libwrap.so.0 => /usr/lib/libwrap.so.0 (0xb6f24000)
        libudev.so.1 => /lib/libudev.so.1 (0xb6ef7000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0xb6edb000)
        libc.so => /lib/ld-musl-armhf.so.1 (0xb6f5b000)
```

dob46, 2023-04-22:
@illiliti 
It doesn't work
```
close(3)                                = 0
munmap(0xb6f38000, 4096)                = 0
access("/sys/bus/platform/devices/vhci_hcd.0", R_OK) = 0
readlink("/sys/bus/platform/devices/vhci_hcd.0/subsystem", "../../../bus/platform", 4096) = 21
readlink("/sys", 0xbed9dbd0, 4064)      = -1 EINVAL (Invalid argument)
readlink("/sys/bus", 0xbed9dbd0, 4068)  = -1 EINVAL (Invalid argument)
readlink("/sys/bus/platform", 0xbed9dbd0, 4077) = -1 EINVAL (Invalid argument)
readlink("/sys/bus/platform/devices", 0xbed9dbd0, 4085) = -1 EINVAL (Invalid argument)
readlink("/sys/bus/platform/devices/vhci_hcd.0", "../../../devices/platform/vhci_h"..., 4096) = 36
readlink("/sys/bus/platform/devices/..", 0xbed9dbd0, 4062) = -1 EINVAL (Invalid argument)
readlink("/sys/devices", 0xbed9dbd0, 4076) = -1 EINVAL (Invalid argument)
readlink("/sys/devices/platform", 0xbed9dbd0, 4085) = -1 EINVAL (Invalid argument)
readlink("/sys/devices/platform/vhci_hcd.0", 0xbed9dbd0, 4096) = -1 EINVAL (Invalid argument)
readlink("/sys/devices/platform/vhci_hcd.0/driver", "../../../bus/platform/drivers/vh"..., 4096) = 38
open("/sys/devices/platform/vhci_hcd.0/uevent", O_RDONLY|O_LARGEFILE) = 3
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6f38000
read(3, "DRIVER=vhci_hcd\nMODALIAS=platfor"..., 1024) = 43
read(3, "", 1024)                       = 0
close(3)                                = 0
munmap(0xb6f38000, 4096)                = 0
statx(AT_FDCWD, "/sys/devices/platform/vhci_hcd.0/nports", AT_STATX_SYNC_AS_STAT|AT_SYMLINK_NOFOLLOW, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0444, stx_size=4096, ...}) = 0
open("/sys/devices/platform/vhci_hcd.0/nports", O_RDONLY|O_LARGEFILE) = 3
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6f38000
readv(3, [{iov_base="16\n", iov_len=4094}, {iov_base="", iov_len=1024}], 2) = 3
readv(3, [{iov_base="", iov_len=4091}, {iov_base="", iov_len=1024}], 2) = 0
close(3)                                = 0
munmap(0xb6f38000, 4096)                = 0
mmap2(NULL, 16384, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6ce5000
readlink("/sys/devices/platform/subsystem", 0xbed9dc04, 4096) = -1 ENOENT (No such file or directory)
readlink("/sys/devices/subsystem", 0xbed9dc04, 4096) = -1 ENOENT (No such file or directory)
writev(2, [{iov_base="libusbip: error: no available us"..., iov_len=46}, {iov_base=NULL, iov_len=0}], 2libusbip: error: no available usb controllers
) = 46
munmap(0xb6ce5000, 16384)               = 0
writev(2, [{iov_base="usbip: error: open vhci_driver\n", iov_len=31}, {iov_base=NULL, iov_len=0}], 2usbip: error: open vhci_driver
) = 31
munmap(0xb6ce9000, 16384)               = 0
munmap(0xb6ced000, 32768)               = 0
munmap(0xb6cf5000, 16384)               = 0
munmap(0xb6cf9000, 32768)               = 0
munmap(0xb6d01000, 16384)               = 0
munmap(0xb6d05000, 32768)               = 0
munmap(0xb6d0d000, 32768)               = 0
munmap(0xb6d15000, 16384)               = 0
munmap(0xb6d19000, 32768)               = 0
munmap(0xb6d21000, 16384)               = 0
munmap(0xb6d25000, 32768)               = 0
munmap(0xb6d2d000, 32768)               = 0
munmap(0xb6d35000, 16384)               = 0
munmap(0xb6d39000, 32768)               = 0
munmap(0xb6d41000, 16384)               = 0
munmap(0xb6d45000, 32768)               = 0
munmap(0xb6d4d000, 16384)               = 0
munmap(0xb6d51000, 32768)               = 0
munmap(0xb6d59000, 32768)               = 0
munmap(0xb6d61000, 16384)               = 0
munmap(0xb6d65000, 32768)               = 0
munmap(0xb6d6d000, 16384)               = 0
munmap(0xb6d71000, 16384)               = 0
munmap(0xb6d75000, 32768)               = 0
munmap(0xb6d7d000, 16384)               = 0
munmap(0xb6d81000, 32768)               = 0
munmap(0xb6d89000, 16384)               = 0
munmap(0xb6d8d000, 16384)               = 0
munmap(0xb6d91000, 32768)               = 0
munmap(0xb6d99000, 16384)               = 0
munmap(0xb6d9d000, 32768)               = 0
munmap(0xb6da5000, 32768)               = 0
[... 33 lines cut ...]
munmap(0xb6f29000, 4096)                = 0
munmap(0xb6e40000, 16384)               = 0
munmap(0xb6e44000, 8192)                = 0
munmap(0xb6e46000, 8192)                = 0
munmap(0xb6e48000, 8192)                = 0
munmap(0xb6e4a000, 8192)                = 0
munmap(0xb6e4c000, 8192)                = 0
munmap(0xb6f2a000, 4096)                = 0
munmap(0xb6f2b000, 8192)                = 0
munmap(0xb6f2d000, 4096)                = 0
munmap(0xb6f2e000, 4096)                = 0
munmap(0xb6f2f000, 4096)                = 0
munmap(0xb6f30000, 4096)                = 0
munmap(0xb6f31000, 4096)                = 0
munmap(0xb6f32000, 4096)                = 0
munmap(0xb6f33000, 4096)                = 0
munmap(0xb6f34000, 4096)                = 0
munmap(0xb6f35000, 4096)                = 0
munmap(0xb6f36000, 4096)                = 0
munmap(0xb6f37000, 4096)                = 0
writev(2, [{iov_base="usbip: error: list imported devi"..., iov_len=36}, {iov_base=NULL, iov_len=0}], 2usbip: error: list imported devices
) = 36
exit_group(1)                           = ?
+++ exited with 1 +++
```

dob46, 2023-04-22:
@illiliti 
It works
```
root@OpenWrt:~/libudev-zero/data# usbip port
Imported USB devices
====================
root@OpenWrt:~/libudev-zero/data# usbip list -l
 - busid 3-1 (04e8:6860)
   Samsung Electronics Co., Ltd : Galaxy A5 (MTP) (04e8:6860)

root@OpenWrt:~/libudev-zero/data# ldd /usr/sbin/usbip
        /lib/ld-musl-armhf.so.1 (0xb6f5c000)
        libusbip.so.0 => /usr/lib/libusbip.so.0 (0xb6f3b000)
        libwrap.so.0 => /usr/lib/libwrap.so.0 (0xb6f25000)
        libudev.so.1 => /usr/lib/libudev.so.1 (0xb6fd7000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0xb6f09000)
        libc.so => /lib/ld-musl-armhf.so.1 (0xb6f5c000)
```

dangowrt, 2023-04-22:
Let's keep this issue open until libudev-zero is actually updated here, so we don't forget about it.

whatsalinux, 2023-04-23:
anyone have a copy of the fixed libudev.so.1 compiled for aarch64?

dob46, 2023-04-26:
@whatsalinux 
This is a ipk file for arm cortex a53
[https://dob46.mazowsze.pl/libudev-zero_1.0.2-1_aarch64_cortex-a53.ipk](url)

dob46, 2023-06-15:
@dangowrt 
I have a question, is it planned to update libudev-zero to version 1.0.2 in OpenWrt 23.05?

TaylorTWBrown, 2023-07-16:
I just compiled libudev-zero 1.0.3 for x86_64 and now USBIP is working.

I'll have to read how to contribute to openwrt!

TaylorTWBrown, 2023-07-17:
I've opened this PR to upgrade the libudev-zero to 1.0.3.

It's my first contribution to OpenWrt, so I appreciate any advice navigating the PR process and standards. Have a look!

https://github.com/openwrt/packages/pull/21586

closed 2023-07-31 by dob46