openwrt/openwrt #12358: mac80211: ath11k: No 160MHz support

Soromeister, 2023-04-09:
### Describe the bug

While 160MHz is selectable in the OpenWRT option, the radio will actually function at 80MHz. This is easily revealed when doing a scan via WiFi Scanner app since it doesn't take the full spectrum, only half of it.

### OpenWrt version

r22528-0c53801968

### OpenWrt target/subtarget

ipq807x/generic

### Device

Xiaomi AX9000

### Image kind

Official downloaded image

### Steps to reproduce

On Radio3, configure a WiFi for 160MHz. Name it a unique name.
Reboot router.
Open the WiFi Scanner app on phone.
Notice the width of the signal only covers 80MHz

### Actual behaviour

Signal of the AP is 80MHz instead of 160MHz.

### Expected behaviour

Signal of the AP should be 160MHz.

### Diffconfig

```text
-ash: ./scripts/diffconfig.sh: not found
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

AlexeiGHub, 2023-04-09:
On my Redmi AX6 with clients(ax200 and smarphone) wich support 160MHz all OK 2400MBit/sec.

closed 2023-04-10 by Soromeister

st7105, 2023-04-13:
AX3600 same problem after ath11-next sync commit.