openwrt/openwrt #12359: v22.03 / MT7621 / sporadic PCI issue / no wifi on 02:00.0

cjdelisle, 2023-04-09:
### Describe the bug

When starting the device, occasionally one radio will fail to come up. In this case it is the 5ghz wifi chip.

## Raw data

Successful reboot:
* `dmesg` - https://gist.github.com/cjdelisle/1c2cc65e6e3ea036db8899b6c6bb480e
* `lspci -vvvv` - https://gist.github.com/cjdelisle/3cfdc0fc31603c44964f9872ded10614
* `cat /proc/interrupts` - https://gist.github.com/cjdelisle/1752cdc8d90e819fddb558c7ac1465cb

Failed reboot (wifi did not come up):
* `dmesg` - https://gist.github.com/cjdelisle/3cd375bf2f76bd731d8bbbd85f3e3a84
* `lspci -vvvv` - https://gist.github.com/cjdelisle/4eb6059c8e714cdfd8f21e1e52ec5cf1
* `cat /proc/interrupts` - https://gist.github.com/cjdelisle/efbd78be8b61b282b0ddfaa157903132

Diff:
* `dmesg` (with timestamps removed) - https://gist.github.com/cjdelisle/9a373796367c5fdf79d9ab661b3a119e
* `lspci` - https://gist.github.com/cjdelisle/6af284e17907b17c2347e60694be28ab
* The proc/interrupts numbers are the same in both cases

---

## Notes

This build uses [an out-of-tree DTS file](https://gist.github.com/cjdelisle/d64d12d653f00f608552d164be6a5a68) but the PCI section is identical to a number of other devices, such as [mt7621_adslr_g7.dts](https://github.com/openwrt/openwrt/blob/openwrt-22.03/target/linux/ramips/dts/mt7621_adslr_g7.dts#L85).

### lspci

The lspci shows that bridge `00:01.0` is configured differently, it has:
* `BridgeCtl`: `SERR-`
* `Capabilities: [70] Express (v2) Root Port (Slot-), MSI 00` -> `LnkCap`: `BwNot+`

I don't know if these are representative of a misconfiguration of the bridge, or an effect of the fact that the radio cannot be accessed. Down in the section of the wifi chip itself, I interpret `AERCap:	First Error Pointer: 0e,` as a PCI error which is why I reported this as a PCI bug rather than a wifi driver issue.

### dmesg

The real elephant in the room seems to be this:

```diff
- pci 0000:00:01.0:   bridge window [mem 0x60200000-0x602fffff]
- pci 0000:00:01.0:   bridge window [mem 0x60300000-0x604fffff pref]
+ pci 0000:00:01.0:   bridge window [mem 0x60000000-0x600fffff]
+ pci 0000:00:01.0:   bridge window [mem 0x60000000-0x600fffff pref]
```

Rather than using `0x60200000-0x602fffff` and `0x60300000-0x604fffff`, it is creating two overlapping bridge windows, which I don't know much about PCI but that looks like something that Could Not Possibly Work.

But how it gets to this decision, or how to nudge it over to doing the right thing every time, this is where I'm stumped.

I notice there are a lot of PCI errors and the driver ends up auto-detecting quite a lot, but I understand from [this thread](https://forum.openwrt.org/t/mt7621-pcie-dts-issues-mt76-stopped-working-with-latest-updates/14623/44) that the PCI configuration is a bit "precarious" in that what fixes one device breaks another.

I guess it should be obvious that this is some kind of a race condition, due to its sporadic nature - so "clearly" a well placed `udelay()` should clear it all up, but I wouldn't know where to try putting it.

### OpenWrt version

r19803-9a599fee93

### OpenWrt target/subtarget

ramips/mt7621

### Device

Comfast CF-EW72 v2

### Image kind

Self-built image

### Steps to reproduce

* Build 22.03 for CF-EW72v2
* Flash and boot
* Reboot a bunch of times
* Eventually you should see it

### Actual behaviour

The mt7663 does not work

### Expected behaviour

The mt7663 works

### Diffconfig

```text
CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_ramips_mt7621_DEVICE_comfast_cf-ew72-v2=y
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_BUSYBOX_CONFIG_FEATURE_SEAMLESS_XZ=y
CONFIG_BUSYBOX_CONFIG_FEATURE_TAR_AUTODETECT=y
CONFIG_BUSYBOX_CONFIG_FEATURE_TAR_UNAME_GNAME=y
CONFIG_BUSYBOX_CONFIG_UNXZ=y
CONFIG_BUSYBOX_CONFIG_XZ=y
CONFIG_BUSYBOX_CONFIG_XZCAT=y
# CONFIG_FEED_luci is not set
# CONFIG_FEED_packages is not set
# CONFIG_FEED_routing is not set
# CONFIG_FEED_telephony is not set
CONFIG_PACKAGE_ip-full=m
CONFIG_PACKAGE_kmod-gre=m
CONFIG_PACKAGE_kmod-iptunnel=m
CONFIG_PACKAGE_libbpf=m
CONFIG_PACKAGE_libelf=m
CONFIG_PACKAGE_pkteeros=y
CONFIG_PACKAGE_zlib=m
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

cjdelisle, 2023-04-09:
Another note:

```diff
- pci 0000:00:01.0: enabling device (0006 -> 0007)
+ pci 0000:00:01.0: enabling device (0000 -> 0003)
```

When the device is broken, the bridge is missing `PCI_COMMAND_MASTER` which seems like it would be important for a bridge to have.

cjdelisle, 2023-04-10:
Added pci-mt7621.c with whatever patches may exist on it here, for reference: https://gist.github.com/cjdelisle/6274d94e48e9cff71de6296f63a684ab

cjdelisle, 2023-04-12:
Update: I have confirmed this bug on [master branch](https://github.com/openwrt/openwrt/commit/e04356362ce3bf84dc554cf27bcc6a991e5fa604). New dmesg is here: https://gist.github.com/cjdelisle/7700473e07fad5dbe09a2f094e6de688

From studying the source code, I have come to the opinion that the "erratic" BARs / bridge windows that are requested by the hardware are a red herring. Although the hardware makes some (in some cases) ridiculous requests, the driver does the right thing and provides reasonable memory maps.

At this point I'm focusing my attention on this line: [pci 0000:00:01.0: enabling device (0000 -> 0003)](https://gist.github.com/cjdelisle/3cd375bf2f76bd731d8bbbd85f3e3a84#file-gistfile1-txt-L269)

* `0003` means `PCI_COMMAND_IO | PCI_COMMAND_MEMORY` while
* `0007` means `PCI_COMMAND_IO | PCI_COMMAND_MEMORY | PCI_COMMAND_MASTER` so the bus `0000:00:01.0` is missing `PCI_COMMAND_MASTER`.

As of around [mt7621-pci 1e140000.pcie: PCIE1 enabled](https://gist.github.com/cjdelisle/3cd375bf2f76bd731d8bbbd85f3e3a84#file-gistfile1-txt-L143), the [PCI_COMMAND_MASTER flag is set to 1](https://gist.github.com/cjdelisle/6274d94e48e9cff71de6296f63a684ab#file-pci-mt7621-c-L594-L596) on "PCI slot 1"

But moments later, we see that [PCI_COMMAND_MASTER is missing](https://github.com/torvalds/linux/blob/v5.15/drivers/pci/setup-res.c#L503) from device `0000:00:01.0`.

So my questions:
1. Is PCI slot 1 from pci-mt7621.c the same thing as `0000:00:01.0` ?
2. Is the `PCI_COMMAND_MASTER` being effectively assigned in pci-mt7621.c ?
3. And if so, is it being removed by software, or by hardware ?

I think my next step will be to `#define DEBUG` to try to get more logs, particularly if `PCI_COMMAND_MASTER` is getting lifted in [__pci_set_master()](https://github.com/torvalds/linux/blob/e62252bc55b6d4eddc6c2bdbf95a448180d6a08d/drivers/pci/pci.c#L4377).

cjdelisle, 2023-04-12:
I have a new kernel with DEBUG defined:
* Successful boot: https://gist.github.com/cjdelisle/5a56ebaf32abad7aaf9a3cef2d861d95
* Failed boot: https://gist.github.com/cjdelisle/9975e63b527a28efe13b66a321a11ca5
* diff: https://gist.github.com/cjdelisle/06a2f67a6eba92460ed4431584fcc38c

It looks like when `PCI_COMMAND_MASTER` is initially set, it does not stick - perhaps the hardware is not yet ready. What I might look at doing next is polling the PCI_COMMAND register until it is stays. Also worth noting that on this system, the device appears to be on bus 2 port 1, so bus 1 port 2 is unused.

cjdelisle, 2023-04-13:
Update: I added some PCI_DEBUG_COMMAND() in a few places:

```c
#define PCI_DEBUG_COMMAND(_dev) \
       do { \
               u16 command_reg = 0; \
               pci_read_config_word(_dev, PCI_COMMAND, &command_reg); \
               pci_info(_dev, "%s PCI_COMMAND = %x\n", __func__, command_reg); \
       } while (0)
```

* Failed dmesg: https://gist.github.com/cjdelisle/bfdfe3624aca9cd38e83cd5a5759f6ff

And you can see that somewhere between `pci_read_bases` and `__pci_setup_bridge`, `PCI_COMMAND_MASTER` is lost. Since this is in the synchronous PCI code, this should be quite easy to track down the line where it is being dropped.

```
[    1.530495] pci 0000:00:01.0: pci_read_bases PCI_COMMAND = 6
[    1.536169] pci 0000:00:01.0: reg 0x10: [mem 0x00000000-0x7fffffff]
[    1.542433] pci 0000:00:01.0: reg 0x14: [mem 0x60510000-0x6051ffff]
[    1.548734] pci 0000:00:01.0: supports D1
[    1.552740] pci 0000:00:01.0: PME# supported from D0 D1 D3hot
[    1.558463] pci 0000:00:01.0: PME# disabled
[    1.559768] pci_bus 0000:00: fixups for bus
[    1.559792] pci 0000:00:00.0: scanning [bus 01-01] behind bridge, pass 0
[    1.559809] pci 0000:00:01.0: scanning [bus 02-02] behind bridge, pass 0
[    1.559822] pci 0000:00:00.0: scanning [bus 00-00] behind bridge, pass 1
[    1.560020] pci_bus 0000:01: scanning bus
[    1.560089] pci 0000:01:00.0: [14c3:7603] type 00 class 0x028000
[    1.566134] pci 0000:01:00.0: pci_read_bases PCI_COMMAND = 0
[    1.571787] pci 0000:01:00.0: reg 0x10: initial BAR value 0x00000000 invalid
[    1.578827] pci 0000:01:00.0: reg 0x10: [mem size 0x00100000]
[    1.584703] pci 0000:01:00.0: PME# supported from D0 D3hot D3cold
[    1.590773] pci 0000:01:00.0: PME# disabled
[    1.592071] pci_bus 0000:01: fixups for bus
[    1.592096] pci 0000:00:00.0: PCI bridge to [bus 01-ff]
[    1.597334] pci 0000:00:00.0:   bridge window [io  0x0000-0x0fff]
[    1.603418] pci 0000:00:00.0:   bridge window [mem 0x60000000-0x600fffff]
[    1.610176] pci 0000:00:00.0:   bridge window [mem 0x60100000-0x601fffff pref]
[    1.617381] pci_bus 0000:01: bus scan returning with max=01
[    1.617396] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
[    1.624013] pci 0000:00:01.0: scanning [bus 00-00] behind bridge, pass 1
[    1.624204] pci_bus 0000:02: scanning bus
[    1.696614] pci 0000:02:00.0: [14c3:7663] type 00 class 0x000280
[    1.702636] pci 0000:02:00.0: pci_read_bases PCI_COMMAND = 0
[    1.708290] pci 0000:02:00.0: reg 0x10: initial BAR value 0x00000000 invalid
[    1.715326] pci 0000:02:00.0: reg 0x10: [mem size 0x00100000 64bit pref]
[    1.722011] pci 0000:02:00.0: reg 0x18: initial BAR value 0x00000000 invalid
[    1.729043] pci 0000:02:00.0: reg 0x18: [mem size 0x00004000 64bit pref]
[    1.735743] pci 0000:02:00.0: reg 0x20: initial BAR value 0x00000000 invalid
[    1.742774] pci 0000:02:00.0: reg 0x20: [mem size 0x00001000 64bit pref]
[    1.749561] pci 0000:02:00.0: supports D1 D2
[    1.753833] pci 0000:02:00.0: PME# supported from D0 D1 D2 D3hot D3cold
[    1.760421] pci 0000:02:00.0: PME# disabled
[    1.760468] pci 0000:02:00.0: 2.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s PCIe x1 link at 0000:00:01.0 (capable of 4.000 Gb/s with 5.0 GT/s PCIe x1 link)
[    1.776550] pci_bus 0000:02: fixups for bus
[    1.776575] pci 0000:00:01.0: PCI bridge to [bus 02-ff]
[    1.781790] pci 0000:00:01.0:   bridge window [io  0x0000-0x0fff]
[    1.787880] pci 0000:00:01.0:   bridge window [mem 0x60000000-0x600fffff]
[    1.794659] pci 0000:00:01.0:   bridge window [mem 0x60000000-0x600fffff pref]
[    1.801849] pci_bus 0000:02: bus scan returning with max=02
[    1.801862] pci_bus 0000:02: busn_res: [bus 02-ff] end is updated to 02
[    1.808465] pci_bus 0000:00: bus scan returning with max=02
[    1.808524] pci 0000:00:00.0: BAR 0: no space for [mem size 0x80000000]
[    1.815136] pci 0000:00:00.0: BAR 0: failed to assign [mem size 0x80000000]
[    1.822070] pci 0000:00:01.0: BAR 0: no space for [mem size 0x80000000]
[    1.828669] pci 0000:00:01.0: BAR 0: failed to assign [mem size 0x80000000]
[    1.835620] pci 0000:00:00.0: BAR 8: assigned [mem 0x60000000-0x600fffff]
[    1.842399] pci 0000:00:00.0: BAR 9: assigned [mem 0x60100000-0x601fffff pref]
[    1.849590] pci 0000:00:01.0: BAR 8: assigned [mem 0x60200000-0x602fffff]
[    1.856366] pci 0000:00:01.0: BAR 9: assigned [mem 0x60300000-0x604fffff pref]
[    1.863576] pci 0000:00:00.0: BAR 1: assigned [mem 0x60500000-0x6050ffff]
[    1.870341] pci 0000:00:01.0: BAR 1: assigned [mem 0x60510000-0x6051ffff]
[    1.877126] pci 0000:00:00.0: BAR 7: assigned [io  0x1e160000-0x1e160fff]
[    1.883904] pci 0000:00:01.0: BAR 7: assigned [io  0x1e161000-0x1e161fff]
[    1.890670] pci 0000:01:00.0: BAR 0: assigned [mem 0x60000000-0x600fffff]
[    1.897447] pci 0000:00:00.0: PCI bridge to [bus 01]
[    1.902407] pci 0000:00:00.0:   bridge window [io  0x1e160000-0x1e160fff]
[    1.909168] pci 0000:00:00.0:   bridge window [mem 0x60000000-0x600fffff]
[    1.915941] pci 0000:00:00.0:   bridge window [mem 0x60100000-0x601fffff pref]
[    1.923146] pci 0000:00:00.0: __pci_setup_bridge PCI_COMMAND = 6
[    1.929132] pci 0000:02:00.0: BAR 0: assigned [mem 0x60300000-0x603fffff 64bit pref]
[    1.936879] pci 0000:02:00.0: BAR 2: assigned [mem 0x60400000-0x60403fff 64bit pref]
[    1.944620] pci 0000:02:00.0: BAR 4: assigned [mem 0x60404000-0x60404fff 64bit pref]
[    1.952357] pci 0000:00:01.0: PCI bridge to [bus 02]
[    1.957300] pci 0000:00:01.0:   bridge window [io  0x1e161000-0x1e161fff]
[    1.964073] pci 0000:00:01.0:   bridge window [mem 0x60200000-0x602fffff]
[    1.970830] pci 0000:00:01.0:   bridge window [mem 0x60300000-0x604fffff pref]
[    1.978035] pci 0000:00:01.0: __pci_setup_bridge PCI_COMMAND = 0
```

cjdelisle, 2023-04-13:
After a bit more debugging, it seems that the PCI_COMMAND flags are being lost from `0000:00:01.0` when `pci_scan_bridge_extend()` is called on `0000:00:00.0`.

```
[    1.576930] pci 0000:00:00.0: pci_scan_child_bus_extend:2868 (e8d:801) PCI_COMMAND = 6
[    1.584832] pci 0000:00:01.0: pci_scan_child_bus_extend:2868 (e8d:801) PCI_COMMAND = 6
[    1.592715] pci 0000:00:00.0: pci_scan_child_bus_extend:2883 (e8d:801) PCI_COMMAND = 6                                                                                                                  
[    1.600613] pci 0000:00:00.0: pci_scan_bridge_extend:1229 (e8d:801) PCI_COMMAND = 6                                                                                                                     
[    1.608262] pci 0000:00:00.0: scanning [bus 01-01] behind bridge, pass 0                                                                                                                                
[    1.608275] pci 0000:00:00.0: pci_scan_bridge_extend:1258 (e8d:801) PCI_COMMAND = 6                                                                                                                     
[    1.615919] pci 0000:00:00.0: pci_scan_bridge_extend:1269 (e8d:801) PCI_COMMAND = 6                                                                                                                     
[    1.623561] pci 0000:00:00.0: pci_scan_child_bus_extend:2885 (e8d:801) PCI_COMMAND = 6
[    1.631443] pci 0000:00:01.0: pci_scan_child_bus_extend:2883 (e8d:801) PCI_COMMAND = 0
```

```c
	/*
	 * Calculate how many hotplug bridges and normal bridges there
	 * are on this bus. We will distribute the additional available
	 * buses between hotplug bridges.
	 */
	for_each_pci_bridge(dev, bus) {

PCI_DEBUG_COMMAND(dev);
		if (dev->is_hotplug_bridge)
			hotplug_bridges++;
		else
			normal_bridges++;
	}

	/*
	 * Scan bridges that are already configured. We don't touch them
	 * unless they are misconfigured (which will be done in the second
	 * scan below).
	 */
	for_each_pci_bridge(dev, bus) {
		cmax = max;

PCI_DEBUG_COMMAND(dev);
		max = pci_scan_bridge_extend(bus, dev, max, 0, 0);
PCI_DEBUG_COMMAND(dev);
		/*
		 * Reserve one bus for each bridge now to avoid extending
		 * hotplug bridges too much during the second scan below.
		 */
		used_buses++;
		if (cmax - max > 1)
			used_buses += cmax - max - 1;
	}
```

cjdelisle, 2023-04-13:
Well, this is really an unfortunate situation and I'm not sure how I can debug any further. I bisected the location where `0000:00:01.0` loses its `PCI_COMMAND` flags, and there's nothing of consequence in that area.

```
[    1.576252] pci 0000:00:00.0: pci_scan_child_bus_extend:2881 (e8d:801) PCI_COMMAND = 6
[    1.584156] pci 0000:00:01.0: pci_scan_child_bus_extend:2881 (e8d:801) PCI_COMMAND = 6
[    1.592077] pci 0000:00:00.0: pci_scan_child_bus_extend:2897 (e8d:801) PCI_COMMAND = 6
[    1.599960] pci 0000:00:01.0: pci_scan_child_bus_extend:2898 (e8d:801) PCI_COMMAND = 0
```

```c
        /*
         * Calculate how many hotplug bridges and normal bridges there
         * are on this bus. We will distribute the additional available
         * buses between hotplug bridges.
         */
        for_each_pci_bridge(dev, bus) {

PCI_DEBUG_COMMAND(dev); /// <-- here it's ok
inspect_dev=dev;
                if (dev->is_hotplug_bridge)
                        hotplug_bridges++;
                else
                        normal_bridges++;
        }

        /*
         * Scan bridges that are already configured. We don't touch them
         * unless they are misconfigured (which will be done in the second
         * scan below).
         */
        for_each_pci_bridge(dev, bus) {
                cmax = max;

PCI_DEBUG_COMMAND(dev);
PCI_DEBUG_COMMAND(inspect_dev); /// <-- here it's not ok (!!)
                max = pci_scan_bridge_extend(bus, dev, max, 0, 0);
PCI_DEBUG_COMMAND(dev);
PCI_DEBUG_COMMAND(inspect_dev);
                /*
                 * Reserve one bus for each bridge now to avoid extending
                 * hotplug bridges too much during the second scan below.
                 */
                used_buses++;
                if (cmax - max > 1)
                        used_buses += cmax - max - 1;
        }
```

neheb, 2023-04-21:
Maybe @paraka can help

paraka, 2023-04-22:
@cjdelisle  You can check this link [0] to better understanding of the pci topology in mt7621:

> Is PCI slot 1 from pci-mt7621.c the same thing as 0000:00:01.0 ?

This is the pcie bridge 1 corresponding with your pcie 1 slot.

> Is the PCI_COMMAND_MASTER being effectively assigned in pci-mt7621.c ?

No, the command master is not set in the pci controller driver anymore. It was at some time but I was told on code revision when this driver was mainlined that it is strange if PCIe controller driver automatically enables PCI bus mastering, prior device driver initialize itself. The kernel has the function pci_set_master() for this purpose which is used by device drivers. So it is driver responsability to enable that. Device drivers for sure enable PCI_COMMAND_MASTER at the time when it is needed.

However the code was added on kernel v5.10 and the issue was also there for you, right? If you want to try to set this bit in master driver code to see what happens just add:

```
@@ -414,6 +416,11 @@ static void mt7621_pcie_enable_port(struct mt7621_pcie_port *port)
        pcie_port_write(port, PCIE_CLASS_CODE | PCIE_REVISION_ID,
                        RALINK_PCI_CLASS);
 
+       /* set 'PCI_COMMAND_MASTER' */
+       val = read_config(pcie, slot, PCI_COMMAND);
+       val |= PCI_COMMAND_MASTER;
+       write_config(pcie, slot, PCI_COMMAND, val);
+
        /* configure RC FTS number to 250 when it leaves L0s */
        val = read_config(pcie, slot, PCIE_FTS_NUM);
        val &= ~PCIE_FTS_NUM_MASK;
```

\[0\]: https://lore.kernel.org/all/CAMhs-H-BA+KzEwuDPzcmrDPdgJBFA2XdYTBvT4R4MEOUB=WQ1g@mail.gmail.com/t/

cjdelisle, 2023-04-22:
Hello @paraka , thank you for looking at this. My version is 5.10 + any patches from OpenWRT 22.03, I see that in this version, [PCI_COMMAND_MASTER is being explicitly set](https://gist.github.com/cjdelisle/6274d94e48e9cff71de6296f63a684ab#file-pci-mt7621-c-L594).

The real problem of course is that on some reboots, the wifi device is inaccessible. The whole issue with PCI_COMMAND_MASTER is that in these "failed boots", it starts out set and then later becomes unset for no apparent reason.

Since this never happens on a cold boot, my instinct is that the process of cycling the hardware is not perfect and the spontaneous loss of the PCI_COMMAND_MASTER is a symptom.

paraka, 2023-04-22:
@cjdelisle With master branch do you experiment the same behaviour? I mean, the driver is not explicitily setting PCI_COMMAND_MASTER, so I want to know if when you have the hardware working on cold reboots that traces are mostly the same and bits also that in your v5.10 configuration. 
Another clarification to be sure I am understanding you properly, do you have sometimes the card working in warm reboots? Or only cold reboots are working for you?

cjdelisle, 2023-04-22:
1. Master branch of [a week ago](https://github.com/openwrt/openwrt/commit/e04356362ce3bf84dc554cf27bcc6a991e5fa604) -> Yes same behavior, here's a [dmesg](https://gist.github.com/cjdelisle/7700473e07fad5dbe09a2f094e6de688#file-dmesg-5-15-L246) which shows PCI_COMMAND_MASTER is unset at the time of "pci 0000:00:01.0: enabling device"
2. Warm reboots it fails roughly 50% of the time, my experience cold reboots are always ok.

paraka, 2023-04-22:
@cjdelisle It looks that some kind of race condition is happening in your device so as you have said the PCI_COMMAND_MASTER is a symptom. Does it works perfectly with any previous kernel version or even legacy pci controller driver?

cjdelisle, 2023-04-22:
The device seems to work fine with the manufacturer's 4.4.198, but I'm not sure how useful that is because it's likely to be based on a fork.

What version would provide you the most useful information? I can build and test that one.

paraka, 2023-04-22:
@cjdelisle What do you mean with "seems to work fine"? Have you ever tested it? I just want to know if the warm reboot issues persists. Because even the driver is different most of the issues that appeared from the beggining of this long migration process to have a real driver in linux tree should be fixed. I'd like to know, which version of the driver is working for you (a link to the code would be fine) and a working dmesg.

cjdelisle, 2023-04-22:
@paraka The manufacturer's version "seems to work fine" in that I was not able to reproduce the problem by rebooting it. However I don't have the source code to it.

I have not yet tested any version that works, but if you can suggest to me a version which is before the big refactoring, I will try to build that version first.

I am also working on trying to get a build with this kernel: https://github.com/ndmsystems/kernel-49/blob/4.9.273-4/arch/mips/rt2880/pci.c and when I have it working, I'll report whether it's ok.

paraka, 2023-04-22:
@cjdelisle 

This has no refactor at all:
https://elixir.bootlin.com/linux/v4.18.20/source/drivers/staging/mt7621-pci/pci-mt7621.c

Start refactoring but still big legacy stuff:
https://elixir.bootlin.com/linux/v4.19.281/source/drivers/staging/mt7621-pci/pci-mt7621.c
https://elixir.bootlin.com/linux/v4.20.17/source/drivers/staging/mt7621-pci/pci-mt7621.c

Note that all of these version does not need the pci phy stuff since it was all in the same controller driver.

cjdelisle, 2023-05-12:
@paraka Sorry for the delay, we had a few false starts before finding something that worked.

We were able to apply a patch to the kernel which implements the 4.18 reset logic in an arch_init function and this has fixed it for us (10 out of 10 reboots work fine).

```diff
Index: linux-5.10.146/arch/mips/ralink/mt7621.c
===================================================================
--- linux-5.10.146.orig/arch/mips/ralink/mt7621.c
+++ linux-5.10.146/arch/mips/ralink/mt7621.c
@@ -13,6 +13,7 @@
 #include <linux/clk.h>
 #include <linux/clkdev.h>
 #include <linux/clk-provider.h>
+#include <linux/delay.h>
 #include <dt-bindings/clock/mt7621-clk.h>
 
 #include <asm/bootinfo.h>
@@ -323,3 +324,53 @@ void prom_soc_init(struct ralink_soc_inf
 	if (!register_vsmp_smp_ops())
 		return;
 }
+
+// See https://github.com/openwrt/openwrt/issues/12359
+#define RALINK_SYSTEM_CONTROL_BASE	0xbe000000
+#define RALINK_PCIE0_RST		(1<<24)
+#define RALINK_PCIE1_RST		(1<<25)
+#define RALINK_PCIE2_RST		(1<<26)
+#define RALINK_RSTCTRL			*(unsigned int *)(RALINK_SYSTEM_CONTROL_BASE + 0x34)
+#define ASSERT_SYSRST_PCIE(val)		\
+	do {								\
+		if (*(unsigned int *)(0xbe00000c) == 0x00030101)	\
+			RALINK_RSTCTRL |= val;				\
+		else							\
+			RALINK_RSTCTRL &= ~val;				\
+	} while(0)
+#define DEASSERT_SYSRST_PCIE(val)	\
+	do {								\
+		if (*(unsigned int *)(0xbe00000c) == 0x00030101)	\
+			RALINK_RSTCTRL &= ~val;				\
+		else							\
+			RALINK_RSTCTRL |= val;				\
+	} while(0)
+
+static int __init mt7621_pci_reset(void)
+{
+	unsigned long val = 0;
+
+	pr_info("Performing MT7621 PCI hardware reset -->\n");
+
+	ASSERT_SYSRST_PCIE(RALINK_PCIE0_RST | RALINK_PCIE1_RST | RALINK_PCIE2_RST);
+
+	*(unsigned int *)(0xbe000060) &= ~(0x3<<10 | 0x3<<3);
+	*(unsigned int *)(0xbe000060) |= 0x1<<10 | 0x1<<3;
+	mdelay(100);
+	*(unsigned int *)(0xbe000600) |= 0x1<<19 | 0x1<<8 | 0x1<<7; // use GPIO19/GPIO8/GPIO7 (PERST_N/UART_RXD3/UART_TXD3)
+	mdelay(100);
+	*(unsigned int *)(0xbe000620) &= ~(0x1<<19 | 0x1<<8 | 0x1<<7);		// clear DATA
+
+	mdelay(100);
+
+	val = RALINK_PCIE0_RST;
+	val |= RALINK_PCIE1_RST;
+	val |= RALINK_PCIE2_RST;
+
+	DEASSERT_SYSRST_PCIE(val);
+
+	pr_info("<-- MT7621 PCI hardware reset complete\n");
+
+	return 0;
+}
+arch_initcall(mt7621_pci_reset);
```

One difference I notice is that in the old code, all reset controls are asserted (and there's a wait) before the any of the GPIO resets are touched, whereas in the new code, the reset and GPIO resets are called together in sequence.

What do you think ?

cjdelisle, 2023-05-12:
Update: confirming my theory, this patch also works 10 out of 10 times (which previously was unheardof)

```patch
--- a/drivers/staging/mt7621-pci/pci-mt7621.c
+++ b/drivers/staging/mt7621-pci/pci-mt7621.c
@@ -469,7 +469,11 @@ static void mt7621_pcie_reset_assert(str
        list_for_each_entry(port, &pcie->ports, list) {
                /* PCIe RC reset assert */
                mt7621_control_assert(port);
+       }
+
+       mdelay(PERST_DELAY_MS);
 
+       list_for_each_entry(port, &pcie->ports, list) {
                /* PCIe EP reset assert */
                mt7621_rst_gpio_pcie_assert(port);
        }
```

paraka, 2023-05-12:
> Update: confirming my theory, this patch also works 10 out of 10 times (which previously was unheardof)
> 
> ```diff
> --- a/drivers/staging/mt7621-pci/pci-mt7621.c
> +++ b/drivers/staging/mt7621-pci/pci-mt7621.c
> @@ -469,7 +469,11 @@ static void mt7621_pcie_reset_assert(str
>         list_for_each_entry(port, &pcie->ports, list) {
>                 /* PCIe RC reset assert */
>                 mt7621_control_assert(port);
> +       }
> +
> +       mdelay(PERST_DELAY_MS);
>  
> +       list_for_each_entry(port, &pcie->ports, list) {
>                 /* PCIe EP reset assert */
>                 mt7621_rst_gpio_pcie_assert(port);
>         }
> ```

Can you try msleep instead of mdelay and see if everything is ok also?

cjdelisle, 2023-05-12:
I think I will need to update to openwrt trunk because in 22.03 the driver only uses mdelay() anywhere, but we can try this next week with 5.15.

paraka, 2023-05-12:
> I think I will need to update to openwrt trunk because in 22.03 the driver
> only uses mdelay() anywhere, but we can try this next week with 5.15.
Ok. My concern with this changes is that most devices work without them. I
think is better to have more testing with other devices to avoid
regressions before sending this patch to mainline.

cjdelisle, 2023-05-12:
How do you foresee this testing taking place? Should we try offering a patch to OpenWRT and then monitor for any related bug reports? I personally don't have any other MT7621 devices on hand.

dfateyev, 2023-05-12:
There was a big discussion in thread https://github.com/openwrt/openwrt/pull/11220, and the changes similar to [above](https://github.com/openwrt/openwrt/issues/12359#issuecomment-1545814330) have been already [merged](https://github.com/openwrt/openwrt/blob/master/target/linux/ramips/patches-5.15/108-PCI-mt7621-Delay-phy-ports-initialization.patch) to master.
The 22.03 branch doesn't have this patch.
These changes fixed PCIe ports detection for a couple of MT7621 devices, so you are encouraged to test it (either from a master snapshot, or applying the patch to 22.03 tree).

cjdelisle, 2023-05-12:
Thanks @dfateyev , when I tested openwrt master a few weeks ago it did have this patch and on my device it still had the problem so it seems to be a bit different.

paraka, 2023-05-12:
> How do you foresee this testing taking place? Should we try offering a patch to OpenWRT and then monitor for any related bug reports? I personally don't have any other MT7621 devices on hand.

I have three devices to test but I don't have time to test this during this week at all. Maybe I can get this a try next weekend. In any case, if you want to submit the patch to openWRT to get more testing that would be awesome :-)

paraka, 2023-05-12:
> Thanks @dfateyev , when I tested openwrt master a few weeks ago it did have this patch and on my device it still had the problem so it seems to be a bit different.

Yes I think they are different but kind of related though. The legacy driver did not take into account nothing about what ports are up and so on when reset and gpio stuff is performed. It just apply all the reset and gpio reset to all of them.  That was the reason to just apply them at once as you are ding in your patch. New driver is totally device tree related and resets and gpios belong to the ports itself. That is the reason to apply them in the way it is being done. Anyway, if with your patch everything seems to also work I will submit your patch to mainline without any problem. Or if you want to submit it I'll give you my Acked-by as maintainer to get it applied if PCI people is happy with it.

cjdelisle, 2023-05-12:
I would not at all recommend the first patch, I can't imagine a patch bashing registers would be accepted in linux.

The second patch is just adding a pause between the reset asserts and the gpio asserts instead of sending them all at once so I might be wrong but I don't see that creating too much risk.

paraka, 2023-05-12:
> I would not at all recommend the first patch, I can't imagine a patch
> bashing registers would be accepted in linux.
>
> The second patch is just adding a pause between the reset asserts and the
> gpio [...]
I meant the second patch with changes only in the driver. The first one is
a clear NAK for me. Let me test in my devices first and come back to you
after that.

dfateyev, 2023-05-13:
I wonder if simply increasing `INIT_PORTS_DELAY_MS` to 500ms or higher could help in this case.
Alternatively, we probably won't need `"msleep(INIT_PORTS_DELAY_MS)"` anymore after the patch above.
It is all about delays during ports initialization, after all.

gsnoff, 2023-05-16:
> I wonder if simply increasing `INIT_PORTS_DELAY_MS` to 500ms or higher could help in this case. Alternatively, we probably won't need `"msleep(INIT_PORTS_DELAY_MS)"` anymore after the patch above. It is all about delays during ports initialization, after all.

@dfateyev @paraka I’ve tested increasing `INIT_PORTS_DELAY_MS` to 500ms, in a succession of 10 reboots the problem did not reappear. I’ve then tested with 200ms, however this time the Wi-Fi chip apparently didn’t get probed at all after some of the reboots. (The `/sys/bus/pci/devices/0000:02:00.0` corresponding to it was not present at all, neither were any related system log entries.)

Therefore I believe that it would be more optimal to resort to introducing a 100ms sleep between PC and EP reset asserts, as it would reduce overall time spent waiting on PCIe to reset its state. Below is the patch which I added to the trunk version and also tested, no problems encountered in a succession of 10 reboots either.

```diff
Index: linux-5.15.111/drivers/pci/controller/pcie-mt7621.c
===================================================================
--- linux-5.15.111.orig/drivers/pci/controller/pcie-mt7621.c
+++ linux-5.15.111/drivers/pci/controller/pcie-mt7621.c
@@ -59,6 +59,7 @@
 #define PCIE_PORT_CNT			3
 
 #define INIT_PORTS_DELAY_MS		100
+#define RC_EP_DELAY_MS			100
 #define PERST_DELAY_MS			100
 
 /**
@@ -324,7 +325,11 @@ static void mt7621_pcie_reset_assert(str
 	list_for_each_entry(port, &pcie->ports, list) {
 		/* PCIe RC reset assert */
 		mt7621_control_assert(port);
+	}
+
+	msleep(RC_EP_DELAY_MS);
 
+	list_for_each_entry(port, &pcie->ports, list) {
 		/* PCIe EP reset assert */
 		mt7621_rst_gpio_pcie_assert(port);
 	}
```

paraka, 2023-05-16:
> @dfateyev @paraka I’ve tested increasing `INIT_PORTS_DELAY_MS` to 500ms, in a succession of 10 reboots the problem did not reappear. I’ve then tested with 200ms, however this time the Wi-Fi chip [...]

@gsnoff  Thanks for testing this using msleep. I think the new added definition is good to have. As I said, I am a little busy now to properly test this changes in my three boards but I I will try to do it ASAP (hopefully this weekend) and report here.

paraka, 2023-05-16:
@gsnoff BTW, what is your failing device?

cjdelisle, 2023-05-16:
@paraka @gsnoff works with me so he has the same type of device that I do (Comfast CF-EW72 V2)

dfateyev, 2023-05-16:
> @gsnoff Thanks for testing this using msleep. I think the new added definition is good to have. As I said, I am a little busy now to properly test this changes in my three boards but I I will try to [...]

I will check these changes on ZBT WE1326, where we earlier had initialization issues with MT7603E (they were fixed with `INIT_PORTS_DELAY_MS`). Although, I don't  expect any issues with new changes, and will provide a feedback, likely on the weekend.

gsnoff, 2023-05-17:
@dfateyev @paraka FYI I’ve just encountered an occurence of this issue **with** the RC-EP delay patch applied (however on v22.03, not trunk). I believe a more thorough testing might still be needed.

dfateyev, 2023-05-17:
> @dfateyev @paraka FYI I’ve just encountered an occurence of this issue **with** the RC-EP delay patch applied (however on v22.03, not trunk). I believe a more thorough testing might still be needed.

You can try with increasing delay values again. In general, the issue is that the port isn't managed to initialize in the given time. I suspect the issue is gone once we reach ~500ms delay. Also, IMO testing with master version is preferable, it has DTS improvements, etc.

dfateyev, 2025-12-23:
I suppose, it is fixed now starting from v24.10.3.