openwrt/openwrt #12364: ath79: [openwrt-22.03 & openwrt-1907] PC machine will not update IP address when Lan IP address changed

chilinkiot, 2023-04-10:
### Describe the bug

Hi, 
      I have ported my router devices ZR2000 (QCA9531) from ar71xx to ATH79，using release branch openwrt-19.07.   Everything works fine except the lan port.  When I changed the IP address/subnet, such as from 192.168.1.1/24 to 192.168.100.1/24, my PC machine will not detect the change and request new IP address. It still use192.168.1.x, it won't update to 192.168.100.x. I fail to ping 192.168.1.1 and 192.168.100.1, network connection is broken. I need to unplug the cable and plug it again or I need to restart my ZR2000 to restore the network.

       Also, When I plug &unplug the cable from Lan port, no events are triggered.  Sometimes we need to run functions depend on the network events. After port to ATH79, we cannot do that.
       That will not happen when I use target ar71xx.   Everything works fine.

device information: 
CPU: QCA9531
FLASH: 16MB
RAM: 256MB
RJ45: 1xLAN， 1xWAN
1 RS232,
1 PCIE slot for 4G modem
OpenWrt:  openwrt-19.07 or openwrt-22.03

---------------------------------------------
my DTS file for my device as below:
#cat qca9531_ciot-zr2000.dtsi 
// SPDX-License-Identifier: GPL-2.0-or-later OR MIT
/dts-v1/;

#include <dt-bindings/gpio/gpio.h>
#include <dt-bindings/input/input.h>

#include "qca953x.dtsi"

/ {
      compatible = "ciot,zr2000", "qca,qca9531";
       model = "ChilinkIoT ZR2000 Board";

      chosen {
             bootargs = "console=ttyS0,57600n8";
       };

	keys {
		compatible = "gpio-keys-polled";
		poll-interval = <20>;

		button0 {
			label = "reset";
			linux,code = <KEY_RESTART>;
			gpios = <&gpio 17 GPIO_ACTIVE_LOW>;
		};
	};
};

&uart {
	status = "okay";
};

&usb0 {
	status = "okay";
};

&usb_phy {
	status = "okay";
};

&spi {
	status = "okay";

	flash@0 {
		compatible = "jedec,spi-nor";
		reg = <0>;
		spi-max-frequency = <25000000>;

		partitions {
			compatible = "fixed-partitions";
[... 508 lines cut ...]
[   90.469434] eth1: link up (100Mbps/Full duplex)
[   96.707638] eth1: link down
[   98.789528] eth1: link up (100Mbps/Full duplex)
[  127.907651] eth1: link down
[  132.069412] eth1: link up (100Mbps/Full duplex)

### OpenWrt version

r204-e29c8e6c  and r0-8655afc0

### OpenWrt target/subtarget

ath79/generic

### Device

ChilinkIoT ZR2000 Board

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-04-10:
Obviously that is how your PC machine reacts to cable down/up. Normally one reduces TTL-s for major network renumbering and maintains both configs instrumented on same wire. e.g keep old router without dhcp until whole network refreshes DHCP to new network.

chilinkiot, 2023-04-24:
> Obviously that is how your PC machine reacts to cable down/up. Normally one reduces TTL-s for major network renumbering and maintains both configs instrumented on same wire. e.g keep old router without dhcp until whole network refreshes DHCP to new network. You perceive normal operation of network as sort of problem. Its not, nothing to do with ath79, network cables dont beam IP addresses without client requesting them.

This has nothing to do with my PC. Router doesn't change the interface state. PC cannot detect any changes. I have reverted my project to target AR71xx, the network interface works fine.   The differences of the build config are the target and device profile, such as below:
CONFIG_TARGET_ar71xx=y
CONFIG_TARGET_ar71xx_generic=y
CONFIG_TARGET_DEVICE_ar71xx_generic_DEVICE_ciot-zr2000=y
-------------------------------------
CONFIG_TARGET_ath79=y
CONFIG_TARGET_ath79_generic=y
CONFIG_TARGET_DEVICE_ath79_generic_DEVICE_ciot-zr2000=y
---------------------------------------------

brada4, 2023-04-24:
I wrongly presumed you restarted router for renumber. Cable is not touched changing IP.
 There is no way DHCP server actively signals change to clients, it has to wait few TTLs with old routing configuration working until devices poll at (1/2-)TTL intervals and adapt new realm.
It is not problem with your porting, or OpenWRT in general. You just observe how networks are supposed to work.