openwrt/openwrt #12366: IGMP stopped working after commit that updates kernel to 5.15.86 on Turris Omnia

PedroM4, 2023-04-10:
### Describe the bug

IPTV channels stopped working after updating my Turris Omnia. I narrowed down the start of the issue to commit:

https://github.com/openwrt/openwrt/commit/738b04c88128d9963b81bff68426980f587f22d9

### OpenWrt version

738b04c88128d9963b81bff68426980f587f22d9

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Turris Omnia

### Image kind

Official downloaded image

### Steps to reproduce

update router firmware with a version that is above the commit that introduces the issue ( including newer kernel versions that that one ) keeping configuration.
IPTV channels stop working ( omcproxy, but same happens with igmproxy ).
Flash a version prior to that commit and IPTV channels work.

### Actual behaviour

IPTV channels stop working ( omcproxy, but same happens with igmproxy ).
Flash a version prior to that commit and IPTV channels work.

### Expected behaviour

IPTV channels should work

### Additional info

maybe these Omnia specific changes of kernel .86 may have something to do with it:
Pali Rohár (9):
ARM: dts: dove: Fix assigned-addresses for every PCIe Root Port
ARM: dts: armada-370: Fix assigned-addresses for every PCIe Root Port
ARM: dts: armada-xp: Fix assigned-addresses for every PCIe Root Port
ARM: dts: armada-375: Fix assigned-addresses for every PCIe Root Port
ARM: dts: armada-38x: Fix assigned-addresses for every PCIe Root Port
ARM: dts: armada-39x: Fix assigned-addresses for every PCIe Root Port
ARM: dts: turris-omnia: Add ethernet aliases
ARM: dts: turris-omnia: Add switch port 6 node
arm64: dts: armada-3720-turris-mox: Add missing interrupt for RTC

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

stklcode, 2023-07-08:
First, I have no idea about the actual issue and don't even have any testcases for that.

But what comes to my mind is, if you already narrowed it down to a small set of potentially breaking changes:

> maybe these Omnia specific changes of kernel .86 may have something to do with it:
> ARM: dts: armada-38x: Fix assigned-addresses for every PCIe Root Port (https://github.com/gregkh/linux/commit/f27dd04e44d82a96e0e7be4d5fb48c84940a7d6e)
> ARM: dts: turris-omnia: Add ethernet aliases (https://github.com/gregkh/linux/commit/b311f8e9f50395173adcd3919ef24f82c7af58f2)
> ARM: dts: turris-omnia: Add switch port 6 node (https://github.com/gregkh/linux/commit/820a5ccca7c30bf0df949e6e7e5b8fdeb4bfb753)

Have you tried to simply revert the changes in a custom 5.15.y build?

PedroM4, 2023-07-08:
man, many thanks for the walk trough. I hinted at those changes here:
https://github.com/openwrt/openwrt/commit/738b04c88128d9963b81bff68426980f587f22d9
but I had no idea on how to revert them to test.
Now I know.

Will try and report back. 

Don't know if this helps you understand it or not, but the output of the ip mroute command on 5.15.85 ( the last working kernel ) is:
root@gateway:~# ip mroute
(192.168.89.100,239.255.255.250) Iif: lan3        State: resolved
(188.81.239.68,232.0.6.6)        Iif: lan4       Oifs: lan3  State: resolved
(188.81.239.68,232.32.0.33)      Iif: lan4       Oifs: lan3  State: resolved
(10.159.224.200,232.32.0.33)     Iif: lan4       Oifs: lan3  State: resolved

the same command on later kernels:
(192.168.89.100,239.255.255.250) Iif: lan3        State: resolved

lan3 is downlink ( the interface where the IPTV box connects to ) lan4 is uplink ( the interface with the public ip address )
so in later kernels it appears the igmp packets coming into lan4 are not processed.

PedroM4, 2023-07-08:
@stklcode I'm getting this:

`Applying /home/finas/OpenWRT.Omnia/target/linux/mvebu/patches-5.15/920-Revert-ARM-dts-armada-38x-Fix-assigned-addresses-for.patch using plaintext: 
patching file arch/arm/boot/dts/armada-385.dtsi
Hunk #1 FAILED at 84.
Hunk #2 FAILED at 103.
Hunk #3 FAILED at 125.
3 out of 3 hunks FAILED -- saving rejects to file arch/arm/boot/dts/armada-385.dtsi.rej
Patch failed!  Please fix /home/finas/OpenWRT.Omnia/target/linux/mvebu/patches-5.15/920-Revert-ARM-dts-armada-38x-Fix-assigned-addresses-for.patch!
make[4]: *** [Makefile:28: /home/finas/OpenWRT.Omnia/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/linux-5.15.120/.prepared_12589e530283135d0889e499f7099ddd] Error 1
make[4]: Leaving directory '/home/finas/OpenWRT.Omnia/target/linux/mvebu'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/finas/OpenWRT.Omnia/target/linux'
time: target/linux/compile#6.45#3.62#7.67
    ERROR: target/linux failed to build.
`

I placed the three files into the patches-5.15 dir of current master. 
Maybe I should try to pull to the 5.15.86 commit and try again, instead of current master?

stklcode, 2023-07-08:
Either that or try `make target/linux/refresh` to update the patches automatically - if possible.

PedroM4, 2023-07-08:
ok, so I tried the first option checking out to the 5.15.86 commit with:
git checkout 738b04c88128d9963b81bff68426980f587f22d9

putting the three patch files in place and compiling and same error. first patch fails in all three chunks.

PedroM4, 2023-07-08:
same thing with the make target/linux/refresh

`atching file arch/arm/boot/dts/armada-385.dtsi
Hunk #1 FAILED at 84.
Hunk #2 FAILED at 103.
Hunk #3 FAILED at 125.
3 out of 3 hunks FAILED -- rejects in file arch/arm/boot/dts/armada-385.dtsi
Patch platform/920-Revert-ARM-dts-armada-38x-Fix-assigned-addresses-for.patch does not apply (enforce with -f)
make[3]: *** [Makefile:18: /home/finas/OpenWRT.Omnia.5/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/linux-5.15.86/.quilt_checked] Error 1
make[3]: Leaving directory '/home/finas/OpenWRT.Omnia.5/target/linux/mvebu'
make[2]: *** [Makefile:11: refresh] Error 2
make[2]: Leaving directory '/home/finas/OpenWRT.Omnia.5/target/linux'
time: target/linux/refresh#0.08#0.00#0.09
    ERROR: target/linux failed to build.
make[1]: *** [target/Makefile:30: target/linux/refresh] Error 1
make[1]: Leaving directory '/home/finas/OpenWRT.Omnia.5'
make: *** [/home/finas/OpenWRT.Omnia.5/include/toplevel.mk:231: target/linux/refresh] Error 2
`

PedroM4, 2023-07-09:
@stklcode 
I was able to advance on this. kernel 6.1.38 and 5.15.120 now work.
I wasn't able to apply the revert patches that you supplied but I was able to edit the kernel files and change them manually. 

there are two issues:

- First issue:

This patch ( https://github.com/gregkh/linux/commit/f27dd04e44d82a96e0e7be4d5fb48c84940a7d6e ) seems to not be in use on kernel 5.15.86 or above ( including 6.1.38 ).
how do I know?
these files:
./build_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-13.1.0_musl_eabi/linux-5.15.120/arch/arm/boot/dts/armada-385.dtsi
./build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/linux-6.1.38/arch/arm/boot/dts/armada-385.dtsi
have the old values that patch ARM-dts-armada-38x-Fix-assigned-addresses-for should have fixed.
So if you check your own build you will see that the values do not correspond to the supposedly new values.

- Second issue:
 patch ARM-dts-turris-omnia-Add-switch-port-6-node needs to be removed.
 I manually edited the files:
./build_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-13.1.0_musl_eabi/linux-5.15.120/arch/arm/boot/dts/armada-385-turris-omnia.dts
./build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/linux-6.1.38/arch/arm/boot/dts/armada-385-turris-omnia.dts
and removed :
`ports@6 {
				reg = <6>;
				label = "cpu";
				ethernet = <&eth0>;
				phy-mode = "rgmii-id";

				fixed-link {
					speed = <1000>;
					full-duplex;
				};
			};`

compiled and now it works perfectly

`BusyBox v1.36.1 (2023-07-07 15:38:38 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r23534-868117cb8b
 -----------------------------------------------------
root@gateway:~# uname -a
Linux gateway 6.1.38 #0 SMP Sun Jun 18 15:06:57 2023 armv7l GNU/Linux
root@gateway:~# ip mroute
(10.173.7.91,232.0.6.6)          Iif: lan4       Oifs: lan3  State: resolved
(10.173.7.89,232.0.6.6)          Iif: lan4       Oifs: lan3  State: resolved
(192.168.89.100,239.255.255.250) Iif: lan3        State: resolved
root@gateway:~# 
`

stklcode, 2023-07-09:
> putting the three patch files in place and compiling and same error. first patch fails in all three chunks.

Sorry for the wrong patches. Seems something got messed up here like spaces vs. tabs. Updated my previous comment. That's why you should never copy&paste patches from the terminal to a browser :see_no_evil: 

> First issue: This patch ( https://github.com/gregkh/linux/commit/f27dd04e44d82a96e0e7be4d5fb48c84940a7d6e ) seems to not be in use on kernel 5.15.86 or above ( including 6.1.38 ).
> [....]
> So if you check your own build you will see that the values do not correspond to the supposedly new values.

The _armada-385.dtsi_ looks fine for me with unchanged patches
```c
                        pcie1: pcie@1,0 {
                                device_type = "pci";
                                assigned-addresses = <0x82000800 0 0x80000 0 0x2000>;
/* ... */
                        pcie2: pcie@2,0 {
                                device_type = "pci";
                                assigned-addresses = <0x82001000 0 0x40000 0 0x2000>;
/* ... */
                        pcie3: pcie@3,0 {
                                device_type = "pci";
                                assigned-addresses = <0x82001800 0 0x44000 0 0x2000>; 
/* ... */
                        pcie4: pcie@4,0 {
                                device_type = "pci";
                                assigned-addresses = <0x82002000 0 0x48000 0 0x2000>;
```

> Second issue: patch ARM-dts-turris-omnia-Add-switch-port-6-node needs to be removed.

Interesting :thinking: 

I don't yet have an idea why this should cause such issue, but that's definitely something we should be able to work with. Maybe some old patch of ours that tried to work around the same issues before 5.15 and now messes up with the upstream changes. But at least there's no obvious one...

I'm pretty busy the next days and don't have any test devices around most times. If nobody else chimes in, I could try to get into it by the end of the week.

PedroM4, 2023-07-09:
@stklcode 
you are right about the  armada-385.dtsi being ok. I guess being up until 6am trying to get this to work didn't help. Sorry!

So I experimented with the fixed revert patches and the only one needed to fix the issue is the 922 one.

One interesting thing is that OpenWRT Omnia page states that port6 is connected to eth1 and not eth0. Not sure if this is outdates, wrong or correct.
https://openwrt.org/toh/turris/turris_omnia

The file armada-385-turris-omnia.dts codes it as being connected to eth0

`/* Connected to 88E6176 switch, port 6 */
&eth0 {
        pinctrl-names = "default";
        pinctrl-0 = <&ge0_rgmii_pins>;
        status = "okay";
        phy-mode = "rgmii";
        buffer-manager = <&bm>;
        bm,pool-long = <0>;
        bm,pool-short = <3>;

        fixed-link {
                speed = <1000>;
                full-duplex;
        };
};

/* Connected to 88E6176 switch, port 5 */
&eth1 {
        pinctrl-names = "default";
        pinctrl-0 = <&ge1_rgmii_pins>;
        status = "okay";
        phy-mode = "rgmii";
        buffer-manager = <&bm>;
        bm,pool-long = <1>;
        bm,pool-short = <3>;

        fixed-link {
                speed = <1000>;
                full-duplex;
        };
};
`

fmfranz, 2023-07-14:
Would be great if this made it into final 23.05. release

PedroM4, 2023-07-21:
I found another interesting thing that may or not be related:

Running ethtool -S eth0 ( or eth1 or eth2 ) will crash the router and automatically reboot it. ethtool -S lanX works fine.

PedroM4, 2023-07-31:
@stklcode I'm curious to know if you reached any conclusion? Actually, did you have time to look at it?

aeri, 2023-09-01:
This could be the same problem as #9115

PedroM4, 2023-09-01:
not sure. this only started happening to me with kernel 5.15.86 and issue is solved when removing the ARM-dts-turris-omnia-Add-switch-port-6-node patch that introduced the issue on that kernel revision.
@stklcode is looking into the issue.

ib54003, 2023-10-14:
@stklcode any update?

stklcode, 2023-10-14:
I'm not actually working on this, so no update from my side.

I had a quick look into the problem couple of weeks ago without any concrete results. And as written before, I do not have any testcases for the actual issue, wo "working" on it blindly is not exactly motivating.

Probably better to get in touch with the Turris team and/or match our kernel patches with theirs.

stklcode, 2025-06-04:
Looks like it. And yes, typically such fixed find their way upstream, either to OpenWRT or to Linux.

Given a modified DTB fixed the issue, sounds like the (upstream kernel) device tree needs some change, as already discussed above.

I couldn't find any obvious hints on that on the Turris repo nor on the LKML.
There's one upstream change to the DTS removing the "cpu" labels from ports 5 and 6 (6 was introduced in a patch named before)

https://github.com/torvalds/linux/commit/0939de7d89d2efc4f565ce1961b4020787e9f297
```diff
--- a/arch/arm/boot/dts/armada-385-turris-omnia.dts
+++b/arch/arm/boot/dts/armada-385-turris-omnia.dts
@@ -479,7 +479,6 @@
 			ports@5 {
 				reg = <5>;
-				label = "cpu";
 				ethernet = <&eth1>;
 				phy-mode = "rgmii-id";
@@ -491,7 +490,6 @@
 			ports@6 {
 				reg = <6>;
-				label = "cpu";
				ethernet = <&eth0>;
				phy-mode = "rgmii-id";
```

As stated before, reverting 2 patches seems to fix the issue:
https://github.com/openwrt/openwrt/issues/12366#issuecomment-1627604755

If the quoted patch (Kernel 6.5) does the trick - though I don't see why it should - , the issue should have been gone in 24.10 (Kernel 6.6), if not, there has to be something else, so either revert it or fix forward - no idea which solution was discovered by the Turris team.

alexojegu, 2025-06-19:
I am on version 24.10.1 and if I enable IGMP Snooping it stops working DHCPv6 (https://forum.openwrt.org/t/ipv6-fail-on-wired-network-and-work-on-wireless-network/232480) so it seems that this patch did not solve it.

I have no knowledge about DTS, but searching the TurrisOS repository for patches with a recent date I can't find anything that might seem related.

alexojegu, 2025-06-19:
I am trying to find the patches https://github.com/openwrt/openwrt/issues/12366#issuecomment-1627604755 in the `master` or `openwrt-24.10` branches but I cannot. If I am not mistaken the path would be: https://github.com/openwrt/openwrt/tree/master/target/linux/mvebu/patches-6.6