openwrt/openwrt #12369: Build error on macbook M2

adonespitogo, 2023-04-11:
### Describe the bug

```
FAILED: CMakeFiles/json-c.dir/json_util.c.o                                                                                                                                                                     /Volumes/OpenWrt/openwrt/staging_dir/host/bin/gcc -D_GNU_SOURCE -I/Volumes/OpenWrt/openwrt/build_dir/hostpkg/json-c-0.15 -O2 -I/Volumes/OpenWrt/openwrt/staging_dir/host/include -I/Volumes/OpenWrt/openwrt/stag
ing_dir/hostpkg/include -I/Volumes/OpenWrt/openwrt/staging_dir/target-aarch64_cortex-a53_musl/host/include -ffunction-sections -fdata-sections -Werror -Wall -Wcast-qual -Wno-error=deprecated-declarations -Wextra -Wwrite-strings -Wno-unused-parameter -Wstrict-prototypes -DNDEBUG -O2 -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.3.sdk -fPIC -D_REENTRANT -MD -MT CMakeFiles/json-c.dir/json_u
til.c.o -MF CMakeFiles/json-c.dir/json_util.c.o.d -o CMakeFiles/json-c.dir/json_util.c.o -c /Volumes/OpenWrt/openwrt/build_dir/hostpkg/json-c-0.15/json_util.c                                                  
/Volumes/OpenWrt/openwrt/build_dir/hostpkg/json-c-0.15/json_util.c:64:35: error: a function declaration without a prototype is deprecated in all versions of C [-Werror,-Wstrict-prototypes]                    
const char *json_util_get_last_err() 
```

### OpenWrt version

v22.03.4

### OpenWrt target/subtarget

I am not able to run openwrt on the target yet.

### Device

Apple MBA M2

### Image kind

Self-built image

### Steps to reproduce

Follow instructions in arm apple m1 arm64:

https://openwrt.org/docs/guide-developer/toolchain/buildroot.exigence.macosx

### Actual behaviour

```
FAILED: CMakeFiles/json-c.dir/json_util.c.o                                                                                                                                                                     /Volumes/OpenWrt/openwrt/staging_dir/host/bin/gcc -D_GNU_SOURCE -I/Volumes/OpenWrt/openwrt/build_dir/hostpkg/json-c-0.15 -O2 -I/Volumes/OpenWrt/openwrt/staging_dir/host/include -I/Volumes/OpenWrt/openwrt/stag
ing_dir/hostpkg/include -I/Volumes/OpenWrt/openwrt/staging_dir/target-aarch64_cortex-a53_musl/host/include -ffunction-sections -fdata-sections -Werror -Wall -Wcast-qual -Wno-error=deprecated-declarations -Wex
tra -Wwrite-strings -Wno-unused-parameter -Wstrict-prototypes -DNDEBUG -O2 -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.3.sdk -fPIC -D_REENTRANT -MD -MT CMakeFiles/json-c.dir/json_u
til.c.o -MF CMakeFiles/json-c.dir/json_util.c.o.d -o CMakeFiles/json-c.dir/json_util.c.o -c /Volumes/OpenWrt/openwrt/build_dir/hostpkg/json-c-0.15/json_util.c                                                  
/Volumes/OpenWrt/openwrt/build_dir/hostpkg/json-c-0.15/json_util.c:64:35: error: a function declaration without a prototype is deprecated in all versions of C [-Werror,-Wstrict-prototypes]                    
const char *json_util_get_last_err()
```

### Expected behaviour

Should build successfully.

### Diffconfig

```text
╰─ ./scripts/diffconfig.sh
CONFIG_TARGET_armvirt=y
CONFIG_TARGET_armvirt_64=y
CONFIG_TARGET_armvirt_64_Default=y
# CONFIG_FEED_luci is not set
# CONFIG_FEED_packages is not set
# CONFIG_FEED_routing is not set
# CONFIG_FEED_telephony is not set
CONFIG_PACKAGE_make=y
# CONFIG_TARGET_IMAGES_GZIP is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

stephensworld, 2023-05-10:
I have the same problem with Macbook M1 Max. Is there already a solution?
Thanks
Stephan

adonespitogo, 2023-05-11:
No solution. Im trashing my M2 for an x86 machine.

dj9pz, 2023-07-28:
The workaround is to manually edit line 64 of `build_dir/hostpkg/json-c-0.15/json_util.c` by adding `void` into the empty parentheses:
`const char *json_util_get_last_err(void)`
and continue building with `make world`.

This is a known problem in upstream json-c and the fix has not yet been issued as part of a new release: [https://github.com/json-c/json-c/issues/818](https://github.com/json-c/json-c/issues/818)

dj9pz, 2023-07-30:
> The workaround is to manually edit line 64 of `build_dir/hostpkg/json-c-0.15/json_util.c` by adding `void` into the empty parentheses: `const char *json_util_get_last_err(void)` and continue building with `make world`.
> 
> This is a known problem in upstream json-c and the fix has not yet been issued as part of a new release: [json-c/json-c#818](https://github.com/json-c/json-c/issues/818)

Update: I've built multiple openwrt images (wr710n with 16MB flash, wr703n with 16MB flash) on my Mac mini M1, both builds succeeded, installed without any problems and the devices are running fine.