openwrt/openwrt #12378: Sysupgrade 22.03.4 on Linksys EA8300 fails

hhasert, 2023-04-12:
### Describe the bug

Doing the standard sysupgrade on 22.03.3 to 22.03.4 (keeping existing config) and waiting for the reboot. The system does not reboot in the sense that I see the linksys logo blinking and afterwards no Wifi anymore.

### OpenWrt version

22.0.3.3

### OpenWrt target/subtarget

22.03.4

### Device

Linksys EA8300

### Image kind

Official downloaded image

### Steps to reproduce

- Download https://downloads.openwrt.org/releases/22.03.4/targets/ipq40xx/generic/openwrt-22.03.4-ipq40xx-generic-linksys_ea8300-squashfs-sysupgrade.bin 
- Shorten the name  and select the flash file in Luci
- Upload it
- wait for reboot 

### Actual behaviour

After reboot, no Wifi (don't have laptop with ethernet so I could not check is I get DHCP

### Expected behaviour

Like always just reboot and maintain settings

### Additional info

Tried twice not a lot of time to experiment and my laptop has no ethernet. 

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

hhasert, 2023-04-12:
I tried the sysupgrade on my TP-Link Archer C7 V2 and that worked perfectly, so I wonder why it breaks on Linksys EA8300

saber716rus, 2023-04-12:
on the asus ax53u, the update went fine.

hhasert, 2023-04-13:
It might be something with the existing config being maintained. It is my main router so experimenting will be a bit hard without the family nagging about the internet being down. I now reboot it with the reset switch on and it then resorts to 22.03.3 and works again with the existing config, so either it triggers the second partition or something during the upgrade itself fails.

hhasert, 2023-04-13:
Could this be related to https://github.com/openwrt/openwrt/issues/12339 ? Seems like It just does not boot at all. Unfortunately I have no serial interface on it so I cannot check.

badulesia, 2023-04-15:
Hi.
Yes it is the very same bug than https://github.com/openwrt/openwrt/issues/12339.
I have tested 22.03.4 on a MR8300 and as expected it fails.
A patch is on its way.
https://github.com/openwrt/openwrt/commit/aad34818b50029e07ed9221ae46f9770d6e29785
You can consider this bug report closed.

hhasert, 2023-04-16:
Any idea if there is going to be a replacement for EA8300 for 22.03.4 ?

badulesia, 2023-04-16:
Unlikely, each time I saw an error/issue with a release, we had to wait for the next one.

hhasert, 2023-04-16:
well I will look at the snapshots then 👍

https://downloads.openwrt.org/releases/22.03-SNAPSHOT/targets/ipq40xx/generic/

One of these should include the bugfix.

Djfe, 2023-05-11:
Please close this issue
22.03.05 was released and addressed this issue :)