openwrt/openwrt #12379: the problem with writing files is due to the lack of necessary user rights.

saber716rus, 2023-04-12:
### Describe the bug

the essence of the problem is that the disk is mounted with root rights, but because of this there is a problem with the samba and transmission packages, more precisely because of the lack of necessary rights to write files on the mounted disk.

### OpenWrt version

r20123-38ccc47687

### OpenWrt target/subtarget

ramips/mt7621

### Device

ASUS RT-AX53U

### Image kind

Official downloaded image

### Steps to reproduce

Samba: just write a file on a remote device
Transmission: file upload

### Actual behaviour

Permission denied

### Expected behaviour

writing files to a mounted disk

### Additional info

delov is in the rights, since he created a folder on the disk with access rights for the transmission user, then the download was successful.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

saber716rus, 2023-04-13:
Transmission and samba packages are installed on openwrt, but they have a common problem - writing to disk due to the lack of the necessary rights.

Disk /dev/sda: 7.68 GiB, 8242855936 bytes, 16099328 sectors
Disk model: Flash Disk
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 2E592ACE-9E2F-1946-97E1-B04C6A046006
Device Start End Sectors Size Type
/dev/sda1 2048 16095231 16093184 7.7G Linux filesystem
root@OpenWrt:~# df -h
Filesystem Size Used Available Use% Mounted on
/dev/root 4.3M 4.3M 0 100% /rom
tmpfs 122.3M 3.4M 118.9M 3% /tmp
/dev/ubi0_1 32.9M 23.4M 7.8M 75% /overlay
overlayfs:/overlay 32.9M 23.4M 7.8M 75% /
tmpfs 512.0K 0 512.0K 0% /dev
/dev/sda1 7.7G 3.5M 7.2G 0% /mnt/sda1
root@OpenWrt:~#
/dev/sda1: UUID="a1b86d1d-2fc8-4048-9bb5-c52a74bc5097" UUID_SUB="5234f58b-508f-4c3d-bdb3-e71a649ffc8a" BLOCK_SIZE="4096" TYPE="btrfs" PARTUUID="0dcec60c-6d46-b943-b850-a74ba5c94915"
root@OpenWrt:~# cat /etc/config/fstab
config global
option anon_swap '0'
option anon_mount '0'
option auto_swap '1'
option auto_mount '1'
option delay_root '5'
option check_fs '0'
config mount
option target '/mnt/sda1'
option uuid 'a1b86d1d-2fc8-4048-9bb5-c52a74bc5097'
option enabled '1'
option options 'relatime'
root@OpenWrt:~#

brada4, 2023-04-14:
You are supposed to `chown` directories you share

saber716rus, 2023-04-14:
> You are supposed to `chown` directories you share

the only temporary solution is to create a folder on the disk and change the user's rights to the folder.

brada4, 2023-04-14:
It is same on every Linux. Permissions stay over reboots on filesystems, even on root of mount point.