openwrt/openwrt #12389: losetup missing from `openwrt-ipq807x-generic-qnap_301w-initramfs-uImage.itb` even though it is used by sysupgrade stage2 script

chrysn, 2023-04-14:
### Describe the bug

It appears that the environment in which sysupgrades are executed from the .itb files depends on losetup, but no losetup binary is present, resulting in failed attempts to install OpenWRT.

### OpenWrt version

r22565-877ec78e23

### OpenWrt target/subtarget

ipq807x/generic

### Device

QNAP 301w

### Image kind

Official downloaded image

### Steps to reproduce

Following the installation instructions for the Qhora 301W (from [its commit message](https://github.com/openwrt/openwrt/commit/bd17683261e1dde1fdd4223745852de1ce8cfc89)), I took a serial console, booted into the .itp image and attempted a `sysupgrade -n openwrt-ipq807x-generic-qnap_301w-squashfs-sysupgrade.bin`, which failed:

```
Thu Apr 13 00:31:02 UTC 2023 upgrade: Switching to ramdisk...
Thu Apr 13 00:31:03 UTC 2023 upgrade: Performing system upgrade...
/lib/upgrade/do_stage2: line 38: losetup: not found
Failed to detach all loop devices. Skip this try.
```

Digging down, I disabled the watchdog (not sure that was necessary, but there was enough scary chatter on the console that I rather relied on my fingers on the power button), and found that as part of the sysupgrade, `losetup --detach-all` was executed in /lib/upgrade/mmc.sh (with errors triggering an immediate reboot) -- but no `losetup` is present, neither in the runtime the .itp boots into, nor in the RAMFS that the sysupgrade changes into (of which I'm not sure whether it's the same environment, a different one or a derived one).

After a sloppy check for whether there's anything loopback-ish around (didn't look like it), I commented out that check, and subsequently flashing succeeded to the point where the machine rebooted into an on-disk OpenWRT.

There were still some discouraging lines in the console output:

```
Thu Apr 13 00:12:36 UTC 2023 upgrade: Switching to ramdisk...
Thu Apr 13 00:12:37 UTC 2023 upgrade: Performing system upgrade...
flashing kernel to /dev/mmcblk0p1
flashing rootfs to /dev/mmcblk0p4
/lib/upgrade/do_stage2: line 62: losetup: not found
/lib/upgrade/do_stage2: line 63: losetup: not found
Failed to mount looped rootfs_data.
```

I can't easily judge yet whether the installation was complete, but either way, I'm in an OpenWRT now.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

kirdesde, 2023-04-15:
This is due to a limitiation of the OpenWrt build system. If building initramfs for multiple devices (this is what the build robots do), the DEVICE_PACKAGE are ignored. 

https://forum.openwrt.org/t/adding-openwrt-support-for-qnap-qhora-301w/96934/662?u=kirdes

Only solutions would be to add the device specific packages as target default packages, but that would increase the image size for all target devices.

kirdesde, 2023-04-21:
> That's not the only solution, you can just comment out the losetup stuff. It'll work fine. So the script can be repurposed to where if losetup isn't there you just proceed without running it.

Not really, losetup is needed to setup an ext4 loop device for the overlay filesystem (used for data persistence), standard OpenWrt way to do that on eMMC devices)

If there's no loop device, OpenWrt fstools may use mmcblk0p9 (rootfs_data) as rootfs overlay, but that is a stock gpt partition and OpenWrt would write directly on the eMMC (bad idea, flash has limited write cycles)

EvanCarroll, 2023-04-21:
It worked fine for me. Maybe it's needed on subsequent updates when you actually have data you want to persist, but on the upgrade of initramfs, to the first squashfs, if there is no losetup and you comment out those lines what negative thing/sideeffect happens? Why can't we infer that if there is no losetup we can just ignore those commands and assume there is no data to persist? Is this really an issue? The reason why losetup isn't included everywhere is because other device builds for the same arch don't need it, right? it's not universally needed? is that a determination made by the port creator here or a policy decision based on this specific device's eMMC? Why is the optimization for eMMC only a factor affecting the install experience on the 301w?

kirdesde, 2023-04-21:
Loop device is only needed on eMMC based devices (QNAP and Zyxel in the ipq807x target)

All other ipq807x devices are NAND based, and on NAND we are using UBI (that provides 3 volumes, incl. a rootfs_data volume)

And those NAND devices have limited space.

It's not someone's decision but depends on the device flash technology.

EvanCarroll, 2023-04-21:
That's really good insight there. Thanks for the clarifying note as to _why_ the losetup is required. I wonder how many write cycles a modern eMMC chip gets.

Firefishy, 2023-04-30:
Workaround until `openwrt-ipq807x-generic-qnap_301w-initramfs-uImage.itb` is fixed with losetup:

```
# Step 1 manually setup IP on local network, replace 192.168.x.y as applicable"
ifconfig br-lan 192.168.x.y
route add default gw 192.168.x.1
echo nameserver 8.8.8.8 >/etc/resolv.conf
# Step 2 set correct date and time
date "2023-04-30 00:01"
# Step 3 update packages and install losetup
opkg update
opkg install losetup
# Step 4 continue sysupgrade as per normal
sysupgrade -n /tmp/openwrt-ipq807x-generic-qnap_301w-squashfs-sysupgrade.bin
```

robimarko, 2023-09-02:
This was fixed by:
https://github.com/openwrt/openwrt/commit/07fe8bc62a866e78e131c3f63a08554a94e931f8