openwrt/openwrt #12392: mt7620 missing builds

George-TL, 2023-04-14:
### Describe the bug

I'm unable to upgrade from '21.02.5' to '21.02.6'.
6 folders:
https://downloads.openwrt.org/releases/21.02.5/targets/ramips/
https://downloads.openwrt.org/releases/22.03.4/targets/ramips/
only 3 folders:
https://downloads.openwrt.org/releases/21.02.6/targets/ramips/

### OpenWrt version

r16688-fa9a932fdb

### OpenWrt target/subtarget

ramips/mt7620

### Device

TP-Link Archer C2 v1

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

lukasz1992, 2023-04-14:
Build is not yet finished, wait for official announcement of 21.02.6

brada4, 2023-04-14:
You can get point release with all packages included via modern firmware selector and its client counterpart "attended sysupgrade" , kind of zero maintenance burden in the end.

George-TL, 2023-04-20:
All missing folders appeared. Thank you.

Djfe, 2023-04-25:
if this is fixed, you may close this issue :)