openwrt/openwrt #12393: Netifd Segmentation Error when Adding Large Number of Interfaces

shripadrayewar, 2023-04-14:
### Describe the bug

Hi
I am encountering an issue with OpenWRT's netifd when attempting to add 40 or more interfaces to my network configuration file. Upon adding the interfaces, netifd gives a segmentation error in dmesg, and the interfaces do not show up with the ip a command. However, the interface configurations are present in the network configuration file.
Follwing is the error im getting while network restart on dmesg

`[ 7400.201808] netifd[22825]: segfault at a0 ip 08057fdc sp bfe16fd0 error 4 in netifd[804b000+1c000]
[ 7400.212899] Code: 8d 65 f4 5b 5e 5f 5d e9 f7 fd ff ff 8d 65 f4 5b 5e 5f 5d c3 80 78 0c 00 0f 84 98 00 00 00 55 89 e5 53 51 8b 58 1c c6 40 0c 00 <8b> 83 a0 00 00 00 48 89 83 a0 00 00 00 f6 05 ac 38 07 08 02 74 2c
[ 7407.152861] netifd[23845]: segfault at a0 ip 08057fdc sp bfb553e0 error 4 in netifd[804b000+1c000]
[ 7407.163195] Code: 8d 65 f4 5b 5e 5f 5d e9 f7 fd ff ff 8d 65 f4 5b 5e 5f 5d c3 80 78 0c 00 0f 84 98 00 00 00 55 89 e5 53 51 8b 58 1c c6 40 0c 00 <8b> 83 a0 00 00 00 48 89 83 a0 00 00 00 f6 05 ac 38 07 08 02 74 2c
[ 7414.155714] netifd[25731]: segfault at a0 ip 08057fdc sp bfba09e0 error 4 in netifd[804b000+1c000]
[ 7414.165320] Code: 8d 65 f4 5b 5e 5f 5d e9 f7 fd ff ff 8d 65 f4 5b 5e 5f 5d c3 80 78 0c 00 0f 84 98 00 00 00 55 89 e5 53 51 8b 58 1c c6 40 0c 00 <8b> 83 a0 00 00 00 48 89 83 a0 00 00 00 f6 05 ac 38 07 08 02 74 2c
[ 7421.032886] netifd[26965]: segfault at a0 ip 08057fdc sp bfa6cd90 error 4 in netifd[804b000+1c000]
[ 7421.043189] Code: 8d 65 f4 5b 5e 5f 5d e9 f7 fd ff ff 8d 65 f4 5b 5e 5f 5d c3 80 78 0c 00 0f 84 98 00 00 00 55 89 e5 53 51 8b 58 1c c6 40 0c 00 <8b> 83 a0 00 00 00 48 89 83 a0 00 00 00 f6 05 ac 38 07 08 02 74 2c
`

Follwoing is my configuration

`
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	list ipaddr '127.0.0.1/8'

config globals 'globals'
	option ula_prefix 'fd3c:3439:836a::/48'

config interface 'wan'
	option device 'eth0'
	option proto 'static'
	option ipaddr '192.168.1.20'
	option netmask '255.255.255.0'
	option gateway '192.168.1.200'
	list dns '8.8.8.8'
	option metric '1'

config interface 'wg_mgmnt'
	option proto 'wireguard'
	list addresses '10.50.1.37/32'
	option private_key '#################################'

config wireguard_wg_mgmnt
	option persistent_keepalive '25'
	option public_key '########################'
	option description 'Management VPN'
	option endpoint_port '54321'
	option endpoint_host '##########'
	list allowed_ips '10.50.0.1'
	list allowed_ips ''
	option preshared_key '######################'
	option route_allowed_ips '1'

config interface 'loopback_ipsec'
	option proto 'static'
	option device 'lo'
	option ipaddr '10.220.220.1'
	option netmask '255.255.255.255'

config interface 'VXLAN_Kankavali'
	option proto 'vxlan'
	option peeraddr '10.220.220.3'
	option ipaddr '10.220.220.1'
	option port '8472'
	option vid '3'

config interface 'VXL_Kanka_IP'
	option proto 'static'
	option device 'VXLAN_Kankavali'
	option ipaddr '172.16.16.1'
	option netmask '255.255.255.252'
[... 696 lines cut ...]
ramips/mt7621

### Device

PBR-M1

### Image kind

Self-built image

### Steps to reproduce

just load network config

### Actual behaviour

netifd unable to create interfaces and segmentaion fault

### Expected behaviour

it should create interfaces

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

shripadrayewar, 2023-04-15:
any one??

shripadrayewar, 2023-04-15:
any one

nbd168, 2023-04-15:
Please give me some more details on what version you are using. I can't find any info on r0-f7c460685b. I also can't reproduce the issue myself with your config

shripadrayewar, 2023-04-15:
i tried multiple versions including on couple platforms like mipsbe and x86.
X86 - 
NAME="OpenWrt"
VERSION="21.02-SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 21.02-SNAPSHOT"
VERSION_ID="21.02-snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r0-6499f973"
OPENWRT_BOARD="x86/generic"
OPENWRT_ARCH="i386_pentium4"
OPENWRT_TAINTS="no-all"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 21.02-SNAPSHOT r0-6499f973"

MIPS:-
NAME="OpenWrt"
VERSION="21.02-SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 21.02-SNAPSHOT"
VERSION_ID="21.02-snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r0-0d716ad8"
OPENWRT_BOARD="ramips/mt7621"
OPENWRT_ARCH="mipsel_24kc"
OPENWRT_TAINTS="no-all"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 21.02-SNAPSHOT r0-0d716ad8"

nbd168, 2023-04-15:
Can you please try a recent version, either 22.03 or master snapshot?

shripadrayewar, 2023-04-15:
tried on 22.03. same issue
[image]

[image]

brada4, 2023-04-15:
What does ifconfig show,i.e how far file gets parsed succesfully (conf file itself looks sane)

nbd168, 2023-04-15:
I managed to reproduce it now. Will work on a fix

shripadrayewar, 2023-04-16:
> What does ifconfig show,i.e how far file gets parsed succesfully (conf file itself looks sane)

ifconfig only shows some random  interfaces from the file.

nbd168, 2023-04-17:
I pushed a fix to OpenWrt master. With that fix, it no longer crashes in my tests with your config.

closed 2023-04-18 by shripadrayewar