openwrt/openwrt #12395: [FeatureRequest] There is no 4k alignment when writing the firmware img to SSD. This is not good for SSD.

fannwong001, 2023-04-14:
### Describe the bug

There is no 4k alignment when writing the firmware img to SSD. This is not good for SSD.
And it will cause slow performance and short life for SSD.
Now the first partition's start sector number is 512. And the second partition's start sector number is not a multiple of 2048 either.
Both the legacy and efi ext4 firmware exist.

### OpenWrt version

r22493-5a8741aee9

### OpenWrt target/subtarget

x86/64

### Device

ASUS

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-04-14:
I used fdisk against current releases and snapshot and ext4 partitions are nicely 4K aligned. Only efiboot is stuffed before 512 1/2k blocks and turns out rather small.

Mijzelf, 2023-04-14:
> Now the first partition's start sector number is 512
512 is a multiple of 8.

fannwong001, 2023-04-15:
> > Now the first partition's start sector number is 512
> 
> 512 is a multiple of 8.

Sorry, I made a mistake and I have modified it.

brada4, 2023-04-15:
Dear, to optimally support 4kB sectors - partition should start at position divisible by eight (512 octet sectors), which is true for all published images, and your bug report. 
```
fdisk -l openwrt-x86-64-generic-ext4-combined.img 
Disk openwrt-x86-64-generic-ext4-combined.img: 120.5 MiB, 126353408 bytes, 246784 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xcc856286

Device                                    Boot Start    End Sectors  Size Id Type
openwrt-x86-64-generic-ext4-combined.img1 *      512  33279   32768   16M 83 Linux
openwrt-x86-64-generic-ext4-combined.img2      33792 246783  212992  104M 83 Linux
```

fannwong001, 2023-04-15:
> Dear, to optimally support 4kB sectors - partition should start at position divisible by eight (512 octet sectors), which is true for all published images, and your bug report.
>
> ```
>
> fdisk -l [...]

That is 4k bytes alignment, but not 4k sectors alignment. 4k sectors is better.

Mijzelf, 2023-04-16:
> That is 4k bytes alignment, but not 4k sectors alignment. 4k sectors is better
Why? Virtually every filesystem uses clusters of 4k bytes. So why would the start address of the containing partition matter, as long as it's 4k byte aligned?

Spudz76, 2023-04-17:
Because of eraseblocks, and secondarily sometimes how the mapper chip does its banking layout.  Also every SSD optimization howto says it must be at least on 4K, which yes 512 satisfies.  Some prefer 1MB alignment of partition starts (which, 2048 satisfies, and the default for every current partition software).  Misalignment could cause the mapper chip having to read two physical flash blocks to get 4K, or erase and rewrite two physical flash blocks to update 4K, leading to degraded speed and wasted write cycles and cache churn where it might get more cache-hits without the wasted reads.

On a magnetic disk I usually set gdisk to align by 8 since it definitely doesn't care about 1MB-aligned striping.  The 1MB comes from the blocksize of some SSD underlying flash layout / striping across the chips / what type of chips they are / etc.

fannwong001, 2023-04-17:
> Because of eraseblocks, and secondarily sometimes how the mapper chip does its banking layout.  Also every SSD optimization howto says it must be at least on 4K, which yes 512 satisfies.  Some prefer 1MB alignment of partition starts (which, 2048 satisfies, and the default for every current partition software).  Misalignment could cause the mapper chip having to read two physical flash blocks to get 4K, or erase and rewrite two physical flash blocks to update 4K, leading to degraded speed and wasted write cycles and cache churn where it might get more cache-hits without the wasted reads.
> 
> 
> 
> On a magnetic disk I usually set gdisk to align by 8 since it definitely doesn't care about 1MB-aligned striping.  The 1MB comes from the blocksize of some SSD underlying flash layout / striping across the chips / what type of chips they are / etc.
> 
> 
> 
> edit: (I have not heard of aligning to 4096 start == 2MB)

2048 or 4096 perform better, just test result.

brada4, 2023-04-25:
You can re-create overlay partition in different place, or better create additional partitions for data as manufacturer recommends. Writes in filesystem will happen at 4k offsets in 4k blocks anyway.

fannwong001, 2023-04-27:
> You can re-create overlay partition in different place, or better create additional partitions for data as manufacturer recommends. Writes in filesystem will happen at 4k offsets in 4k blocks anyway.

I find that it costs more cpu when install the openwrt to ssd than usb device.

brada4, 2023-04-27:
Would not be a big waste to align /overlay to 2MB mimicking Windows, but the alignment requirement >4kB is not something generally hinted by SSD manufacturers. Can you get some measurements either relocating /overlay or adding aligned-your-way partition after one you have (along with SSD name from dmesg/lsblk etc)?

fannwong001, 2023-04-28:
> Would not be a big waste to align /overlay to 2MB mimicking Windows, but the alignment requirement >4kB is not something generally hinted by SSD manufacturers. Can you get some measurements either [...]

Sorry I can not. I have replaced the ssd with a usb device. ssd is now used for other functions. I do not want to mess around anymore.

closed 2023-04-28 by fannwong001

Spudz76, 2023-04-28:
> alignment requirement >4kB is not something generally hinted by SSD manufacturers

Yeah and ~5K mile oil changes aren't hinted by vehicle manufacturers either but your stuff will burn out right around the end of warranty (which the manufacturer loves, right?) unless you ignore their propaganda and do the science yourself.  Just because something works for the warranty target doesn't mean it's ideal.

brada4, 2023-04-28:
Following analogy you are suggesting to cut off one cylinder for oil economy. Show that improves acceleration and maintains top speed.

Spudz76, 2023-04-28:
No, my analogy is perfect. :)  You spend more money to protect the longevity, just like "spending" some space for longevity (and a couple MB is not even as expensive as new oil 3x as often as "manufacturer suggested", we're not wasting two thirds of an SSD).

Spudz76, 2023-04-29:
Internal 4kb alignment doesn't matter if the device underneath the mapper chip is in 1MB stripes.  You still must align the partition starts to 1MB then for optimal treatment of the eraseblocks and such.  SSDs don't generally advertise their actual layout and just be lazy and say 4kb which isn't technically true but might be functionally true.