openwrt/openwrt #12397: On Redmi AX6, iw list : 4 streams at 160MHz

AlexeiGHub, 2023-04-14:
On Redmi AX6, iw list :

```
                        HE RX MCS and NSS set 160 MHz
                                1 streams: MCS 0-11
                                2 streams: MCS 0-11
                                3 streams: MCS 0-11
                                4 streams: MCS 0-11
                                5 streams: not supported
                                6 streams: not supported
                                7 streams: not supported
                                8 streams: not supported
                        HE TX MCS and NSS set 160 MHz
                                1 streams: MCS 0-11
                                2 streams: MCS 0-11
                                3 streams: MCS 0-11
                                4 streams: MCS 0-11
                                5 streams: not supported
                                6 streams: not supported
                                7 streams: not supported
                                8 streams: not supported
```

Does that mean it now supports 4 streams at 160MHz :) ?  Was there 2 before?

_Originally posted by @AlexeiGHub in https://github.com/openwrt/openwrt/issues/12333#issuecomment-1501217363_

brada4, 2023-04-16:
iw tool displays only capability maximums, not full matrix.
e.g you have wifi client card that does both 2.4 and 5 and advertises 80MHz ac capability which does not apply for every channel.
 your router supports 160 2x2 and 80 4x4, so both maximums are shown.

AlexeiGHub, 2023-04-16:
> so both maximums are shown
Possible, but as of previous FW 2.5.0.1 this information was correct 4 streams for 80 MHz width and 2 streams for 160 MHz width

brada4, 2023-04-16:
Please complete bug form noting exact revisions bug appeared in and the impact it has.

AlexeiGHub, 2023-04-17:
Form request created https://github.com/openwrt/openwrt/issues/12416#event-9028234382

closed 2023-04-17 by AlexeiGHub