openwrt/openwrt #12407: Openthread on OpenWrt

ngovancuong2735, 2023-04-16:
### Describe the bug

I tried build OpenThread for MT7688 use OpenWrt but when I build have that error:

mipsel-openwrt-linux-g++: warning: environment variable 'STAGING_DIR' not defined
mipsel-openwrt-linux-g++: warning: environment variable 'STAGING_DIR' not defined
/home/cuong/Documents/MT7688_OW19/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.3.0_musl/bin/../lib/gcc/mipsel-openwrt-linux-musl/8.3.0/../../../../mipsel-openwrt-linux-musl/bin/ld: cannot find -lanl
collect2: error: ld returned 1 exit status
make[2]: *** [tools/CMakeFiles/steering-data.dir/build.make:114: tools/steering-data] Error 1
make[1]: *** [CMakeFiles/Makefile2:4572: tools/CMakeFiles/steering-data.dir/all] Error 2
[ 97%] Building CXX object tests/unit/CMakeFiles/otbr-test-unit.dir/test_pskc.cpp.o
mipsel-openwrt-linux-g++: warning: environment variable 'STAGING_DIR' not defined
[ 97%] Building CXX object tests/unit/CMakeFiles/otbr-test-unit.dir/test_task_runner.cpp.o
mipsel-openwrt-linux-g++: warning: environment variable 'STAGING_DIR' not defined
[ 97%] Linking CXX executable otbr-test-unit
mipsel-openwrt-linux-g++: warning: environment variable 'STAGING_DIR' not defined
mipsel-openwrt-linux-g++: warning: environment variable 'STAGING_DIR' not defined
/home/cuong/Documents/MT7688_OW19/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.3.0_musl/bin/../lib/gcc/mipsel-openwrt-linux-musl/8.3.0/../../../../mipsel-openwrt-linux-musl/bin/ld: cannot find -lCppUTest
/home/cuong/Documents/MT7688_OW19/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.3.0_musl/bin/../lib/gcc/mipsel-openwrt-linux-musl/8.3.0/../../../../mipsel-openwrt-linux-musl/bin/ld: cannot find -lCppUTestExt
/home/cuong/Documents/MT7688_OW19/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.3.0_musl/bin/../lib/gcc/mipsel-openwrt-linux-musl/8.3.0/../../../../mipsel-openwrt-linux-musl/bin/ld: cannot find -lanl
collect2: error: ld returned 1 exit status
make[2]: *** [tests/unit/CMakeFiles/otbr-test-unit.dir/build.make:194: tests/unit/otbr-test-unit] Error 1
make[1]: *** [CMakeFiles/Makefile2:4647: tests/unit/CMakeFiles/otbr-test-unit.dir/all] Error 2
[ 97%] Linking CXX static library libotbr-mdns.a
[ 97%] Built target otbr-mdns
make: *** [Makefile:166: all] Error 2

**How to fix it.**
**I use OpenWrt 19.07 use toolchain-mipsel_24kc_gcc-8.3.0_musl.**

### OpenWrt version

19.07

### OpenWrt target/subtarget

/toolchain-mipsel_24kc_gcc-8.3.0_musl/bin/../lib/gcc/mipsel-openwrt-linux-musl/8.3.0/../../../../mipsel-openwrt-linux-musl/bin/ld

### Device

Acer A715

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-04-16:
Use :official: port from openthread? https://github.com/openthread/ot-br-posix/tree/main/etc/openwrt/openthread-br
Note that 19.07 is EOL, you might want to try current releases.

ngovancuong2735, 2023-04-16:
> Use :official: port from openthread? https://github.com/openthread/ot-br-posix/tree/main/etc/openwrt/openthread-br Note that 19.07 is EOL, you might want to try current releases.
But my HC is using 19.07 so upgrading to a higher version is quite difficult for us. Is there any way to build OpenThread on 19.07.

brada4, 2023-04-16:
You need to extract dependencies like `cannot find -lanl` from cmakelists and add them to local package descritpion file.
Best way is to practice against current releases and try to backport what is missing.

ngovancuong2735, 2023-04-16:
> You need to extract dependencies like `cannot find -lanl` from cmakelists and add them to local package descritpion file. Best way is to practice against current releases and try to backport what is missing.

Thank for your help. I will check it.

jow-, 2023-05-02:
Closing this is as this is not an OpenWrt bug but a general support request. Consider using the forums for such inquiries.

closed 2023-05-02 by jow-