openwrt/openwrt #12408: Fail to start WiFi 5GHz in AP mode after upgrading wpad-basic-wolfssl and hostapd-common

jwagra, 2023-04-16:
### Describe the bug

Packages upgrading breaks WiFI 5GHz, while 2.4GHz works fine 
Found this when updating to OpenWrt 22.03.4 and then after returning to 22.03.3
Recreating wireless network by LuCI does not affect in AP mode, Client mode starts
Now back to OpenWrt 22.03.3 without upgrading packages

### OpenWrt version

r20028-43d71ad93e

### OpenWrt target/subtarget

ipq40xx/generic

### Device

ASUS RT-AC58U

### Image kind

Official downloaded image

### Steps to reproduce

Upgrading wpad-basic-wolfssl on root from 2022-01-16-cff80b4f-15.2 to 2022-01-16-cff80b4f-16.2...
Downloading https://downloads.openwrt.org/releases/22.03.3/packages/arm_cortex-a7_neon-vfpv4/base/wpad-basic-wolfssl_2022-01-16-cff80b4f-16.2_arm_cortex-a7_neon-vfpv4.ipk
Upgrading hostapd-common on root from 2022-01-16-cff80b4f-15.2 to 2022-01-16-cff80b4f-16.2...
Downloading https://downloads.openwrt.org/releases/22.03.3/packages/arm_cortex-a7_neon-vfpv4/base/hostapd-common_2022-01-16-cff80b4f-16.2_arm_cortex-a7_neon-vfpv4.ipk

### Additional info

System Log:

Sun Apr 16 12:04:05 2023 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy1.conf (phy wlan1) --> new PHY
Sun Apr 16 12:04:06 2023 kern.warn kernel: [   95.955540] ath10k_ahb a800000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
Sun Apr 16 12:04:06 2023 kern.warn kernel: [   95.955599] ath10k_ahb a800000.wifi: msdu-desc: 2500  skid: 32
Sun Apr 16 12:04:06 2023 kern.info kernel: [   95.991325] ath10k_ahb a800000.wifi: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
Sun Apr 16 12:04:06 2023 kern.info kernel: [   95.992303] ath10k_ahb a800000.wifi: wmi print 'free: 53252 iram: 13432 sram: 35752'
Sun Apr 16 12:04:07 2023 kern.info kernel: [   96.279996] ath10k_ahb a800000.wifi: rts threshold -1
Sun Apr 16 12:04:07 2023 kern.warn kernel: [   96.281944] ath10k_ahb a800000.wifi: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
Sun Apr 16 12:04:07 2023 kern.info kernel: [   96.299234] br-lan: port 3(wlan1) entered blocking state
Sun Apr 16 12:04:07 2023 kern.info kernel: [   96.299287] br-lan: port 3(wlan1) entered disabled state
Sun Apr 16 12:04:07 2023 kern.info kernel: [   96.304120] device wlan1 entered promiscuous mode
Sun Apr 16 12:04:07 2023 daemon.notice hostapd: Frequency 5180 (primary) not allowed for AP mode, flags: 0x311979 RADAR
Sun Apr 16 12:04:07 2023 daemon.err hostapd: Primary frequency not allowed
Sun Apr 16 12:04:07 2023 daemon.warn hostapd: wlan1: IEEE 802.11 Configured channel (36) or frequency (5180) (secondary_channel=1) not found from the channel list of the current mode (2) IEEE 802.11a
Sun Apr 16 12:04:07 2023 daemon.warn hostapd: wlan1: IEEE 802.11 Hardware does not support configured channel
Sun Apr 16 12:04:07 2023 daemon.err hostapd: Could not select hw_mode and channel. (-3)
Sun Apr 16 12:04:07 2023 daemon.notice hostapd: wlan1: interface state UNINITIALIZED->DISABLED
Sun Apr 16 12:04:07 2023 daemon.notice hostapd: wlan1: AP-DISABLED
Sun Apr 16 12:04:07 2023 daemon.err hostapd: wlan1: Unable to setup interface.
Sun Apr 16 12:04:07 2023 daemon.notice hostapd: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
Sun Apr 16 12:04:07 2023 kern.info kernel: [   96.312507] device wlan1 left promiscuous mode
Sun Apr 16 12:04:07 2023 kern.info kernel: [   96.313695] br-lan: port 3(wlan1) entered disabled state
Sun Apr 16 12:04:07 2023 kern.warn kernel: [   96.418585] ath10k_ahb a800000.wifi: peer-unmap-event: unknown peer id 0
Sun Apr 16 12:04:07 2023 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Sun Apr 16 12:04:07 2023 daemon.notice hostapd: wlan1: CTRL-EVENT-TERMINATING
Sun Apr 16 12:04:07 2023 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan1 wasn't started
Sun Apr 16 12:04:07 2023 daemon.notice netifd: radio1 (3593): Command failed: ubus call hostapd config_add {"iface":"wlan1", "config":"/var/run/hostapd-phy1.conf"} (Invalid argument)
Sun Apr 16 12:04:07 2023 daemon.notice netifd: radio1 (3593): Usage: ubus [<options>] <command> [arguments...]
Sun Apr 16 12:04:07 2023 daemon.notice netifd: radio1 (3593): Options:
Sun Apr 16 12:04:07 2023 daemon.notice netifd: radio1 (3593):  -s <socket>:		Set the unix domain socket to connect to
Sun Apr 16 12:04:07 2023 daemon.notice netifd: radio1 (3593):  -t <timeout>:		Set the timeout (in seconds) for a command to complete
Sun Apr 16 12:04:07 2023 daemon.notice netifd: radio1 (3593):  -S:			Use simplified output (for scripts)
Sun Apr 16 12:04:07 2023 daemon.notice netifd: radio1 (3593):  -v:			More verbose output
Sun Apr 16 12:04:07 2023 daemon.notice netifd: radio1 (3593):  -m <type>:		(for monitor): include a specific message type
Sun Apr 16 12:04:07 2023 daemon.notice netifd: radio1 (3593): 			(can be used more than once)
Sun Apr 16 12:04:07 2023 daemon.notice netifd: radio1 (3593):  -M <r|t>		(for monitor): only capture received or transmitted traffic
Sun Apr 16 12:04:07 2023 daemon.notice netifd: radio1 (3593):
[... 136 lines cut ...]
                 * short GI for 40 MHz
                 * max A-MPDU length exponent
                 * min MPDU start spacing
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Maximum associated stations in AP mode: 32
        Supported extended features:
                * [ VHT_IBSS ]: VHT-IBSS
                * [ RRM ]: RRM
                * [ SET_SCAN_DWELL ]: scan dwell setting
                * [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
                * [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
                * [ AQL ]: Airtime Queue Limits (AQL)
                * [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
                * [ DEL_IBSS_STA ]: deletion of IBSS station support
                * [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
                * [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-04-16:
First dont start hostapd from command line, use uci/luci/ service wpad , once you run it ubus cannot control it anymore.
And set regulatory domain to get CH36 (or use auto-select if that is crowded)
-> reboot to kill out unsolicited processes
-> go to LuCI and set regulatory domain and auto and restart interface an check that good channel with good power gets selected. 
-> probably run `logread -f` to observe DFS activities better.

jwagra, 2023-04-16:
Thank you for the response. The output is the same as "System Log" 
I found a solution:
The "driver default" Country Code doesn't seem to work anymore, it now requires a specific Country Code to be set in the Advanced Settings (/etc/config/wireless).

Now:

Sun Apr 16 13:58:19 2023 daemon.notice hostapd: Remove interface 'wlan1'
Sun Apr 16 13:58:19 2023 daemon.notice hostapd: wlan1: interface state DFS->DISABLED
Sun Apr 16 13:58:19 2023 daemon.notice hostapd: wlan1: AP-DISABLED
Sun Apr 16 13:58:19 2023 daemon.notice hostapd: wlan1: CTRL-EVENT-TERMINATING
Sun Apr 16 13:58:19 2023 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Sun Apr 16 13:58:19 2023 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan1 wasn't started
Sun Apr 16 13:58:19 2023 daemon.notice hostapd: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
Sun Apr 16 13:58:19 2023 kern.info kernel: [ 1170.956503] device wlan1 left promiscuous mode
Sun Apr 16 13:58:19 2023 kern.info kernel: [ 1170.956748] br-lan: port 3(wlan1) entered disabled state
Sun Apr 16 13:58:20 2023 kern.info kernel: [ 1171.044300] ath10k_ahb a800000.wifi: mac flush null vif, drop 0 queues 0xffff
Sun Apr 16 13:58:20 2023 kern.warn kernel: [ 1171.046897] ath10k_ahb a800000.wifi: peer-unmap-event: unknown peer id 0
Sun Apr 16 13:58:20 2023 daemon.notice netifd: Wireless device 'radio1' is now down
Sun Apr 16 13:58:21 2023 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy1.conf (phy wlan1) --> new PHY
Sun Apr 16 13:58:22 2023 kern.warn kernel: [ 1173.631019] ath10k_ahb a800000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
Sun Apr 16 13:58:22 2023 kern.warn kernel: [ 1173.631080] ath10k_ahb a800000.wifi: msdu-desc: 2500  skid: 32
Sun Apr 16 13:58:22 2023 kern.info kernel: [ 1173.667055] ath10k_ahb a800000.wifi: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
Sun Apr 16 13:58:22 2023 kern.info kernel: [ 1173.668054] ath10k_ahb a800000.wifi: wmi print 'free: 53252 iram: 13432 sram: 35752'
Sun Apr 16 13:58:22 2023 kern.info kernel: [ 1173.953209] ath10k_ahb a800000.wifi: rts threshold -1
Sun Apr 16 13:58:22 2023 kern.warn kernel: [ 1173.954176] ath10k_ahb a800000.wifi: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
Sun Apr 16 13:58:22 2023 kern.info kernel: [ 1173.963099] br-lan: port 3(wlan1) entered blocking state
Sun Apr 16 13:58:22 2023 kern.info kernel: [ 1173.968677] br-lan: port 3(wlan1) entered disabled state
Sun Apr 16 13:58:22 2023 kern.info kernel: [ 1173.974164] device wlan1 entered promiscuous mode
Sun Apr 16 13:58:22 2023 daemon.notice hostapd: wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
Sun Apr 16 13:58:22 2023 daemon.notice hostapd: wlan1: interface state COUNTRY_UPDATE->HT_SCAN
Sun Apr 16 13:58:23 2023 daemon.notice hostapd: Switch own primary and secondary channel to get secondary channel with no Beacons from other BSSes
Sun Apr 16 13:58:23 2023 daemon.notice hostapd: wlan1: interface state HT_SCAN->DFS
Sun Apr 16 13:58:23 2023 daemon.notice hostapd: wlan1: DFS-CAC-START freq=5200 chan=40 sec_chan=-1, width=1, seg0=42, seg1=0, cac_time=60s
Sun Apr 16 13:58:24 2023 daemon.notice netifd: Wireless device 'radio1' is now up

closed 2023-04-16 by jwagra

brada4, 2023-04-16:
Actual bug was that it ever transmitted outside regulatory domain. The default "global" domain permits only few channels and only if netcard has passive radar scan, i.e not send any signal until confirming that air is clean for that meaning that if netcard does not support that kind of scan it will not transmit anything by default.