openwrt/openwrt #12409: Readmi AX6000 terrible download speed in AX 160 MHz mode

atolismesh, 2023-04-16:
### Describe the bug

Readmi AX6000 terrible (5-10 Mbit/s) download speed in AX 160 MHz mode. 
If I reduce channel width down to 80 MHz AX, the speed increases to normal - 800-900 Mbit/s. 

With the February snapshots, the speed was normal (800 -1600) on the same devices. 

### OpenWrt version

r22573-72780e3eac

### OpenWrt target/subtarget

mediatek/filogic

### Device

Xiaomi Redmi Router AX6000 (stock layout)

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-04-16:
No apple device supports 160MHz channels. Do other STAs magically work when you force them to 80mhz?

brada4, 2023-04-17:
Most likely cause is interference in one half of 160MHz
Read logread for ACS/DFS messages while channel is auto mode. Having any radar in range will not permit to use wider bands.

riKirsti, 2023-04-17:
> No apple device supports 160MHz channels. Do other STAs magically work when you force them to 80mhz?

@brada4 Really? Apple must made some mistakes:)
[image]
[image]
https://support.apple.com/guide/deployment/macbook-pro-wi-fi-specification-details-dep2ac3e3b51/1/web/1.0
https://support.apple.com/guide/deployment/ipad-wi-fi-specification-details-depf9bb7e412/1/web/1.0

atolismesh, 2023-04-17:
All devices worked fine with a channel width of 160 MHz until March and stopped working after I installed one of the March and next snapshots. 

Thank you for info.
Yes, I know that Apple devices do not support channel width 160 in 5 GHz. 

Now I will try to check the DFS operation on channels 52-64

brada4, 2023-04-17:
I mistook iphones being all "apple devices" ;-)

Check the regulatory details in logread , Apple has own supplicant which may be different if not be better than linux-android one.

Enable ACS, it may change between 36 and 100 optimally

Example 5km from small airfield with 2 planes dropping parachutes on weekends switching on radar when planes take off (most of AP-s in village drop to ch36):
```
Sat Apr 15 11:40:35 2023 daemon.notice hostapd: wlan0: interface state DFS->ENABLED
Sat Apr 15 11:42:57 2023 daemon.notice hostapd: wlan0: DFS-RADAR-DETECTED freq=5660 ht_enabled=0 chan_offset=0 chan_width=3 cf1=5690 cf2=0
Sat Apr 15 11:42:57 2023 daemon.notice hostapd: wlan0: DFS-NEW-CHANNEL freq=5260 chan=52 sec_chan=1
```

atolismesh, 2023-04-17:
iw dev phy1-ap0 survey dump 
shows noise level -92 dBm on each channel

I can't downgrade because the February snapshots are no longer available.
When I first discovered this issue in March, I thought it was related to @robimarko  #12274 .
But now I understand that they are not related and that's why I opened this issue.

robimarko, 2023-04-17:
This device is using mt76 and completely different SoC so it has nothing to do with ath11k

atolismesh, 2023-04-17:
> This device is using mt76 and completely different SoC so it has nothing to do with ath11k

@robimarko 
Sorry, the title of your message ("mac80211") slightly confused me. 
I thought that your PR  patched common mac80211.c/cfg80211.c, not only ath11k

brada4, 2023-04-17:
I cannot figure out which may be relevant (like changing MCS map or so):
https://github.com/openwrt/mt76/commits/master/mt7915

atolismesh, 2023-04-17:
Additional information. I just did an experiment - I installed an old router with HE160 support (Xiaomi AX3000). On the same channels with HE160, all devices operate at normal download and upload speed.

atolismesh, 2023-04-17:
> Basically install snapshot SDK and go back tu supposedly good times, then roll up till it breaks.

Thanks! I'll try it in the next few days

brada4, 2023-04-17:
Whatever snapshot you take, just take matching SDK so that module-package is not rejected.

Neustradamus, 2023-05-22:
@atolismesh: Have you tried?

atolismesh, 2023-10-31:
The issue is not reproducible after final 23.05.0 release

closed 2023-10-31 by atolismesh