openwrt/openwrt #12415: 22.03.04 factory on ea6350v3 softbrick

smilehappymore, 2023-04-17:
### Describe the bug

from wiki directions. boot from linksys partition and format with factory image.  This works fine with 22.03.03.  Rooter will boot up with green light after flashing in linksys.  Flashing with 22.03.04 factory however turns the router in softbrick.  no green light.  cannot access gui via manual ip.  Reflashing 22.03.03 factory from linksys partition restores it to operating condition.

### OpenWrt version

22.03.04

### OpenWrt target/subtarget

ipq40xx

### Device

ea6350v3

### Image kind

Official downloaded image

### Steps to reproduce

use recovery method to access linksys partition and flash new firmware factory imge.

### Actual behaviour

router reboots after flash for 20 seconds and front led remains off.  cannot access router

### Expected behaviour

router should reboot after flash and enter a green solid led.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-04-17:
You have to use sysupgrade image to upgrade from openwrt to openwrt.
Factory image is for one-time move from factory/vendor firmware.

smiehm, 2023-04-17:
> You have to use sysupgrade image to upgrade from openwrt to openwrt. Factory image is for one-time move from factory/vendor firmware.

no. this is not the upgrade method for this router.

brada4, 2023-04-17:
There are two methods

boot linksys and install factory every time

XOR

use sysupgrade rewriting other boot slot

You cannot install factory from OpenWRT.

smiehm, 2023-04-17:
> There are two methods
> 
> boot linksys and install factory every time
> 
> XOR
> 
> use sysupgrade rewriting other boot slot firstly one with linksys
> 
> You cannot install factory from OpenWRT.

I have already stated that I am flashing from linksys.
Second,  flashing with sysupgrade will not work flashing from linksys.
I feel like you are trolling or you are not even reading the bug report.  Please be respectful of other people.

Shine-, 2023-04-17:
@brada4 Check correctness of your claim before claiming something... In fact, for this device and related ones, it's officially possible to flash factory image via sysupgrade (introduced with af4f635eed28). Actually, that's possible for many many other devices as well and also it's the only way to upgrade from one build to another when the partition layout has changed inbetween.

@smiehm Your issue should already be fixed with 22.03 Snapshot (074db57936 or later), available from [here](https://downloads.openwrt.org/releases/22.03-SNAPSHOT/targets/ipq40xx/generic/).

@ Devs: You might want to consider pulling 22.03.4, since I assume most Linksys NAND targets are broken, not just MR8300 as stated in the release notes.

smiehm, 2023-04-17:
I can only verify sysupgrade method does not work for this router in the linksys partition as it will give an error message on flashing with the official 22.03.4 sysupgrade file of 6mb.  I cannot verify snapshot fix.

Djfe, 2023-05-11:
Please retest with 2022.03.05 :)

closed 2023-05-25 by smiehm