openwrt/openwrt #12418: MR8300 snapshot flash does not keep configuration even when option is selected

infantvin, 2023-04-17:
### Describe the bug

During the flash of a new firmware from snapshot on Linksys MR8300, I notice that even though I have selected the option to keep the configuration details during the flash process, the flash procedure never saves the existing settings of the router OS.
It always resets to the defaults which is incorrect behavior.

You can test this by flashing any new snapshot image to a MR8300 and selecting the option to save the configuration and not to change it. It will always get reset to defaults post the flash of the new snapshot image.

### OpenWrt version

OpenWrt SNAPSHOT r22573-72780e3eac

### OpenWrt target/subtarget

ipq40xx/generic

### Device

MR8300

### Image kind

Official downloaded image

### Steps to reproduce

Flash a openwrt image on a MR8300
Make config changes. Save

Download latest snapshot image for this device.
Flash it with the checkbox to store config settings.

Check if your settings are present/retained as the option was selected. They will be reset to defaults.

### Actual behaviour

Settings/whole config on router resets to default.

### Expected behaviour

The settings of Wireless should be retained and the router needs no additional configuration.

### Additional info

Happens all the time.

### Diffconfig

```text
NA
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Forst, 2023-04-29:
Happened to me as well, when upgrading from

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r21946-b246385126'
DISTRIB_TARGET='ipq40xx/generic'
DISTRIB_ARCH='arm_cortex-a7_neon-vfpv4'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r21946-b246385126'
DISTRIB_TAINTS=''
```

to

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r22664-1f41b6bb83'
DISTRIB_TARGET='ipq40xx/generic'
DISTRIB_ARCH='arm_cortex-a7_neon-vfpv4'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r22664-1f41b6bb83'
DISTRIB_TAINTS='no-all'
```

Linksys EA8300 (ipq40xx/generic).

Had to restore a config backup after upgrading.

badulesia, 2023-04-30:
Your original snapshot r21946 was old and didn't already include the fix. Try now with r22664.

badulesia, 2023-05-02:
It's probably this one
https://github.com/openwrt/openwrt/pull/11692

infantvin, 2023-05-04:
Thank you. I will test a upgrade this week and just be sure. If you can wait its fine, else if other users confirm this is fixed, we can close this out.  Happy that this has been addressed too.

badulesia, 2023-05-05:
A couple of days ago, I upgraded from 5.15.109 to .110 using a sysupgrade file, and keeping the settings.

infantvin, 2023-05-05:
In that case, we can close this issue. I will reopen this if the issue reoccurs or I find it happening.
Thank you for helping fix this again.

badulesia, 2023-05-06:
Hi.
I think you can, it is now working in the best way possible.

infantvin, 2023-05-08:
The settings seem to be saved. Closing.

closed 2023-05-08 by infantvin