openwrt/openwrt #12420: snapshot mac80211 crash -  driver-ops.h Failed check-sdata-in-driver check, flags: 0x1...

gtxaspec, 2023-04-18:
### Describe the bug

Latest snapshot/master on ipq806x for the R7800 crashes wifi, sometimes on boot, sometimes a bit after boot, and then wifi goes down:

```
[   69.602719] ------------[ cut here ]------------
[   69.602771] WARNING: CPU: 0 PID: 974 at backports-6.1-rc8/net/mac80211/driver-ops.h:611 __ieee80211_flush_queues+0x1a0/0x1a4 [mac80211]
[   69.606626] phy0-ap-vlan1: Failed check-sdata-in-driver check, flags: 0x1
[   69.618429] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet ath10k_pci ath10k_core ath pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211 cfg80211 cdc_ncm cdc_ether usbnet slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c crc_ccitt compat ledtrig_usbport seqiv cmac leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom ohci_platform ohci_hcd phy_qcom_ipq806x_usb ahci fsl_mph_dr_of ehci_platform ehci_fsl sd_mod ahci_platform libahci_platform libahci libata scsi_mod scsi_common ehci_hcd ramoops reed_solomon pstore gpio_button_hotplug crc32c_generic
[   69.682582] CPU: 0 PID: 974 Comm: hostapd Not tainted 5.15.106 #0
[   69.704599] Hardware name: Generic DT based system
[   69.710677] [<c030e7e8>] (unwind_backtrace) from [<c030a194>] (show_stack+0x14/0x20)
[   69.715365] [<c030a194>] (show_stack) from [<c0650c00>] (dump_stack_lvl+0x40/0x4c)
[   69.723264] [<c0650c00>] (dump_stack_lvl) from [<c031fe3c>] (__warn+0x90/0x104)
[   69.730643] [<c031fe3c>] (__warn) from [<c031ff24>] (warn_slowpath_fmt+0x74/0x84)
[   69.737848] [<c031ff24>] (warn_slowpath_fmt) from [<bf224794>] (__ieee80211_flush_queues+0x1a0/0x1a4 [mac80211])
[   69.745514] [<bf224794>] (__ieee80211_flush_queues [mac80211]) from [<bf1ecc44>] (sta_set_sinfo+0xdd4/0xef4 [mac80211])
[   69.755741] [<bf1ecc44>] (sta_set_sinfo [mac80211]) from [<bf1ed000>] (sta_info_destroy_addr_bss+0x4c/0x70 [mac80211])
[   69.766243] [<bf1ed000>] (sta_info_destroy_addr_bss [mac80211]) from [<bf19c428>] (nl80211_del_station+0xe8/0x2f4 [cfg80211])
[   69.777016] [<bf19c428>] (nl80211_del_station [cfg80211]) from [<c08dd3c8>] (genl_rcv_msg+0x164/0x35c)
[   69.788372] [<c08dd3c8>] (genl_rcv_msg) from [<c08dc2c4>] (netlink_rcv_skb+0xd8/0x148)
[   69.797570] [<c08dc2c4>] (netlink_rcv_skb) from [<c08dc9ec>] (genl_rcv+0x30/0x48)
[   69.805467] [<c08dc9ec>] (genl_rcv) from [<c08db8e4>] (netlink_unicast+0x190/0x2b8)
[   69.813020] [<c08db8e4>] (netlink_unicast) from [<c08dbbfc>] (netlink_sendmsg+0x1f0/0x460)
[   69.820486] [<c08dbbfc>] (netlink_sendmsg) from [<c083b2b4>] (____sys_sendmsg+0x1f4/0x248)
[   69.828823] [<c083b2b4>] (____sys_sendmsg) from [<c083d0e0>] (___sys_sendmsg+0xb0/0xf8)
[   69.837068] [<c083d0e0>] (___sys_sendmsg) from [<c083d278>] (sys_sendmsg+0x48/0x84)
[   69.844965] [<c083d278>] (sys_sendmsg) from [<c0300060>] (ret_fast_syscall+0x0/0x48)
[   69.852604] Exception stack(0xc2ba9fa8 to 0xc2ba9ff0)
[   69.860592] 9fa0:                   00000000 00000000 0000000b beb8f218 00000000 00000000
[   69.865544] 9fc0: 00000000 00000000 00c83910 00000128 00000004 beb8f260 00000001 00000000
[   69.873700] 9fe0: beb8f1c0 beb8f1b0 b6f12aec b6f11f78
[   69.884438] ---[ end trace d3a8d28e9409e446 ]---
```

i am forced to use the ath10k driver due to -ct not working well with ap/vlan and multicast.  this was not an issue with previously, this has only happened within the snapshot's of the last week or so. 

Confirmed issue happens on multiple R7800's all on latest master `r22580-e11d00d44c`.

Full dmesg:

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.15.106 (builder@buildhost) (arm-openwrt-linux-muslgnueabi-gcc (OpenWrt GCC 12.2.0 r22580-e11d00d44c) 12.2.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Mon Apr 17 22:11:22 2023
[    0.000000] CPU: ARMv7 Processor [512f04d0] revision 0 (ARMv7), cr=10c5787d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Netgear Nighthawk X4S R7800
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000042000000-0x000000005fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000042000000-0x000000005fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000042000000-0x000000005fffffff]
[    0.000000] percpu: Embedded 12 pages/cpu s16716 r8192 d24244 u49152
[    0.000000] pcpu-alloc: s16716 r8192 d24244 u49152 alloc=12*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 121800
[    0.000000] Kernel command line: 
[    0.000000] Bootloader command line (ignored): console=ttyHSL1,115200n8
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 472104K/491520K available (7407K kernel code, 619K rwdata, 1788K rodata, 1024K init, 247K bss, 19416K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[... 493 lines cut ...]
```

### OpenWrt version

snapshot

### OpenWrt target/subtarget

ipq806x

### Device

R7800

### Image kind

Official downloaded image

### Steps to reproduce

install latest snapshot

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

timocapa, 2023-04-21:
This also happens on ct firmware. Sorry, I killed all kallsyms etc to reduce image size, so this wont be much of help.

```
[ 3509.175280] ------------[ cut here ]------------
[ 3509.180019] WARNING: CPU: 0 PID: 1292 at backports-6.1.24/net/mac80211/driver-ops.h:611 0x82848370 [mac80211@700b17da+0x8b380]
[ 3509.191654] phy2-ap0.sta4: Failed check-sdata-in-driver check, flags: 0x1
[ 3509.198560] Modules linked in: ath9k ath10k_pci ath9k_common nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet ath9k_hw ath10k_core ath nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211 lzo cfg80211 rng_core nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 lzo_rle lzo_decompress lzo_compress libcrc32c compat sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac crypto_acompress gpio_button_hotplug crc32c_generic [last unloaded: ath9k]
[ 3509.264207] CPU: 0 PID: 1292 Comm: hostapd Not tainted 5.15.107 #0
[ 3509.270605] Stack : 80597d08 00000004 00000000 00000000 00000000 00000000 00000000 00000000
[ 3509.279190]         00000000 00000000 00000000 00000000 00000000 00000001 828fd9e8 0b5813bb
[ 3509.287698]         828fda80 00000000 00000000 828fd888 00000038 80306484 00000000 ffffffea
[ 3509.296342]         0000015e 828fd894 0000015e 8062ffd8 805a7108 828fd9c8 00000000 82848370
[ 3509.304900]         00000009 80b3f790 80630000 00000028 00000018 8035de70 00000000 80b10000
[ 3509.313418]         ...
[ 3509.315926] Call Trace:
[ 3509.315964] [<80306484>] 0x80306484
[ 3509.322013] [<82848370>] 0x82848370 [mac80211@700b17da+0x8b380]
[ 3509.328041] [<8035de70>] 0x8035de70
[ 3509.331610] [<800665b4>] 0x800665b4
[ 3509.335149] [<800665bc>] 0x800665bc
[ 3509.338700] [<800846bc>] 0x800846bc
[ 3509.342260] [<82848370>] 0x82848370 [mac80211@700b17da+0x8b380]
[ 3509.348348] [<800847d0>] 0x800847d0
[ 3509.351962] [<82848370>] 0x82848370 [mac80211@700b17da+0x8b380]
[ 3509.358069] [<8280adf0>] 0x8280adf0 [mac80211@700b17da+0x8b380]
[ 3509.364144] [<803ff218>] 0x803ff218
[ 3509.367789] [<8280b308>] 0x8280b308 [mac80211@700b17da+0x8b380]
[ 3509.373929] [<81f13cdc>] 0x81f13cdc [cfg80211@bdc44f1f+0x446c0]
[ 3509.380008] [<804557b0>] 0x804557b0
[ 3509.383554] [<80453a48>] 0x80453a48
[ 3509.387101] [<81f13c08>] 0x81f13c08 [cfg80211@bdc44f1f+0x446c0]
[ 3509.393172] [<8045564c>] 0x8045564c
[ 3509.396714] [<804546f8>] 0x804546f8
[ 3509.400270] [<8030614c>] 0x8030614c
[ 3509.403837] [<80455078>] 0x80455078
[ 3509.407386] [<80453cd8>] 0x80453cd8
[ 3509.410942] [<80454150>] 0x80454150
[ 3509.414480] [<803ed0b0>] 0x803ed0b0
[ 3509.418087] [<801ba2ec>] 0x801ba2ec
[ 3509.421684] [<80454120>] 0x80454120
[ 3509.425236] [<800b6a70>] 0x800b6a70
[ 3509.428791] [<803f56bc>] 0x803f56bc
[ 3509.432537] [<80453dfc>] 0x80453dfc
[ 3509.436094] [<803e1bd8>] 0x803e1bd8
[ 3509.439671] [<802d2a00>] 0x802d2a00
[ 3509.443233] [<803e1530>] 0x803e1530
[ 3509.446872] [<803e3b74>] 0x803e3b74
[ 3509.450462] [<802d2a00>] 0x802d2a00
[ 3509.454010] [<802d2b8c>] 0x802d2b8c
[ 3509.457598] [<803e3d88>] 0x803e3d88
[ 3509.461154] [<803e09cc>] 0x803e09cc
[ 3509.464706] [<8006dc84>] 0x8006dc84
[ 3509.468379] [<8006e2ec>] 0x8006e2ec
[ 3509.471956] [<8020cd7c>] 0x8020cd7c
[ 3509.475585] 
[ 3509.477101] ---[ end trace f4bed8c6c36366db ]---
```

gtxaspec, 2023-05-21:
@nbd168 any insight on this?

timocapa, 2023-05-21:
Does it still happen? I've reverted the patch locally for myself but I've planned to build again soon-ish

gtxaspec, 2023-05-21:
@timocapa Still happening, I don't have access to a build system so I can't revert it myself yet...

gtxaspec, 2023-05-24:
@timocapa i see kernel 6.1 is in testing now in snapshot, if you compile again you may want to upgrade to the latest one to see if this is resolved. FYI.

timocapa, 2023-05-24:
Won't help much until someone ports my targets to 6.1

gtxaspec, 2023-06-19:
still occurring on latest master.

gtxaspec, 2023-06-24:
@nbd168 anyway you could check on this? still occurring in today's master.

jannispinter, 2023-07-09:
I have exactly the same problem on ipq40xx with ath10k-ct on OpenWrt 23.05 branch.

gtxaspec, 2023-08-07:
latest master, still happening:

```
[  112.977125] WARNING: CPU: 1 PID: 1026 at backports-6.1.24/net/mac80211/driver-ops.h:611 __ieee80211_flush_queues+0x170/0x174 [mac80211]
[  112.980785] phy1-ap-vlan40: Failed check-sdata-in-driver check, flags: 0x1
[  112.992776] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_inet ath10k_pci ath10k_core ath pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211 cfg80211 cdc_ncm cdc_ether usbnet slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c crc_ccitt compat ledtrig_usbport sha512_arm seqiv cmac leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom ohci_platform ohci_hcd phy_qcom_ipq806x_usb ahci fsl_mph_dr_of ehci_platform ehci_fsl sd_mod ahci_platform libahci_platform libahci libata scsi_mod scsi_common ehci_hcd ramoops reed_solomon pstore gpio_button_hotplug crc32c_generic
[  113.053539] CPU: 1 PID: 1026 Comm: hostapd Not tainted 6.1.42 #0
[  113.075566] Hardware name: Generic DT based system
[  113.081733]  unwind_backtrace from show_stack+0x10/0x14
[  113.086330]  show_stack from dump_stack_lvl+0x40/0x4c
[  113.091453]  dump_stack_lvl from __warn+0x84/0xc4
[  113.096660]  __warn from warn_slowpath_fmt+0xec/0x14c
[  113.101348]  warn_slowpath_fmt from __ieee80211_flush_queues+0x170/0x174 [mac80211]
[  113.106399]  __ieee80211_flush_queues [mac80211] from sta_set_sinfo+0xe84/0xfa4 [mac80211]
[  113.113853]  sta_set_sinfo [mac80211] from sta_info_destroy_addr_bss+0x44/0x5c [mac80211]
[  113.122187]  sta_info_destroy_addr_bss [mac80211] from nl80211_del_station+0xe0/0x2b4 [cfg80211]
[  113.130437]  nl80211_del_station [cfg80211] from genl_rcv_msg+0x190/0x3c8
[  113.139284]  genl_rcv_msg from netlink_rcv_skb+0xb8/0x11c
[  113.145964]  netlink_rcv_skb from genl_rcv+0x28/0x34
[  113.151348]  genl_rcv from netlink_unicast+0x1d8/0x2b8
[  113.156383]  netlink_unicast from netlink_sendmsg+0x1d0/0x428
[  113.161330]  netlink_sendmsg from ____sys_sendmsg+0x1e0/0x274
[  113.167148]  ____sys_sendmsg from ___sys_sendmsg+0x6c/0xa4
[  113.172876]  ___sys_sendmsg from sys_sendmsg+0x44/0x74
[  113.178257]  sys_sendmsg from ret_fast_syscall+0x0/0x54
[  113.183378] Exception stack(0xc3643fa8 to 0xc3643ff0)
[  113.188500] 3fa0:                   00000000 00000000 0000000e beee1838 00000000 00000000
[  113.193712] 3fc0: 00000000 00000000 beee1d80 00000128 00000004 beee1880 00000001 00000000
[  113.201873] 3fe0: beee17e0 beee17d0 b6ef9164 b6ef85d8
```

oldium, 2023-09-21:
Flushing is broken (`__ieee80211_flush_queues` is common to all stack traces), now it is not possible to easily revert anything on kernel 6.1 unfortunately (I guess patches are part of the kernel).

```
Thu Sep 21 22:27:09 2023 kern.warn kernel: [  903.805979] ------------[ cut here ]------------
Thu Sep 21 22:27:09 2023 kern.warn kernel: [  903.806000] WARNING: CPU: 0 PID: 1587 at backports-6.5/net/mac80211/driver-ops.h:645 __ieee80211_flush_queues+0x1ac/0x1c4 [mac80211]
Thu Sep 21 22:27:09 2023 kern.warn kernel: [  903.806294] phy0-ap0-vlan1: Failed check-sdata-in-driver check, flags: 0x1
Thu Sep 21 22:27:09 2023 kern.warn kernel: [  903.806302] Modules linked in: ath9k ath9k_common pppoe ppp_async nft_fib_inet nf_flow_table_inet ath9k_hw ath10k_pci ath10k_core ath pppox ppp_generic nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_fwd_netdev nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_dup_netdev nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack_netlink nf_conntrack_bridge nf_conntrack mac80211 lzo cfg80211 slhc rng_core nfnetlink_queue nfnetlink_log nfnetlink nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_dup_netdev nf_defrag_ipv6 nf_defrag_ipv4 lzo_rle lzo_decompress lzo_compress libcrc32c crc_ccitt compat ledtrig_usbport rfkill sha512_generic seqiv jitterentropy_rng drbg md5 hmac cmac crypto_acompress fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd gpio_button_hotplug
Thu Sep 21 22:27:09 2023 kern.warn kernel: [  903.806640]  usbcore nls_base usb_common crc32c_generic
Thu Sep 21 22:27:09 2023 kern.warn kernel: [  903.806666] CPU: 0 PID: 1587 Comm: hostapd Not tainted 6.1.52 #0
Thu Sep 21 22:27:09 2023 kern.warn kernel: [  903.806682] Stack : 80790000 00000028 80c1cc64 800c0d28 00000000 00000004 00000000 00000000
Thu Sep 21 22:27:09 2023 kern.warn kernel: [  903.806725]         824ada24 80940000 80820000 806d8190 818ca1e8 00000001 824ad9c8 1e41ddcd
Thu Sep 21 22:27:09 2023 kern.warn kernel: [  903.806767]         00000000 00000000 806d8190 824ad8f8 ffffefff 00000000 00000000 ffffffea
Thu Sep 21 22:27:09 2023 kern.warn kernel: [  903.806807]         00000144 824ad904 00000144 807939c0 00000001 806d8190 00000000 82549c8c
Thu Sep 21 22:27:09 2023 kern.warn kernel: [  903.806848]         00000009 82aaa4e0 80790000 00000028 00000000 00000020 0806e098 80940000
Thu Sep 21 22:27:09 2023 kern.warn kernel: [  903.806889]         ...
Thu Sep 21 22:27:09 2023 kern.warn kernel: [  903.806899] Call Trace:
Thu Sep 21 22:27:09 2023 kern.warn kernel: [  903.806904] [<80066d4c>] show_stack+0x28/0xf0
Thu Sep 21 22:27:09 2023 kern.warn kernel: [  903.806938] [<805f8670>] dump_stack_lvl+0x38/0x60
Thu Sep 21 22:27:09 2023 kern.warn kernel: [  903.806956] [<80085260>] __warn+0x9c/0xe8
Thu Sep 21 22:27:09 2023 kern.warn kernel: [  903.806976] [<80085338>] warn_slowpath_fmt+0x8c/0xac
Thu Sep 21 22:27:09 2023 kern.warn kernel: [  903.807007] [<82549c8c>] __ieee80211_flush_queues+0x1ac/0x1c4 [mac80211]
Thu Sep 21 22:27:09 2023 kern.warn kernel: [  903.807333] [<8250b1c8>] sta_set_sinfo+0xef8/0x1040 [mac80211]
Thu Sep 21 22:27:09 2023 kern.warn kernel: [  903.807576]
Thu Sep 21 22:27:09 2023 kern.warn kernel: [  903.807664] ---[ end trace 0000000000000000 ]---
```

oldium, 2023-09-23:
For me this happens just for the _first_ AP-STA-DISCONNECTED only. The scenario is always the same - AP-STA-DISCONNECTED from hostapd followed by the error. It does not matter who is first on my Archer C7 v4 and v5, whether ath9k or ath10k driver. Always just the first one. Maybe this is some off-by-one or wrong usage of index 0?

regae, 2023-09-24:
same here with tp-link onhub and engenius eap1200h

oldium, 2023-09-25:
I tried to compile kernel with Kernel Debug Information enabled (with reduced option enabled) and added `RSTRIP:=:` (as mentioned [here](https://forum.openwrt.org/t/disable-package-strip-when-build-bin/111384/2)) to mac80211 `Makefile`, so here is more complete stack trace:

```
Sun Sep 24 21:02:04 2023 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED 7a:bd:bc:d2:13:72
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.405978] ------------[ cut here ]------------
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.405999] WARNING: CPU: 0 PID: 1583 at backports-6.5/net/mac80211/driver-ops.h:645 __ieee80211_flush_queues+0x1ac/0x1c4 [mac80211]
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.406544] phy0-ap0-vlan1: Failed check-sdata-in-driver check, flags: 0x1Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.406551] Modules linked in: ath9k ath9k_common pppoe ppp_async nft_fib_inet nf_flow_table_inet ath9k_hw ath10k_pci ath10k_core ath pppox ppp_generic nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_fwd_netdev nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_dup_netdev nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack_netlink nf_conntrack_bridge nf_conntrack mac80211 lzo cfg80211 slhc rng_core nfnetlink_queue nfnetlink_log nfnetlink nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_dup_netdev nf_defrag_ipv6 nf_defrag_ipv4 lzo_rle lzo_decompress lzo_compress libcrc32c crc_ccitt compat ledtrig_usbport rfkill sha512_generic seqiv jitterentropy_rng drbg md5 hmac cmac crypto_acompress fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd gpio_button_hotplug
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.406946]  usbcore nls_base usb_common crc32c_generic
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.406972] CPU: 0 PID: 1583 Comm: hostapd Not tainted 6.1.52 #0
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.406988] Stack : 807b0000 00000028 80c1c864 800c1d40 00000000 00000004 00000000 00000000
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.407031]         8289ba24 80970000 80830000 806ea684 818d3668 00000001 8289b9c8 499dad96
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.407071]         00000000 00000000 806ea684 8289b8f8 ffffefff 00000000 00000000 ffffffea
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.407111]         0000013d 8289b904 0000013d 807b19c0 00000001 806ea684 00000000 830476dc
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.407151]         00000009 84d504e0 807b0000 00000028 00000000 00000020 00000000 80970000
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.407191]         ...
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.407201] Call Trace:
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.407206] [<80066e4c>] show_stack+0x28/0xf0
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.407241] [<80608e98>] dump_stack_lvl+0x38/0x60
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.407263] [<80086240>] __warn+0x9c/0xe8
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.407283] [<80086318>] warn_slowpath_fmt+0x8c/0xac
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.407316] [<830476dc>] __ieee80211_flush_queues+0x1ac/0x1c4 [mac80211]
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.408080] [<8300af50>] __sta_info_destroy_part2+0x1a0/0x2e8 [mac80211]
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.408797] [<8300b43c>] sta_info_destroy_addr_bss+0x5c/0x84 [mac80211]
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.409514] [<82995478>] nl80211_del_station+0xd4/0x154 [cfg80211]
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.409759] [<804bfff8>] genl_rcv_msg+0x198/0x3d4
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.409799] [<804beb6c>] netlink_rcv_skb+0xb8/0x164
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.409817] [<804bf60c>] genl_rcv+0x30/0x48
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.409836] [<804be090>] netlink_unicast+0x230/0x354
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.409853] [<804be4d8>] netlink_sendmsg+0x324/0x488
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.409871] [<804388e0>] ____sys_sendmsg+0xc4/0x29c
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.409892] [<8043a488>] ___sys_sendmsg+0x7c/0xcc
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.409911] [<8043a9b4>] sys_sendmsg+0x4c/0xb8
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.409927] [<8006e1e4>] syscall_common+0x34/0x58
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.409943]
Sun Sep 24 21:02:04 2023 kern.warn kernel: [ 1196.409947] ---[ end trace 0000000000000000 ]---
```

regae, 2023-09-25:
tp-link onhub
```
IEEE 802.11: disassociated
[  139.055044] ------------[ cut here ]------------
[  139.055096] WARNING: CPU: 1 PID: 1214 at backports-6.5/net/mac80211/driver-ops.h:645 __ieee80211_flush_queues+0x170/0x188 [mac80211]
[  139.058906] phy1-ap0-vlan5: Failed check-sdata-in-driver check, flags: 0x1
[  139.070818] Modules linked in: ath10k_pci ath10k_core ath snd_soc_lpass_ipq806x snd_soc_lpass_cpu pppoe mac80211 cfg80211 xt_time xt_physdev xt_nat xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_REDIRECT xt_PROTO xt_MASQUERADE xt_LOG xt_CT vpd_sysfs ts_kmp ts_fsm ts_bm snd_soc_storm snd_soc_max98357a snd_soc_lpass_platform pppox nf_log_syslog netconsole memconsole_coreboot memconsole lcc_ipq806x iptable_nat ip6table_raw coreboot_table compat br_netfilter configs ledtrig_usbport cryptodev ip6table_nat ip6t_NPT ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT nf_reject_ipv6 msdos ip6_gre ip_gre gre ip6_tunnel oid_registry tunnel6 ip_tunnel raw_diag snd_rawmidi snd_seq_device snd_pcm_oss snd_mixer_oss snd_hwdep dns_resolver nls_utf8 nls_iso8859_1 nls_cp437 wp512 twofish_generic twofish_common tea serpent_generic khazad cast6_generic cast5_generic cast_common camellia_generic blowfish_generic blowfish_common anubis crypto_user algif_skcipher
[  139.071445]  algif_rng algif_hash algif_aead af_alg sha512_arm sha1_arm kpp cmac uas leds_gpio ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl tpm_i2c_infineon ehci_hcd gpio_button_hotplug vfat fat f2fs ext4 mbcache jbd2 tpm crc32_generic
[  139.164333] CPU: 1 PID: 1214 Comm: hostapd Not tainted 6.1.55 #0
[  139.186174] Hardware name: Generic DT based system
[  139.192342]  unwind_backtrace from show_stack+0x10/0x14
[  139.196938]  show_stack from dump_stack_lvl+0x40/0x4c
[  139.202059]  dump_stack_lvl from __warn+0x84/0xc4
[  139.207267]  __warn from warn_slowpath_fmt+0x108/0x194
[  139.211954]  warn_slowpath_fmt from __ieee80211_flush_queues+0x170/0x188 [mac80211]
[  139.217028]  __ieee80211_flush_queues [mac80211] from sta_set_sinfo+0xf04/0x104c [mac80211]
[  139.224548]  sta_set_sinfo [mac80211] from sta_info_destroy_addr_bss+0x48/0x60 [mac80211]
[  139.232881]  sta_info_destroy_addr_bss [mac80211] from nl80211_del_station+0xe0/0x2b4 [cfg80211]
[  139.241229]  nl80211_del_station [cfg80211] from genl_rcv_msg+0x190/0x3c8
[  139.250067]  genl_rcv_msg from netlink_rcv_skb+0xb8/0x11c
[  139.256746]  netlink_rcv_skb from genl_rcv+0x28/0x34
[  139.262128]  genl_rcv from netlink_unicast+0x16c/0x270
[  139.267162]  netlink_unicast from netlink_sendmsg+0x1d0/0x428
[  139.272110]  netlink_sendmsg from ____sys_sendmsg+0x1e0/0x274
[  139.277929]  ____sys_sendmsg from ___sys_sendmsg+0x6c/0xa4
[  139.283657]  ___sys_sendmsg from sys_sendmsg+0x44/0x74
[  139.289037]  sys_sendmsg from ret_fast_syscall+0x0/0x54
[  139.294158] Exception stack(0xc2145fa8 to 0xc2145ff0)
[  139.299284] 5fa0:                   012bfed0 bedb6f98 0000000f bedb6f98 00000000 00000001
[  139.304497] 5fc0: 012bfed0 bedb6f98 012bb348 00000128 00000000 00000001 00000004 00000000
[  139.312654] 5fe0: b6baaf88 bedb6f68 b6ba7380 b6b1b050
[  139.320903] ---[ end trace 0000000000000000 ]---
```

oldium, 2023-09-26:
Maybe this is regular, but because of `WARN_ONCE` we see it just once. It looks like the VIF (virtual interface associated to the physical interface) does not have the `IEEE80211_SDATA_IN_DRIVER` flag set for some reason. I added some logging and see that the flag is set for all my three main interfaces (phy0-ap0, phy1-ap0, phy1-ap1), but that is all. I have multiple VLANs and the log contains `phy0-ap0-vlan5`, so I _guess_ this has something to do with wifi-VLAN interface setup. Is VLAN also commonality of the problem for you?

regae, 2023-09-26:
the only problem with vlan for me is i can not use 802.11r, when FT is enabled after roaming always bind to the wrong interface ( to master interface ), not the vlan interface.
tested with local pmk and r1 push, same result.

oldium, 2023-09-26:
@regae I have that too. I _guess_ that this is because of the roaming mechanism of 802.11r - it does not use passwords to reauthenticate, but I use password-based VLAN assignment. Then when you roam you get the default VLAN interface being assigned (no password matches). I disabled the 802.11r because of this. I was thinking about some scripting magic here with hostapd+hotplugd - when the client authenticates, add to other routers static mapping for mac->vlan and when the client finally disconnects (but how to detect that it was not roamed...) delete the assignment to allow usage of different password (i.e. vlan) next time.

regae, 2023-09-26:
have you look at this [commit](https://w1.fi/cgit/hostap/commit/src/ap/wpa_auth.c?id=ba150059d1ec964add8f29eb2c92dd6dfde97308) ?
i'm kinda curious with this vlan issue

oldium, 2023-09-26:
Hm, interesting. Unfortunately, I am not so deep into the 802.11 protocols to tell if that makes any difference. But the description of the commit looks promising.

oldium, 2023-09-26:
But let's focus on the `check-sdata-in-driver` error. Anyone willing to debug this would be appreciated 😊 (I am “debugging” this by adding `pr_info` lines around, updating new patch with `quilt` and recompiling)

Situation: hostapd with VLANs (`vlan_file` option is set in the corresponding `hostapd.conf`). The code expects that `flags` attribute of `sdata` is updated with `IEEE80211_SDATA_IN_DRIVER` flag (see `net/mac80211/driver-ops.c`, function `drv_add_interface`, file in OpenWrt build `build_dir/target-mips_24kc_musl/linux-ath79_generic/backports-6.5/net/mac80211/driver-ops.c`). In the _latest_ kernel (OpenWrt is using 6.5 kernel backports...) the corresponding line is [here](https://github.com/torvalds/linux/blob/6465e260f48790807eef06b583b38ca9789b6072/net/mac80211/driver-ops.c#L73). It _looks like_ this line of code is not called for interfaces associated with additional VLANs.

oldium, 2023-09-26:
Just from reading the original bug report it looks like this commit is responsible for the “error” in kernel log: https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=package/kernel/mac80211/patches/subsys/331-wifi-mac80211-flush-queues-on-STA-removal.patch;h=00232ec1b9e9f6d077c5b17827bde675c9fec2b6;hb=d54c91bd9ab3c54ee06923eafbd67047816a37e4

oldium, 2023-09-30:
I believe this patch fixes the issue. Copy it to source tree into `package/kernel/mac80211/patches/subsys` and rebuild the mac80211 package (if you have already built image from sources, you can do `make package/kernel/mac80211/{clean,compile} && make`).
[999-flush_sdata.patch](https://github.com/openwrt/openwrt/files/12774212/999-flush_sdata.patch)

regae, 2023-10-02:
i think its only needed in drv_flush function.
@nbd168 thoughts?

```
@@ -638,11 +638,14 @@ static inline void drv_flush(struct ieee
 			     struct ieee80211_sub_if_data *sdata,
 			     u32 queues, bool drop)
 {
-	struct ieee80211_vif *vif = sdata ? &sdata->vif : NULL;
+	struct ieee80211_vif *vif;
 
 	might_sleep();
 
-	if (sdata && !check_sdata_in_driver(sdata))
+	sdata = get_bss_sdata(sdata);
+	vif = sdata ? &sdata->vif : NULL;
+
+	if (!check_sdata_in_driver(sdata))
 		return;
 
 	trace_drv_flush(local, queues, drop);
```

oldium, 2023-10-02:
I'm more worried that the kernel developers haven't noticed this problem for a long time - they probably don't use this feature. This also means that this feature (VLANs) is not well tested before making kernel changes.

oldium, 2023-10-02:
Can anyone test the patch at https://github.com/openwrt/openwrt/issues/12420#issuecomment-1741711350? If it works for you (I'm sure it should, but you never know… 😅), I can create a Request-for-Test pull request and send it to upstream then.

regae, 2023-10-02:
> Can anyone test the patch at [#12420 (comment)](https://github.com/openwrt/openwrt/issues/12420#issuecomment-1741711350)? If it works for you (I'm sure it should, but you never know… 😅), I can create a Request-for-Test pull request and send it to upstream then.

got kernel panic on engenius eap1200h when runing `wifi down`
```
<1>[   50.262948] CPU 0 Unable to handle kernel paging request at virtual address 00000b58, epc == 87049de8, ra == 87049de8
<4>[   50.273732] Oops[#1]:
<4>[   50.276039] CPU: 0 PID: 1271 Comm: hostapd Not tainted 5.4.257 #0
<4>[   50.282209] $ 0   : 00000000 00000001 00000001 00000000
<4>[   50.287506] $ 4   : 86dcba4c 00000010 00000010 00000000
<4>[   50.292803] $ 8   : 86f2e598 0000dc00 00000004 57d2e6a6
<4>[   50.298100] $12   : 00000000 00000001 87fe2d30 00000000
<4>[   50.303398] $16   : 0000ffff 00000000 876b84a0 00000000
<4>[   50.308696] $20   : 86f4b008 876b84a0 876b93c8 854b3200
<4>[   50.313994] $24   : 8066d240 00000000                  
<4>[   50.319291] $28   : 86dca000 86dcba48 00000000 87049de8
<4>[   50.324589] Hi    : 00129ce9
<4>[   50.327502] Lo    : 57d2e6a6
<4>[   50.330562] epc   : 87049de8 __ieee80211_flush_queues+0x154/0x1fc [mac80211]
<4>[   50.337749] ra    : 87049de8 __ieee80211_flush_queues+0x154/0x1fc [mac80211]
<4>[   50.344890] Status: 1100dc03	KERNEL EXL IE 
<4>[   50.349134] Cause : 00800008 (ExcCode 02)
<4>[   50.353198] BadVA : 00000b58
<4>[   50.356112] PrId  : 00019750 (MIPS 74Kc)
<4>[   50.360080] Modules linked in: ath9k ath9k_common ath9k_hw ath10k_pci ath10k_core ath mac80211 lzo cfg80211 rng_core lzo_rle lzo_decompress lzo_compress hwmon compat configs sha512_generic sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac ghash_generic gf128mul gcm ctr cmac ccm gpio_button_hotplug aead crypto_null cryptomgr crc32c_generic crypto_hash
<4>[   50.392367] Process hostapd (pid: 1271, threadinfo=f93328e6, task=0b3a0bd4, tls=77ecddfc)
<4>[   50.400647] Stack : 00000000 0000ffff 876b96c0 876ba5ec 00000000 876b84a0 876b84a0 00000000
<4>[   50.409120]         00000000 876b84a0 86f4b008 8701c67c 86f4afb8 87fe2d00 00000000 876b84a0
<4>[   50.417595]         86f2e580 876b84a0 876b84a0 8701e3a8 876b84a0 87053440 00000001 86f2e580
<4>[   50.426069]         86f2e580 870524b4 8068fb94 86f4ad78 86f2e5b0 00000000 876b84a0 86f4ad78
<4>[   50.434544]         86f2e5b0 870543cc 876b84a0 876b84a0 876b93c8 854b3200 00000000 86f4ad78
<4>[   50.443019]         ...
<4>[   50.445495] Call Trace:
<4>[   50.448046] [<87049de8>] __ieee80211_flush_queues+0x154/0x1fc [mac80211]
<4>[   50.454924] [<8701c67c>] ieee80211_ibss_leave+0x2e0/0x1ec4 [mac80211]
<4>[   50.461529] [<8701e3a8>] ieee80211_recalc_idle+0x18/0x40 [mac80211]
<4>[   50.467963] [<870524b4>] ieee80211_iter_chan_contexts_atomic+0xf84/0xff4 [mac80211]
<4>[   50.475778] Code: 02002825  0dc126bc  02402025 <8e230b58> 24020004  1062001d  00000000  1620ffd6  00000000 
<4>[   50.485663] 
<4>[   50.487552] ---[ end trace 8d46a8bf700bfd47 ]---
<0>[   50.503739] Kernel panic - not syncing: Fatal exception
```
but fine with tp-link onhub, no panic when wifi down.
weird

oldium, 2023-10-02:
My first guess is a null pointer exception. Do you use IBSS (Ad-Hoc/Peer-to-Peer config) together with VLANs?

regae, 2023-10-03:
more weird, patch was removed but still happened.
haven't notice this issue before, will dig more.
thank you

oldium, 2023-10-03:
You may be facing some race condition in the shutdown path. Without knowing further details, during my searches I found this (random) [patch](https://lore.kernel.org/all/20230829201610.cc648236f434.I205a557b06a27f58afe3880f4db43d554210c88d@changeid/) - if you want to give it a try.

oldium, 2023-10-13:
@regae May I kindly ask you to test this in #13620, please? If that works for you (meaning to test it on router, where you observed _only_ the check-sdata-in-driver issue), may I add your Tested-By to the upstream commit (for that I would need your name and email address)?

closed 2023-11-29 by dangowrt

gtxaspec, 2023-11-29:
are these fixed merged into master yet?

oldium, 2023-11-30:
Yay, there was recent activity, not a direct reply to my patch. My patch was mentioned in another Pull Request https://www.spinics.net/lists/linux-wireless/msg243106.html, so it looks good 😊

oldium, 2023-12-03:
🎉 Patch is now part of queue for kernel 6.6: https://git.kernel.org/pub/scm/linux/kernel/git/stable/stable-queue.git/tree/queue-6.6/wifi-mac80211-do-not-pass-ap_vlan-vif-pointer-to-dri.patch

oldium, 2023-12-15:
FYI: The patch is now part of v6.6.5 kernel.

oldium, 2023-12-24:
Merged. Enjoy 😁

nmpu, 2024-04-15:
Has this fix made it to a public release? My scenario is identical to yours. First disconnect on password-based VLAN.
```
Model: Extreme Networks WS-AP3825i
Architecture: e500v2
Target Platform: mpc85xx/p1020
Firmware Version: OpenWrt 23.05.2 r23630-842932a63d / LuCI openwrt-23.05 branch git-24.086.45142-09d5a38
Kernel Version: 5.15.137
```