openwrt/openwrt #12430: mwlwifi: update to version 10.3.9.0-20230311

pesa1234, 2023-04-20:
### Describe the bug

Frequently disconnection on devices on both 2.4ghz and 5ghz.
Standard configuration.
Wrt1900acs

### OpenWrt version

SNAPSHOT r22599 (Linux 5.15.107)

### OpenWrt target/subtarget

mvebu cortex a9

### Device

wrt1900acs vè

### Image kind

Self-built image

### Steps to reproduce

Change name of WiFi.
Assign a wpa2 key.
Connect a device.
After some times I get disconnected, distance, device router 5mt.

### Additional info

Previous driver, perfect.
I'm not the only one with this issue, also other people.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-04-20:
anything interesting in `logread`?

xophr, 2023-04-20:
Similar behavior with **wrt32x** on same build (r22599).

brada4, 2023-04-20:
The question is to run `logread -f` while it disconnects and post the recorded log (minus identifiable MAC addreses if applicable)

anomeome, 2023-04-20:
_AMSDU_ state (en/dis)abled due to previous 88W8864 issue where ii was recommended to disable in order to deal with issue.

pesa1234, 2023-04-20:
`Thu Apr 20 22:18:00 2023 cron.err crond[2868]: USER root pid 20863 cmd /etc/2watchdog.sh
Thu Apr 20 22:20:00 2023 cron.err crond[2868]: USER root pid 20941 cmd /etc/2watchdog.sh
Thu Apr 20 22:22:00 2023 cron.err crond[2868]: USER root pid 21008 cmd /etc/2watchdog.sh
Thu Apr 20 22:23:10 2023 kern.debug kernel: [ 3725.249779] ieee80211 phy0: Mac80211 start BA bc:7f:a4:09:xx:yy
Thu Apr 20 22:23:11 2023 kern.debug kernel: [ 3725.630703] ieee80211 phy0: Stop BA a8:34:6a:58:yy:xx
Thu Apr 20 22:24:00 2023 cron.err crond[2868]: USER root pid 21094 cmd /etc/2watchdog.sh
Thu Apr 20 22:26:00 2023 cron.err crond[2868]: USER root pid 21163 cmd /etc/2watchdog.sh
Thu Apr 20 22:28:00 2023 cron.err crond[2868]: USER root pid 21241 cmd /etc/2watchdog.sh
Thu Apr 20 22:28:19 2023 kern.debug kernel: [ 4034.261597] ieee80211 phy0: Mac80211 start BA a8:34:6a:58:yy:xx
Thu Apr 20 22:28:20 2023 kern.debug kernel: [ 4035.097687] ieee80211 phy0: Stop BA 0c:43:f9:1f:fe:05
Thu Apr 20 22:30:00 2023 cron.err crond[2868]: USER root pid 21321 cmd /etc/2watchdog.sh
Thu Apr 20 22:32:00 2023 cron.err crond[2868]: USER root pid 21389 cmd /etc/2watchdog.sh
Thu Apr 20 22:32:19 2023 daemon.info hostapd: phy1-ap0: STA bc:7f:a4:09:xx:yy IEEE 802.11: authenticated
Thu Apr 20 22:32:19 2023 daemon.info hostapd: phy1-ap0: STA bc:7f:a4:09:xx:yy IEEE 802.11: associated (aid 6)
Thu Apr 20 22:32:19 2023 daemon.notice hostapd: phy0-ap0: Prune association for bc:7f:a4:09:xx:yy
Thu Apr 20 22:32:19 2023 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED bc:7f:a4:09:xx:yy
Thu Apr 20 22:32:19 2023 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED bc:7f:a4:09:xx:yy auth_alg=open
Thu Apr 20 22:32:19 2023 daemon.info hostapd: phy1-ap0: STA bc:7f:a4:09:xx:yy WPA: pairwise key handshake completed (RSN)
Thu Apr 20 22:32:19 2023 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED bc:7f:a4:09:xx:yy
Thu Apr 20 22:32:38 2023 kern.debug kernel: [ 4293.017074] ieee80211 phy1: Mac80211 start BA bc:7f:a4:09:xx:yy
Thu Apr 20 22:32:40 2023 kern.debug kernel: [ 4294.493666] ieee80211 phy1: Mac80211 start BA bc:7f:a4:09:xx:yy
Thu Apr 20 22:32:42 2023 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED b0:60:88:69:yy:xx
Thu Apr 20 22:32:43 2023 daemon.info hostapd: phy0-ap0: STA b0:60:88:69:yy:xx IEEE 802.11: authenticated
Thu Apr 20 22:32:43 2023 daemon.info hostapd: phy0-ap0: STA b0:60:88:69:yy:xx IEEE 802.11: associated (aid 1)
Thu Apr 20 22:32:43 2023 daemon.notice hostapd: phy0-ap0: AP-STA-CONNECTED b0:60:88:69:yy:xx auth_alg=open
Thu Apr 20 22:32:43 2023 daemon.info hostapd: phy0-ap0: STA b0:60:88:69:yy:xx WPA: pairwise key handshake completed (RSN)
Thu Apr 20 22:32:43 2023 daemon.notice hostapd: phy0-ap0: EAPOL-4WAY-HS-COMPLETED b0:60:88:69:yy:xx
Thu Apr 20 22:32:44 2023 kern.debug kernel: [ 4298.926997] ieee80211 phy0: Mac80211 start BA b0:60:88:69:yy:xx
Thu Apr 20 22:32:45 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.181.235 b0:60:88:69:yy:xx
Thu Apr 20 22:32:45 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.181.235 b0:60:88:69:yy:xx F15_Plus_2
Thu Apr 20 22:32:47 2023 daemon.notice hostapd: phy0-ap0: STA-OPMODE-N_SS-CHANGED bc:7f:a4:09:xx:yy 1
Thu Apr 20 22:32:47 2023 daemon.info hostapd: phy0-ap0: STA bc:7f:a4:09:xx:yy IEEE 802.11: authenticated
Thu Apr 20 22:32:47 2023 daemon.info hostapd: phy0-ap0: STA bc:7f:a4:09:xx:yy IEEE 802.11: associated (aid 2)
Thu Apr 20 22:32:47 2023 daemon.notice hostapd: phy1-ap0: Prune association for bc:7f:a4:09:xx:yy
Thu Apr 20 22:32:47 2023 daemon.notice hostapd: phy1-ap0: AP-STA-DISCONNECTED bc:7f:a4:09:xx:yy
Thu Apr 20 22:32:47 2023 daemon.notice hostapd: phy0-ap0: AP-STA-CONNECTED bc:7f:a4:09:xx:yy auth_alg=open
Thu Apr 20 22:32:47 2023 daemon.info hostapd: phy0-ap0: STA bc:7f:a4:09:xx:yy WPA: pairwise key handshake completed (RSN)
Thu Apr 20 22:32:47 2023 daemon.notice hostapd: phy0-ap0: EAPOL-4WAY-HS-COMPLETED bc:7f:a4:09:xx:yy
Thu Apr 20 22:33:08 2023 daemon.info hostapd: phy1-ap0: STA bc:7f:a4:09:xx:yy IEEE 802.11: authenticated
Thu Apr 20 22:33:08 2023 daemon.info hostapd: phy1-ap0: STA bc:7f:a4:09:xx:yy IEEE 802.11: associated (aid 6)
Thu Apr 20 22:33:08 2023 daemon.notice hostapd: phy0-ap0: Prune association for bc:7f:a4:09:xx:yy
Thu Apr 20 22:33:08 2023 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED bc:7f:a4:09:xx:yy
Thu Apr 20 22:33:08 2023 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED bc:7f:a4:09:xx:yy auth_alg=open
Thu Apr 20 22:33:08 2023 daemon.info hostapd: phy1-ap0: STA bc:7f:a4:09:xx:yy WPA: pairwise key handshake completed (RSN)
Thu Apr 20 22:33:08 2023 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED bc:7f:a4:09:xx:yy
Thu Apr 20 22:33:38 2023 daemon.info hostapd: phy0-ap0: STA bc:7f:a4:09:xx:yy IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Thu Apr 20 22:33:58 2023 kern.debug kernel: [ 4373.365185] ieee80211 phy1: Mac80211 start BA bc:7f:a4:09:xx:yy
Thu Apr 20 22:34:00 2023 cron.err crond[2868]: USER root pid 21467 cmd /etc/2watchdog.sh
Thu Apr 20 22:34:47 2023 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED b0:60:88:69:yy:xx
Thu Apr 20 22:34:47 2023 daemon.info hostapd: phy0-ap0: STA b0:60:88:69:yy:xx IEEE 802.11: authenticated
Thu Apr 20 22:34:47 2023 daemon.info hostapd: phy0-ap0: STA b0:60:88:69:yy:xx IEEE 802.11: associated (aid 1)
Thu Apr 20 22:34:47 2023 daemon.notice hostapd: phy0-ap0: AP-STA-CONNECTED b0:60:88:69:yy:xx auth_alg=open
[... cut ...]
Thu Apr 20 22:37:02 2023 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED ac:1e:9e:af:yy:xx
Thu Apr 20 22:37:03 2023 kern.debug kernel: [ 4557.939675] ieee80211 phy1: Mac80211 start BA ac:1e:9e:af:yy:xx
Thu Apr 20 22:37:32 2023 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED bc:7f:a4:09:xx:yy
Thu Apr 20 22:37:32 2023 daemon.info hostapd: phy0-ap0: STA ac:1e:9e:af:yy:xx IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Thu Apr 20 22:37:34 2023 daemon.info hostapd: phy1-ap0: STA bc:7f:a4:09:xx:yy IEEE 802.11: authenticated`

olimpiumarius, 2023-04-20:
I don't realize if I had disconnections but I noticed a strange ping, honestly I spent three hours on the TV watching movies on netflix and I had no disconnection problems, all the partitions move relatively ok, only this ping shows us that there are some problems !

you can find my settings here on the forum for more details!
[https://forum.openwrt.org/t/linksys-wrt1900acs-v2-openwrt-snapshot/157825/5](https://github.com/kaloz/mwlwifi/pull/url)

```
Pinging 192.168.15.1 with 32 bytes of data:
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=3ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=13ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=4ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=3ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=4ms TTL=64
Reply from 192.168.15.1: bytes=32 time=5ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=2ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=15ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=2ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=3ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
[... 137 lines cut ...]
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=3278ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64
Reply from 192.168.15.1: bytes=32 time=8ms TTL=64
Reply from 192.168.15.1: bytes=32 time=2441ms TTL=64
Reply from 192.168.15.1: bytes=32 time=9ms TTL=64
Reply from 192.168.15.1: bytes=32 time=40ms TTL=64
Reply from 192.168.15.1: bytes=32 time=42ms TTL=64
Reply from 192.168.15.1: bytes=32 time=36ms TTL=64
Reply from 192.168.15.1: bytes=32 time=3408ms TTL=64
Reply from 192.168.15.1: bytes=32 time=20ms TTL=64
Reply from 192.168.15.1: bytes=32 time=1ms TTL=64

Ping statistics for 192.168.15.1:
    Packets: Sent = 217, Received = 217, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 1ms, Maximum = 3462ms, Average = 218ms

```
https://github.com/kaloz/mwlwifi/pull/408

pesa1234, 2023-04-20:
I don't know if it can help, but I get disconnection when I have all devices 2.4 and 5ghz connected... 
I have 5 devices 5ghz and 5 at 2.4ghz

brada4, 2023-04-20:
1/ The default hostapd inactivity limit is 300s , but any activity should reset it. You can try setting it to something like 600s, still it does not explain disconnect in one minute/
2/ slow pings - seems bufferbloat which certainly should not be there (?)
3/ do disconnects stop if you stop 5GHz radio (presuming both AP-s are named same - if not - make extra 2.4 AP under same name as 5GHz for test,  note you need to change the subtle "network" assignment down the web form)

pesa1234, 2023-04-20:
I didn't see slow ping... I change name to 5g wifi, nobody is connected to 5g but I get same issue...

olimpiumarius, 2023-04-20:
just now I tried another test, the ping is high and there are disconnections, I opened putty via ssh while doing a speed test, I lost connection with putty during the speedtest and at that moment the ping was also high...

pesa1234, 2023-04-20:
So, what can I do to help...? I can try to compile only drivers and understand which commit give issues...

anomeome, 2023-04-20:
See [post](https://github.com/kaloz/mwlwifi/pull/408#issuecomment-1516245692), but this should be fought in only one place, I would suggest upstream is the correct place as other OS are probably impacted.

pesa1234, 2023-04-21:
> See [post](https://github.com/kaloz/mwlwifi/pull/408#issuecomment-1516245692), but this should be fought in only one place, I would suggest upstream is the correct place as other OS are probably impacted.

Ok, I was able to self compile drivers. I will let you know which commit damage the driver (I hope)

In the same time my suggestion is to come back to previous version here on snapshot...

tamer-hassan, 2023-04-23:
thanks @pesa1234 
had the same issue with latest snapshots on WRT3200ACM
I confirm that rolling back to r22257 snapshot ([divested build](https://divested.dev/unofficial-openwrt-builds/mvebu-linksys/20230310-01/)) which was a day prior to the merge of pr containing this troublesome 1-line patch, the frequent mass disconnections do not happen, at least while having the following additional 2 options for wifi config
```
	option max_inactivity '86400'
	option disassoc_low_ack '0'
```

closed 2023-04-30 by pesa1234