openwrt/openwrt #12438: qca-ssdk: build fails with make >= v4.4 installed on host system

pwned-pixel, 2023-04-21:
### Describe the bug

I'm doing builds for Xiaomi AX3600 and Zyxel NBG7815. Both need qca-ssdk package as a hard requirement.
Build fails as soon make >= v4.4 is installed on host system.

### OpenWrt version

master

### OpenWrt target/subtarget

ipq807x

### Device

generic

### Image kind

Self-built image

### Steps to reproduce

Build Environment:

Arch Linux latest with make >= v4.4 installed on host system (other systems are affected also with make >= 4.4 installed e. g. Fedora 38).

Description:

I'm doing builds for Xiaomi AX3600 and Zyxel NBG7815. Both need qca-ssdk package as a hard requirement.
Build fails as soon make >= v4.4 is installed on host system.

For make v4.3 it is working as expected.

### Actual behaviour

For

`make -j1 V=sc package/kernel/qca-ssdk/{clean,compile}`

the last lines from the log are:

```
make[5]: Entering directory '/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-2022-09-12-628b22bc/src/adpt/sfp'
aarch64-openwrt-linux-musl-gcc -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-2022-09-12-628b22bc=qca-ssdk-2022-09-12-628b22bc -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/user/openwrt/nbg7815/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/usr/include -I/home/user/openwrt/nbg7815/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/include/fortify -I/home/user/openwrt/nbg7815/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/include  -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-2022-09-12-628b22bc/include -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-2022-09-12-628b22bc/include/common -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-2022-09-12-628b22bc/include/api -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-2022-09-12-628b22bc/include/fal -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-2022-09-12-628b22bc/include/ref -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-2022-09-12-628b22bc/include/adpt -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-2022-09-12-628b22bc/include/hsl -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-2022-09-12-628b22bc/include/hsl/phy -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-2022-09-12-628b22bc/include/sal/os -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-2022-09-12-628b22bc/include/sal/os/linux -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-2022-09-12-628b22bc/include/sal/sd -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-2022-09-12-628b22bc/include/sal/sd/linux/hydra_howl -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-2022-09-12-628b22bc/include/sal/sd/linux/uk_interface -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-2022-09-12-628b22bc/include/init -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-2022-09-12-628b22bc/include/hsl/isisc -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-2022-09-12-628b22bc/include/hsl/hppe -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-2022-09-12-628b22bc/include/adpt/hppe -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-2022-09-12-628b22bc/include/hsl/mht -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-2022-09-12-628b22bc/include/hsl/sfp -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-2022-09-12-628b22bc/include/adpt/sfp -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-2022-09-12-628b22bc/include/shell_lib -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.106 -I/home/user/openwrt/nbg7815/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/bin/../lib/gcc//120200/include/ -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.106/include -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.106/source -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.106/source/include -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.106/source/arch/arm64/mach-msm/include -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.106/arch/arm64/mach-msm/include -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.106/source/arch/arm64/include -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.106/arch/arm64/include -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.106/source/arch/arm64/include/asm -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.106/arch/arm64/include/generated -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.106/include/generated/uapi -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.106/include/uapi -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.106/arch/arm64/include/uapi -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.106/source/arch/arm64/include/asm/mach -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.106/arch/arm64/include/generated/uapi -I/home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.106/source/include/uapi -include /home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/linux-5.15.106/include/linux/kconfig.h -DIN_ACL -DIN_FDB -DIN_IGMP -DIN_LEAKY -DIN_LED -DIN_MIB -DIN_MIRROR -DIN_MISC -DIN_PORTCONTROL -DIN_PORTVLAN -DIN_QOS -DIN_RATE -DIN_STP -DIN_VLAN -DIN_COSMAP -DIN_IP -DIN_NAT -DIN_FLOW -DIN_TRUNK -DIN_SEC -DIN_QM -DIN_INTERFACECONTROL -DIN_CTRLPKT -DIN_SERVCODE -DIN_RSS_HASH -DIN_MALIBU_PHY -DIN_AQUANTIA_PHY -DIN_QCA803X_PHY -DIN_QCA808X_PHY -DIN_SFP_PHY -DIN_PHY_I2C_MODE -DIN_VSI -DIN_PPPOE -DIN_BM -DIN_SHAPER -DIN_POLICER -DIN_UNIPHY -DUK_IF -DUK_NL_PROT= -DUK_MINOR_DEV=254 -DIN_NETLINK  -Wall -DVERSION=\"3.0.0.2\" -DBUILD_DATE=\"2023-04-21-08:50:05\" -DOS=\"linux\" -D"KBUILD_STR(s)=#s" -D"KBUILD_MODNAME=KBUILD_STR(qca-ssdk)" -DISISC -DHPPE -DMHT -DIN_SFP -DKVER34 -DKVER32 -DLNX26_22 -D__KERNEL__ -DKERNEL_MODULE -DMODULE -Os -pipe -march=armv8-a -mcpu=cortex-a53+crypto -fno-caller-saves -fno-strict-aliasing -fno-common -Wno-format-security -Wno-pointer-sign -Wno-unused-but-set-variable -Wno-error=unused-result -mcmodel=large -fno-stack-protector -MM adpt_sfp.c > /home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-2022-09-12-628b22bc/build/linux/KSLIB/adpt_sfp.d.tmp
sed 's,\(adpt_sfp\)\.o[ :]*,\1.o adpt_sfp.d : ,g' < /home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-2022-09-12-628b22bc/build/linux/KSLIB/adpt_sfp.d.tmp > /home/user/openwrt/nbg7815/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-2022-09-12-628b22bc/build/linux/KSLIB/adpt_sfp.d
[... cut ...]
make: *** [/home/user/openwrt/nbg7815/include/toplevel.mk:231: package/kernel/qca-ssdk/compile] Fehler 2
```

### Additional info

For reference:

https://forum.openwrt.org/t/need-help-fixing-my-toolchain-qca-ssdk-compile-fails/149368

### Diffconfig

```text
CONFIG_TARGET_ipq807x=y
CONFIG_TARGET_ipq807x_generic=y
CONFIG_TARGET_MULTI_PROFILE=y
CONFIG_TARGET_DEVICE_ipq807x_generic_DEVICE_xiaomi_ax3600=y
CONFIG_TARGET_DEVICE_ipq807x_generic_DEVICE_zyxel_nbg7815=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

notgood, 2023-05-03:
Confirming, unable to build qca-ssdk from OpenWRT master on Arch workstation.

Downgrading make from 4.4 to 4.3 allows build to proceed.

Ansuel, 2023-05-06:
Yep we are aware of that... Thinking of reporting this to make mailing list as regression

robimarko, 2023-05-10:
I find it interesting that as @Ansuel found out commenting out the `vpath` processing for `.d` files it compiles and seems to work fine

Ansuel, 2023-05-10:
with further check, commenting out vpath is not correct... that just disable the entire dependency way used on ssdk to detect changes on consequent builds...  ok for our scenario... not ok as a fix :(

Ansuel, 2023-05-10:
I also had fun with building intermediate version from source between 4.3.90 and 4.3 and the problem was still there... also for some reason some commit doesn't compile at all...

robimarko, 2023-05-23:
So, it seems that backporting support for building as a proper module works around our make issue.
So, testing from more people would be great, it works for me on Qnap.
https://github.com/openwrt/openwrt/pull/12704

PR #12704 (2023-05-23): kernel: qca-ssdk: backport support for building as kernel module
Currently, SSDK is rather special in the sense that its not being built as a proper out of tree module at all but rather like a userspace application and that involves a lot of make magic which unfortunately broke with make version 4.4 and newer.

Luckily QCA finally added a way to build SSDK as an out of tree module and it uses the kernel buildsystem which makes it compile with make 4.4 as well.
So lets backport the support for it and switch to using it.

Fixes: #12438
closed 2023-05-23 by openwrt-bot