openwrt/openwrt #12456: SNAPSHOT x86 64

xh0823, 2023-04-22:
### Describe the bug

X86 upgraded today's snapshot and failed to start successfully 
[image]
[image]
[image]

### OpenWrt version

SNAPSHOT r22181-a13af31231

### OpenWrt target/subtarget

x86 64

### Device

x86 64

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

illumiN8i, 2023-04-23:
Same issue here. Broken after updating to latest snapshot.

illumiN8i, 2023-04-24:
I was able to get some more details by using failsafe mode. Not sure what it means, but hopefully it helps.
[image]