openwrt/openwrt #12459: Possible spin lock race condition in net/bridge code

Brain2000, 2023-04-23:
### Describe the bug

I've experienced intermittent crash issues throughout version 22 (and possibly earlier) when VLANs are used, especially when a VLAN bridge is added/removed, which happens every time a station connects/disconnects. I've also noticed other people's Oops stack traces seem to occur immediately as a VLAN STA is added/removed.

So I compiled the kernel with the following lock check build options:
```
[*] Enable kernel lock checking
[*] Compile the kernel with detect Soft Lockups
[*] Compile the kernel with detect Hard Lockups
```
and it quickly logged a potential locking issue. I don't know if this might have something to do with the crashes.
For now I have changed all spin_lock(&br->lock) to spin_lock_bh(&br->lock) as a test, just to see what happens. But it may need someone much more familiar with this code to determine if this is an issue, and what the remedy would be:

```
Sat Apr 22 18:53:56 2023 kern.warn kernel: [   26.153812]
Sat Apr 22 18:53:56 2023 kern.warn kernel: [   26.155337] ========================================================
Sat Apr 22 18:53:56 2023 kern.warn kernel: [   26.161693] WARNING: possible irq lock inversion dependency detected
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.168049] 5.15.108 #0 Tainted: G S
Sat Apr 22 18:53:56 2023 kern.warn kernel: [   26.172932] --------------------------------------------------------
Sat Apr 22 18:53:56 2023 kern.warn kernel: [   26.179285] swapper/0/0 just changed the state of lock:
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.184515] ffffff80028a2b98 (&br->lock){+.-.}-{2:2}, at: br_forward_delay_timer_expired+0x24/0x1ec
Sat Apr 22 18:53:56 2023 kern.warn kernel: [   26.193618] but this lock took another, SOFTIRQ-unsafe lock in the past:
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.200320]  (fs_reclaim){+.+.}-{0:0}
Sat Apr 22 18:53:56 2023 kern.warn kernel: [   26.200345]
Sat Apr 22 18:53:56 2023 kern.warn kernel: [   26.200345]
Sat Apr 22 18:53:56 2023 kern.warn kernel: [   26.200345] and interrupts could create inverse lock ordering between them.
Sat Apr 22 18:53:56 2023 kern.warn kernel: [   26.200345]
Sat Apr 22 18:53:56 2023 kern.warn kernel: [   26.215390]
Sat Apr 22 18:53:56 2023 kern.warn kernel: [   26.215390] other info that might help us debug this:
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.221918] Chain exists of:
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.221918]   &br->lock --> offload_lock#2 --> fs_reclaim
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.221918]
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.231704]  Possible interrupt unsafe locking scenario:
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.231704]
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.238488]        CPU0                    CPU1
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.243009]        ----                    ----
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.247529]   lock(fs_reclaim);
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.250665]                                local_irq_disable();
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.256574]                                lock(&br->lock);
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.262140]                                lock(offload_lock#2);
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.268143]   <Interrupt>
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.270754]     lock(&br->lock);
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.273976]
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.273976]  *** DEADLOCK ***
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.273976]
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.279885] 1 lock held by swapper/0/0:
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.283713]  #0: ffffffc008003da8 ((&p->forward_delay_timer)){+.-.}-{0:0}, at: call_timer_fn+0x0/0x240
Sat Apr 22 18:53:56 2023 kern.warn kernel: [   26.293028]
Sat Apr 22 18:53:56 2023 kern.warn kernel: [   26.293028] the shortest dependencies between 2nd lock and 1st lock:
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.300850]   -> (fs_reclaim){+.+.}-{0:0} {
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.305034]      HARDIRQ-ON-W at:
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.308343]                         lock_acquire+0xfc/0x2c0
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.313911]                         fs_reclaim_acquire+0x84/0xbc
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.319915]                         kmem_cache_alloc_trace+0x3c/0x2e0
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.326353]                         init_rescuer+0x4c/0xdc
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.331836]                         workqueue_init+0xc8/0x1dc
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.337578]                         kernel_init_freeable+0x1c0/0x354
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.343928]                         kernel_init+0x20/0x120
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.349408]                         ret_from_fork+0x10/0x20
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.354976]      SOFTIRQ-ON-W at:
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.358284]                         lock_acquire+0xfc/0x2c0
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.363849]                         fs_reclaim_acquire+0x84/0xbc
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.369850]                         kmem_cache_alloc_trace+0x3c/0x2e0
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.376286]                         init_rescuer+0x4c/0xdc
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.381765]                         workqueue_init+0xc8/0x1dc
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.387504]                         kernel_init_freeable+0x1c0/0x354
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.393851]                         kernel_init+0x20/0x120
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.399330]                         ret_from_fork+0x10/0x20
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.404896]      INITIAL USE at:
Sat Apr 22 18:53:56 2023 kern.debug kernel: [   26.408117]                        lock_acquire+0xfc/0x2c0
[... 318 lines cut ...]
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-compat=y
CONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-json=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-lua-runtime=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-lua=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_umdns=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wpad-wolfssl=y
CONFIG_WOLFSSL_HAS_NO_HW=y
CONFIG_WPA_WOLFSSL=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ptpt52, 2023-04-23:
mark.

LGA1150, 2023-05-06:
Issue identified:
In `br_offload_port_state`, either `rhashtable_init` or `rhashtable_free_and_destroy` function is called inside a spin lock critical section, while both the rhashtable operation may sleep, potentially causing a dead lock.
@nbd168

LGA1150, 2023-05-17:
Follow up: [Patchwork](https://patchwork.ozlabs.org/project/openwrt/patch/20230515102640.1321685-1-dqfext@gmail.com/)