openwrt/openwrt #12462: ccache 4.6 doesn't build with gcc 13

mchehab, 2023-04-23:
### Describe the bug

Trying to build OpenWRT on Fedora 38 (with gcc13) doesn't work:

```
make[3]: Entering directory '/home/mchehab/devel/lede-gaivota/tools/ccache'
MAKEFLAGS="" /home/mchehab/devel/lede-gaivota/staging_dir/host/bin/ninja  -j1 -C /home/mchehab/devel/lede-gaivota/build_dir/host/ccache-4.6 
ninja: Entering directory `/home/mchehab/devel/lede-gaivota/build_dir/host/ccache-4.6'
[1/108] Building CXX object src/CMakeFiles/ccache_framework.dir/Args.cpp.o
FAILED: src/CMakeFiles/ccache_framework.dir/Args.cpp.o 
/home/mchehab/devel/lede-gaivota/staging_dir/host/bin/g++ -DXXH_STATIC_LINKING_ONLY -Dnssv_CONFIG_SELECT_STRING_VIEW=nssv_STRING_VIEW_NONSTD -I/home/mchehab/devel/lede-gaivota/build_dir/host/ccache-4.6 -I/home/mchehab/devel/lede-gaivota/build_dir/host/ccache-4.6/src -O2 -I/home/mchehab/devel/lede-gaivota/staging_dir/host/include -DNDEBUG -include /home/mchehab/devel/lede-gaivota/build_dir/host/ccache-4.6/config.h -Wall -std=c++14 -MD -MT src/CMakeFiles/ccache_framework.dir/Args.cpp.o -MF src/CMakeFiles/ccache_framework.dir/Args.cpp.o.d -o src/CMakeFiles/ccache_framework.dir/Args.cpp.o -c /home/mchehab/devel/lede-gaivota/build_dir/host/ccache-4.6/src/Args.cpp
In file included from src/Util.hpp:21,
                 from src/Args.hpp:22,
                 from src/Args.cpp:19:
src/Stat.hpp:132:3: error: 'uint64_t' does not name a type
  132 |   uint64_t size() const;
      |   ^~~~~~~~
src/Stat.hpp:28:1: note: 'uint64_t' is defined in header '<cstdint>'; did you forget to '#include <cstdint>'?
   27 | #include <string>
  +++ |+#include <cstdint>
   28 | 
src/Stat.hpp:134:3: error: 'uint64_t' does not name a type
  134 |   uint64_t size_on_disk() const;
      |   ^~~~~~~~
src/Stat.hpp:134:3: note: 'uint64_t' is defined in header '<cstdint>'; did you forget to '#include <cstdint>'?
src/Stat.hpp:219:8: error: 'uint64_t' does not name a type
  219 | inline uint64_t
      |        ^~~~~~~~
src/Stat.hpp:219:8: note: 'uint64_t' is defined in header '<cstdint>'; did you forget to '#include <cstdint>'?
src/Stat.hpp:225:8: error: 'uint64_t' does not name a type
  225 | inline uint64_t
      |        ^~~~~~~~
src/Stat.hpp:225:8: note: 'uint64_t' is defined in header '<cstdint>'; did you forget to '#include <cstdint>'?
src/Util.hpp: In function 'int64_t Util::size_change_kibibyte(const Stat&, const Stat&)':
src/Util.hpp:352:41: error: 'const class Stat' has no member named 'size_on_disk'
  352 |   return (static_cast<int64_t>(new_stat.size_on_disk())
      |                                         ^~~~~~~~~~~~
src/Util.hpp:353:43: error: 'const class Stat' has no member named 'size_on_disk'
  353 |           - static_cast<int64_t>(old_stat.size_on_disk()))
      |                                           ^~~~~~~~~~~~
ninja: build stopped: subcommand failed.
```

Updating to ccache 4.6.3 works:

```
commit 189fecf2987aad4ac0847078ec6adf06c0e415a8
Author: Mauro Carvalho Chehab <mchehab@kernel.org>
Date:   Sun Apr 23 08:46:28 2023 +0100

    tools/ccache: update to 4.6.3
    
    Full changelog: https://ccache.dev/releasenotes.html#_ccache_4_6_3
    
    Signed-off-by: Mauro Carvalho Chehab <mchehab@kernel.org>

diff --git a/tools/ccache/Makefile b/tools/ccache/Makefile
index 9b2937dccd64..0cf528bb5f63 100644
--- a/tools/ccache/Makefile
+++ b/tools/ccache/Makefile
@@ -8,11 +8,11 @@ include $(TOPDIR)/rules.mk
 include $(INCLUDE_DIR)/target.mk
 
 PKG_NAME:=ccache
-PKG_VERSION:=4.6
+PKG_VERSION:=4.6.3
 
 PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
 PKG_SOURCE_URL:=https://github.com/ccache/ccache/releases/download/v$(PKG_VERSION)
-PKG_HASH:=3d2bb860f4359169e640f60cf7cc11da5fab5fb9aed55230d78141e49c3945e9
+PKG_HASH:=1e3a251bb112632553b8255a78661fe526c3a16598496d51128c32b218fd8b22
 
 include $(INCLUDE_DIR)/host-build.mk
 include $(INCLUDE_DIR)/cmake.mk
```

PS.: I reverted 3de4572ed308 ("tools/ccache: fix build with musl and gcc 12"), as ccache 4.6.3 already has it fixed. From its changelog (https://ccache.dev/releasenotes.html#_ccache_4_6_3):
```
Build improvements

    Fixed build with musl when using GCC 12.
    [contributed by Khem Raj]
```

### OpenWrt version

22.03 tree on changeset bf81a921715c ("ath79: create Aruba AP-105 APBoot compatible image")

### OpenWrt target/subtarget

ramips/mt7621

### Device

Netgear R6260

### Image kind

Self-built image

### Steps to reproduce

select ccache and run make.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

mchehab, 2023-04-23:
Fix patch: https://github.com/openwrt/openwrt/pull/12467

Ansuel, 2023-05-06:
@mchehab guess this can be closed as fixed? tell me if you still have this and i will reopen

closed 2023-05-06 by Ansuel

mchehab, 2023-05-07:
Yes, the issue was fixed. Thanks!