openwrt/openwrt #12469: Banana Pi R3 wds button sends reset

trunneml, 2023-04-23:
### Describe the bug

The Banana Pi R3 has two buttons:

1. WPS
2. Reset

When pressing the WPS button on my banana pi r3 a reset event is handled and the router reboots. The reset button has no function at all.

Also the activation for failsafe ealy in button needs the WPS button and not the reset button.

### OpenWrt version

r22619-bf3b876a63

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R3

### Image kind

Official downloaded image

### Steps to reproduce

Press `WPS` Button.
Device Reboots.

### Actual behaviour

Device reboots when pressing WPS

### Expected behaviour

Device reboots when pressing Reset.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-04-24:
Can you check that corner of the board:
https://wiki.banana-pi.org/File:BPI-R3_interface_750.jpg
Are they really swapped or case does not match board?

alanswanson, 2023-04-28:
If you read the documentation https://openwrt.org/toh/sinovoip/bananapi_bpi-r3#buttons you would know that due to a hardware bug we cannot use the reset button when a NVM drive is connected and hence the reset function is remapped to the WPS button.

trunneml, 2023-07-18:
Ah yes you are right. I only found this page https://openwrt.org/toh/sinovoip/bananapi_bpi_r3_v1.0_v1.1 where this info is missing.

closed 2023-07-18 by trunneml