openwrt/openwrt #12470: dnsmasq: server list option and strictorder is in reverse

barhom, 2023-04-23:
### Describe the bug

When settings dnsmasq to not be used for local lookups (noresolv) but still keep dhcp clients requests for dns have allservers= and strictorder=1

```
config dnsmasq
        option allservers '0'
        option strictorder '1'
        list server '1.1.1.1'
        list server '8.8.8.8'
       option noresolv '1'
```

You expect that 1.1.1.1 is the first DNS server configured in dnsmasq however the opposite is actually true,
```
Sun Apr 23 21:56:28 2023 daemon.info dnsmasq[4423]: using nameserver 8.8.8.8#53
Sun Apr 23 21:56:28 2023 daemon.info dnsmasq[4423]: using nameserver 1.1.1.1#53
```

### OpenWrt version

r16399+159-c67509efd7

### OpenWrt target/subtarget

ipq807x/ipq60xx

### Device

GL Technologies, Inc. AX1800

### Image kind

Self-built image

### Steps to reproduce

set

config dnsmasq
        option allservers '0'
        option strictorder '1'
        list server '1.1.1.1'
        list server '8.8.8.8'
       option noresolv '1'

### Actual behaviour

8.8.8.8 is the first dns used

### Expected behaviour

1.1.1.1 should be the first dns used

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

barhom, 2023-04-25:
> dnsmasq.conf.example

How would one configure strict-order for command-line servers then?
It seems to be following the order I put them in but in reverse right now.

Is this not something you think people would want?

brada4, 2023-04-25:
It is a documented upstream feature that command-line servers are not sorted. If you want sorting you have to use resolv.conf method.

brada4, 2023-04-25:
> Is this not something you think people would want?

Likely, ask upstream to make order of forwarders predictable. Sooner or later it will land back in OpenWRT.

brada4, 2023-04-27:
Here (follow the thread) is discussion on how strict-order actually worsens the situation.
https://lists.thekelleys.org.uk/pipermail/dnsmasq-discuss/2023q1/016877.html