openwrt/openwrt #12491: CONFIG_MT753X_GSW definition missing in kernel_menconfig

pameruoso, 2023-04-28:
### Describe the bug

When running make kernel_menuconfig the CONFIG_MT753X_GSW is not available. I guess it's because there is no definition to load the driver which is present in the folder target/linux/mediatek/files/drivers/net/phy/mtk. The driver is not compiled and I'm having problems with my mt7986a (ex5601-t0) to recognize the 2.5 gbit ports

Probably something like the following is needed:
https://github.com/openwrt/openwrt/blob/2c530fcb972c112e7a2b10f9c21ac6d276624b5e/target/linux/mediatek/patches-5.15/500-gsw-rtl8367s-mt7622-support.patch#L22

### OpenWrt version

latest

### OpenWrt target/subtarget

MT7986A

### Device

ex5601-t0 (mt7986a)

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ptpt52, 2023-04-29:
mt753x is deprecated for mt7622/mt798x, we now use dsa driver mt7530

pameruoso, 2023-04-29:
@ptpt52 thanks for clarifying, but can you please check this issue I'm having and suggest why with 5.4 and that driver i can set correctly the registers?
https://forum.openwrt.org/t/adding-openwrt-support-for-zyxel-ex5601-t0/155914/36?u=skyscreaper

pameruoso, 2023-05-16:
The issue can be closed. The new driver is working fine, I had to fix the assert and deassert timings in dts

closed 2023-05-16 by pameruoso