openwrt/openwrt #12501: RT-AX53U Bad page map in process dynamic_dns_upd  pte:08505b8f pmd:837f6000

zz64, 2023-04-29:
### Describe the bug

Hi,
this is just output that i collected from running dmsg.
apparently does not affect any thing. (not hung running system)
```
[1173641.165572] BUG: Bad page map in process dynamic_dns_upd  pte:08505b8f pmd:837f6000
[1173641.173514] page:936c14f1 refcount:1 mapcount:-1 mapping:00000000 index:0x77dcf pfn:0x8505
[1173641.182000] anon flags: 0x80014(uptodate|lru|swapbacked)
[1173641.187840] raw: 00080014 80b05b88 80b1637c 83c4b969 00077dcf 00000000 fffffffe 00000001
[1173641.196400] raw: 81430800
[1173641.199193] page dumped because: bad pte
[1173641.203322] page->mem_cgroup:81430800
[1173641.207392] addr:0045f000 vm_flags:00100877 anon_vma:83c5ccb0 mapping:81889b78 index:4f
[1173641.215582] file:busybox fault:0x80134bb4 mmap:0x8012f430 readpage:0x802506f0
[1173641.223139] CPU: 1 PID: 26210 Comm: dynamic_dns_upd Not tainted 5.10.161 #0
[1173641.230243] Stack : 00000000 807293a8 809b0000 809c0000 00000000 00000000 00000000 00000000
[1173641.238767]         00000000 00000000 00000000 00000000 00000000 00000001 83f99a40 41683194
[1173641.247287]         83f99ad8 00000000 00000000 83f998e8 00000038 8039a284 ffffffea 00000000
[1173641.255808]         83f998f4 000001d6 807ce868 ffffffff 83f99a20 806f8134 80000000 80700000
[1173641.264328]         81889b78 809c0000 807c0000 809c0000 00000001 80403fb8 00000004 80990004
[1173641.272848]         ...
[1173641.275459] Call Trace:
[1173641.275477] [<8039a284>] 0x8039a284
[1173641.281777] [<80403fb8>] 0x80403fb8
[1173641.285427] [<80007b08>] 0x80007b08
[1173641.289069] [<80007b10>] 0x80007b10
[1173641.292716] [<8037f3a0>] 0x8037f3a0
[1173641.296367] [<8007fcdc>] 0x8007fcdc
[1173641.300016] [<80163704>] 0x80163704
[1173641.303702] [<80134bb4>] 0x80134bb4
[1173641.307350] [<8012f430>] 0x8012f430
[1173641.310996] [<802506f0>] 0x802506f0
[1173641.314642] [<80165f48>] 0x80165f48
[1173641.318284] [<801a27d0>] 0x801a27d0
[1173641.321942] [<802aca1c>] 0x802aca1c
[1173641.325594] [<80323350>] 0x80323350
[1173641.329247] [<801da194>] 0x801da194
[1173641.332899] [<80133d28>] 0x80133d28
[1173641.336559] [<801c2f10>] 0x801c2f10
[1173641.340221] [<80166e30>] 0x80166e30
[1173641.343888] [<8013ef34>] 0x8013ef34
[1173641.347535] [<80140760>] 0x80140760
[1173641.351182] [<802ad2bc>] 0x802ad2bc
[1173641.354836] [<80167140>] 0x80167140
[1173641.358479] [<80141380>] 0x80141380
[1173641.362138] [<8016ed24>] 0x8016ed24
[1173641.365782] [<801b6908>] 0x801b6908
[1173641.369447] [<8002c800>] 0x8002c800
[1173641.373095] [<801b1b98>] 0x801b1b98
[1173641.376747] [<80224e8c>] 0x80224e8c
[1173641.380397] [<802260bc>] 0x802260bc
[1173641.384043] [<801a7244>] 0x801a7244
[1173641.387698] [<801b0f04>] 0x801b0f04
[1173641.391345] [<801b0280>] 0x801b0280
[1173641.394992] [<801b13f8>] 0x801b13f8
[1173641.398634] [<800189d4>] 0x800189d4
[1173641.402282] [<801b2204>] 0x801b2204
[1173641.405940] [<80014578>] 0x80014578
[1173641.409594] 
```

### OpenWrt version

r20028-43d71ad93e

### OpenWrt target/subtarget

ramips/mt7621

### Device

ASUS RT-AX53U

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-04-29:
Is it happening repeatedly? Whhat is in logread around same moment?

zz64, 2023-04-30:
I discovered this by chance, I don't even know when it happened.
when i check the logread, i only see dhcp requests and nothing else.
Now, I restarted the device, and if it happens again, I will come back to report it.

don't hesitate to ask, if I have to do anything to reproduce the bug.
The OpenWrt deserves the contribution.

brada4, 2023-04-30:
It looks like a bit flipped in memory metadata at least under normal hardware.
- overclock
- high temperature
- undervoltage due to hungry usb/pcie device and weak power circuit
- EMI flips the bit, like broke microwave or nearby electric discharge(s)
- SMI-like interference that flips bit at one physical address
- Some buggy device or its driver that flips random or fixed bit

If you are sure thermal conditions are good and fireproof you can run stress-ng to stress memory, though it may take longer than natural occurence to trigger hardware bug artificially.