openwrt/openwrt #12502: [BUG, WEB] Unannounced versions are listed in website's Firmware Selector

michael9dk, 2023-04-29:
### Describe the bug

The Firmware Selector on website, shows the latest build, despite it hasn't finished building / isn't announced as a public release.

This causes confusion for (new?) users, that don't see the official announcements of a new release, before downloading a new build.

Installing Attended Upgrades solve this issue, but the websites Firmware Selector is not consistent with it.

### OpenWrt version

openwrt website

### OpenWrt target/subtarget

N/A

### Device

N/A

### Image kind

Official downloaded image

### Actual behaviour

A user can download a unofficial build from Firmware Selector on the website. There is no warning about wether it's a official release.

### Expected behaviour

a) only update Firmware Selector AFTER the build pipeline has succeeded (and are approved/ready for the public).
b) add a notification in Firmware Selector, that shows a "work in progress"-warning, and links to the current officially release.

### Additional info

Background is my question at https://www.reddit.com/r/openwrt/comments/132f4zw/22035_soft_release/ji87d1i/

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-04-30:
It has own issue tracker:
https://gitlab.com/openwrt/web/firmware-selector-openwrt-org/-/issues

Djfe, 2023-05-01:
This is not a bug, the devs just didn't have the time to do announcements, yet. This is a community project.
But you may upgrade already.
The release is a fixup release for 22.03.04
The "biggest" fix is that the boot issues for 22.03.04 were solved.
All other commits are in the git repo if you want to read them
https://github.com/openwrt/openwrt/commits/v22.03.5