openwrt/openwrt #12508: WDS Kernel Oops on WRX36

smileys29, 2023-05-01:
### Describe the bug

I have a WRX36 functioning as a Router and setup as 5GHz WDS Access Point.  I have two RT3200's connecting as wireless bridges (WDS Client).  The WRX36 will come up with a kernel oops, looking like something going on with hostapd:

`[220349.275379] ------------[ cut here ]------------
[220349.275418] phy0-ap0.sta2: Failed check-sdata-in-driver check, flags: 0x1
[220349.279118] WARNING: CPU: 3 PID: 1882 at __ieee80211_flush_queues+0x174/0x180 [mac80211]
[220349.285846] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet ath11k_ahb ath11k wireguard pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211 libchacha20poly1305 iptable_mangle iptable_filter ipt_REJECT ipt_ECN ip_tables chacha_neon cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY x_tables slhc sch_cake qrtr_smd qrtr qmi_helpers poly1305_neon nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic libcrc32c libchacha crc_ccitt compat sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw
[220349.286098]  cls_flow cls_basic act_skbedit act_mirred act_gact ifb ip6_udp_tunnel udp_tunnel seqiv jitterentropy_rng drbg michael_mic kpp hmac cmac leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom qca_nss_dp qca_ssdk gpio_button_hotplug aquantia hwmon crc32c_generic
[220349.383223] CPU: 3 PID: 1882 Comm: hostapd Not tainted 5.15.108 #0
[220349.405459] Hardware name: Dynalink DL-WRX36 (DT)
[220349.411447] pstate: 80400005 (Nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[220349.416311] pc : __ieee80211_flush_queues+0x174/0x180 [mac80211]
[220349.423514] lr : __ieee80211_flush_queues+0x174/0x180 [mac80211]
[220349.429504] sp : ffffffc00d0c37f0
[220349.435574] x29: ffffffc00d0c37f0 x28: ffffff8003f6e9c0 x27: ffffffc00d0c3dd0
[220349.438881] x26: ffffff80020ae880 x25: ffffffc008c260c0 x24: ffffffc000c05000
[220349.446086] x23: 0000000000000000 x22: 0000000000000007 x21: ffffff8012e50e20
[220349.453290] x20: ffffff8006350880 x19: 0000000000000007 x18: ffffffc008b27bc8
[220349.460495] x17: 0000000000000000 x16: 0000000000000002 x15: 0000000000001722
[220349.467699] x14: 00000000000007b6 x13: ffffffc00d0c3518 x12: ffffffc008b7fbc8
[220349.474905] x11: 0000000000000004 x10: ffffffc008b7fbc8 x9 : 0000000000000000
[220349.482109] x8 : ffffffc008b27b78 x7 : ffffffc008b27bc8 x6 : 0000000000000001
[220349.489314] x5 : ffffff803fddd6e0 x4 : 0000000000000000 x3 : 0000000000000027
[220349.496520] x2 : 0000000000000027 x1 : 0000000000000023 x0 : 000000000000003d
[220349.503726] Call trace:
[220349.510922]  __ieee80211_flush_queues+0x174/0x180 [mac80211]
[220349.513617]  ieee80211_flush_queues+0x18/0x24 [mac80211]
[220349.519259]  sta_set_sinfo+0xbc4/0xc50 [mac80211]
[220349.524640]  sta_info_destroy_addr_bss+0x50/0x7c [mac80211]
[220349.529329]  ieee80211_color_change_finish+0x1dbc/0x2064 [mac80211]
[220349.535146]  cfg80211_check_station_change+0x1408/0x4950 [cfg80211]
[220349.541482]  genl_family_rcv_msg_doit+0xb8/0x11c
[220349.547816]  genl_rcv_msg+0xd4/0x1cc
[220349.552415]  netlink_rcv_skb+0x5c/0x130
[220349.556061]  genl_rcv+0x38/0x50
[220349.560054]  netlink_unicast+0x1f8/0x2f4
[220349.563267]  netlink_sendmsg+0x1a0/0x3dc
[220349.567173]  ____sys_sendmsg+0x288/0x2c0
[220349.571166]  ___sys_sendmsg+0x84/0xf0
[220349.575158]  __sys_sendmsg+0x48/0xb0
[220349.578804]  __arm64_sys_sendmsg+0x24/0x30
[220349.582538]  invoke_syscall.constprop.0+0x5c/0x104
[220349.586532]  do_el0_svc+0x6c/0x15c
[220349.591389]  el0_svc+0x18/0x54
[220349.594862]  el0t_64_sync_handler+0xe8/0x114
[220349.597990]  el0t_64_sync+0x184/0x188
[220349.602503] ---[ end trace a20829cdd69f73b6 ]---
`

It appears the router looses signal to the closest RT3200, about 5ft away, but separated by a wall.  I get this on the RT3200:
`Sat Apr 29 19:25:03 2023 daemon.notice wpa_supplicant[1441]: wl1-sta0: CTRL-EVENT-BEACON-LOSS
`

### OpenWrt version

r22652-9a26669510

### OpenWrt target/subtarget

ipq807x/generic

### Device

Dynalink DL-WRX36

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-05-01:
You likely need newer snapshot , dmesg looks like [CVE-2022-47522] fix without driver support

smileys29, 2023-05-01:
@brada4 Thanks for the reply on this.  snapshots on both the RT3200 and the WRX36 are a couple days old at this point.
 
 I saw @nbd168 released a patch for this CVE on March 30 for the MT76.  Is this something that needs to be addressed in the ath11k driver @robimarko?

robimarko, 2023-05-03:
There shouldn't be, but TX flush is kind of broken in ath11k currently

gtxaspec, 2023-05-21:
similar issue on ath10k using the ath10k driver... #12420 , running latest master same.

oldium, 2023-11-22:
This looks related to #12420 with fix in pull request #13620.