openwrt/openwrt #12509: nftables Error: Could not process rule: Resource busy

jsalatiel, 2023-05-01:
### Describe the bug

I have created a file /usr/share/nftables.d/ruleset-post/log.nft containing the following:
```
insert rule inet fw4 reject_to_wan oifname { "pppoe-wan1" } ct state new meter blkdrp { tcp dport . ip saddr timeout 3600s limit rate 1/hour } counter log prefix "reject2wan"                        
```

Everything worked just fine after the first /etc/init.d/firewall restart.

The problem is that no restart of firewall will work anymore. I will always get:
```
In file included from /dev/stdin:349:1-53:
/usr/share/nftables.d/ruleset-post/log.nft:2:82-150: Error: Could not process rule: Resource busy
insert rule inet fw4 reject_to_wan oifname { "pppoe-wan1" } ct state new meter blkdrp { tcp dport . ip saddr timeout 3600s limit rate 1/hour } counter log prefix "reject2wan"
                                                                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```

Any ideas why ?

### OpenWrt version

r20028-43d71ad93e

### OpenWrt target/subtarget

ramips/mt7621

### Device

MikroTik RouterBOARD 750Gr3

### Image kind

Official downloaded image

### Steps to reproduce

 In the description

### Actual behaviour

Firewall Fails to restart

### Expected behaviour

Firewall would restart just fine

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-05-01:
It is chain-post rule i.e non-default chain shall be made beforehand. See readme in that directory.

jsalatiel, 2023-05-01:
Hi @brada4 , i read the readme previously and I am gonna break this in two question then:
1) Why does it work on the first firewall restart. Technically  the chain is already there. reject_to_wan is a "default" chain, at least I am not creating that with custom rules, it is fw4 default rules creating it.
2) I had tried to  adding that log.nft into /usr/share/nftables.d/chain-pre/reject_to_wan , but apparently even if that file is loaded the rules are not inserted. I am  experienced with iptables, but starting now with nftables, so maybe I am doing something wrong with the nft rule.
When i try to add the the rule by /usr/share/nftables.d/**ruleset-post**/log.nft, I basically have inside that file the whole nft command:

```
# cat /usr/share/nftables.d/ruleset-post/log.nft
insert rule inet fw4 reject_to_wan oifname { "pppoe-wan1"} ct state new meter blkdrp { tcp dport . ip saddr timeout 3600s limit rate 1/hour } counter log prefix "reject2wan"
```
but i suppose if i try to add in /usr/share/nftables.d/**chain-pre**/**reject_to_wan**/log.nft I should skip the `insert rule inet fw4 reject_to_wan` because I understand that's already taken by the path name, or not? Either way, it does not work, so I may be missing something, since I get no errors, it just does not add the rule to the chain.

```
# cat /usr/share/nftables.d/chain-pre/reject_to_wan/log.nft
oifname { "pppoe-wan1"  } ct state new meter blkdrp { tcp dport . ip saddr timeout 3600s limit rate 1/hour } counter log prefix "reject2wan"
```

jsalatiel, 2023-05-01:
Thanks, i have added the uci include of type script. And it worked.
Good to know that there is no include support within reject_to_wan. So i suppose that only input, forward and output are supported.

Still can not figure it out what causes "Could not process rule: Resource busy" after the first run.

brada4, 2023-05-01:
nft sometimes mysteriously denies modifying existing ruleset
try `service firewall stop ; service firewall start` to flush rules and make itlike first time.

jsalatiel, 2023-05-01:
Stop and start works.
Is that what happens (stop/start not restart) when the firewall reload is triggered by some interface up and down?

brada4, 2023-05-02:
It does full restart only with `restart`
The `reload` is only additive `start` 
Note that states are not flushed in either case except for masquerade automatic flush
Interface trigger does limited `restart` on rules applying to particular interface.

I think error handling is tough in a simple nft script but check the discussion:
 https://bugzilla.redhat.com/show_bug.cgi?id=2013173#c3
Seems though changing priority to same value is rejected in your case (?)