openwrt/openwrt #12512: Acer RT-AC51U traffic rules does not always work on 22.03.4

rikke-rp, 2023-05-01:
### Describe the bug

When enabling simple traffic rules to block all internet access for a device by MAC address, the rules does not always apply. This is encountered on version 22.03.4 for the Acer RT-AC51U device.

The rule is as follows:
```
firewall.@rule[1]=rule
firewall.@rule[1].name='block internet access'
firewall.@rule[1].src='lan'
firewall.@rule[1].dest='wan'
firewall.@rule[1].target='REJECT'
firewall.@rule[1].src_mac='94:3C:C6:xx:xx:xx'
firewall.@rule[1].enabled='1'
```
I've censored the last 6 digits of the mac address on purpose.

After enabling the rule, the device is still online. Another rule to block specific port has same problem.

After downgrading to old version 21.02.3 everything works again. I'm just gonna report this bug because it is open source and I like the work you are doing. My problem is already solved by downgrading, because I already have a 2nd Acer RT-AC51U on version 21.02.3 with no problems at all.

### OpenWrt version

22.03.4

### OpenWrt target/subtarget

i already downgraded

### Device

Acer RT-AC51U

### Image kind

Official downloaded image

### Steps to reproduce

1. Flash Acer RT-AC51U with firmware version 22.03.4
2. Set rule: (replace mac address with your own device's mac address)
```
firewall.@rule[1]=rule
firewall.@rule[1].name='block internet access'
firewall.@rule[1].src='lan'
firewall.@rule[1].dest='wan'
firewall.@rule[1].target='REJECT'
firewall.@rule[1].src_mac='94:3C:C6:xx:xx:xx'
firewall.@rule[1].enabled='1'
```
3. $ uci commit firewall
4. $ /etc/init.d/firewall restart

### Actual behaviour

The internet access is not blocked on device with the specified mac address connected to the OpenWrt router

### Expected behaviour

The internet access should be blocked on device with the specified mac address connected to the OpenWrt router

### Additional info

I'm using a setup of 2 routers now, and I discovered this issue when the newest router got flashed with latest and greatest stable 30th of April 2023.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-05-01:
Can you post resulting rendering in `nft list ruleset`

rikke-rp, 2023-05-02:
I don't have "nft" installed `-ash: nft: not found`, and I only have the working version installed now. Unless you want to see the ruleset in the working version 21.02.3

francisuk1989, 2023-05-03:
> Unless you want to see the ruleset in the working version 21.02.3

Keep in mind that, **21.02.6** is based on iptables/FW3, 22.03.4/SNAPSHOTS will be nftables based.

If you fine with SSH, Then you can add this simple rule into /etc/config/firewall

SSH into 192.168.1.1
opkg update
opkg install nano
nano /etc/config/firewall
Press PageDown till the end
Copy/Paste the following, Replace 94:3C:C6:xx:xx:xx with a real mac address.

```
config rule
	option name 'Block Internet Access'
	list src_mac '94:3C:C6:xx:xx:xx'
	list proto 'all'
	option src 'lan'
	option dest 'wan'
	option target 'DROP'
```
Once pasted, Ctrl+X and then Y and then enter (to save the file)

Last enter this command to apply the firewall rules.

`/etc/init.d/firewall restart`

I can confirm this works on the latest snapshots so 22.03.4 should work fine.

rikke-rp, 2023-05-25:
I now have a new router (ASUS RT-AX53U) with version 22.03.3 and I encouter same problem as with ASUS RT-AC51U and version 22.03.4. I guess the nftables is to blame. 
I changed the rule to what [francisuk1989](https://github.com/francisuk1989) suggested but it did not work. enabling and disabling the rule only works 10% of the time or so, and when I enable the rule, the "list src_mac" changes back to "option src_mac". Changing the REJECT to DROP did not work either.  I will see if I can find a way to fix my problem so it works like the old iptables.

brada4, 2023-05-26:
> Can you post resulting rendering in `nft list ruleset`

rikke-rp, 2023-05-30:
> 
This newer router has it installed by default
```
$ nft list ruleset
table inet fw4 {
	chain input {
		type filter hook input priority filter; policy accept;
		iifname "lo" accept comment "!fw4: Accept traffic from loopback"
		ct state established,related accept comment "!fw4: Allow inbound established and related flows"
		tcp flags syn / fin,syn,rst,ack jump syn_flood comment "!fw4: Rate limit TCP syn packets"
		iifname "br-lan" jump input_lan comment "!fw4: Handle lan IPv4/IPv6 input traffic"
		iifname "wan" jump input_wan comment "!fw4: Handle wan IPv4/IPv6 input traffic"
	}

	chain forward {
		type filter hook forward priority filter; policy drop;
		ct state established,related accept comment "!fw4: Allow forwarded established and related flows"
		iifname "br-lan" jump forward_lan comment "!fw4: Handle lan IPv4/IPv6 forward traffic"
		iifname "wan" jump forward_wan comment "!fw4: Handle wan IPv4/IPv6 forward traffic"
		jump handle_reject
	}

	chain output {
		type filter hook output priority filter; policy accept;
		oifname "lo" accept comment "!fw4: Accept traffic towards loopback"
		ct state established,related accept comment "!fw4: Allow outbound established and related flows"
		oifname "br-lan" jump output_lan comment "!fw4: Handle lan IPv4/IPv6 output traffic"
		oifname "wan" jump output_wan comment "!fw4: Handle wan IPv4/IPv6 output traffic"
	}

	chain prerouting {
		type filter hook prerouting priority filter; policy accept;
		iifname "br-lan" jump helper_lan comment "!fw4: Handle lan IPv4/IPv6 helper assignment"
	}

	chain handle_reject {
		meta l4proto tcp reject with tcp reset comment "!fw4: Reject TCP traffic"
		reject comment "!fw4: Reject any other traffic"
	}

	chain syn_flood {
		limit rate 25/second burst 50 packets return comment "!fw4: Accept SYN packets below rate-limit"
		drop comment "!fw4: Drop excess packets"
	}

	chain input_lan {
		jump accept_from_lan
	}

	chain output_lan {
		jump accept_to_lan
	}

	chain forward_lan {
		jump accept_to_wan comment "!fw4: Accept lan to wan forwarding"
		jump accept_to_lan
	}

	chain helper_lan {
	}

	chain accept_from_lan {
		iifname "br-lan" counter packets 67 bytes 7404 accept comment "!fw4: accept lan IPv4/IPv6 traffic"
	}

	chain accept_to_lan {
		oifname "br-lan" counter packets 20 bytes 3624 accept comment "!fw4: accept lan IPv4/IPv6 traffic"
	}

	chain input_wan {
		meta nfproto ipv4 udp dport 68 counter packets 0 bytes 0 accept comment "!fw4: Allow-DHCP-Renew"
		icmp type echo-request counter packets 0 bytes 0 accept comment "!fw4: Allow-Ping"
		meta nfproto ipv4 meta l4proto igmp counter packets 0 bytes 0 accept comment "!fw4: Allow-IGMP"
		meta nfproto ipv6 udp dport 546 counter packets 0 bytes 0 accept comment "!fw4: Allow-DHCPv6"
		ip6 saddr fe80::/10 icmpv6 type . icmpv6 code { mld-listener-query . no-route, mld-listener-report . no-route, mld-listener-done . no-route, mld2-listener-report . no-route } counter packets 0 bytes 0 accept comment "!fw4: Allow-MLD"
		icmpv6 type { destination-unreachable, time-exceeded, echo-request, echo-reply, nd-router-solicit, nd-router-advert } limit rate 1000/second counter packets 13 bytes 704 accept comment "!fw4: Allow-ICMPv6-Input"
		icmpv6 type . icmpv6 code { packet-too-big . no-route, parameter-problem . no-route, nd-neighbor-solicit . no-route, nd-neighbor-advert . no-route, parameter-problem . admin-prohibited } limit rate 1000/second counter packets 1 bytes 72 accept comment "!fw4: Allow-ICMPv6-Input"
		jump reject_from_wan
	}

	chain output_wan {
		jump accept_to_wan
	}

	chain forward_wan {
		icmpv6 type { destination-unreachable, time-exceeded, echo-request, echo-reply } limit rate 1000/second counter packets 0 bytes 0 accept comment "!fw4: Allow-ICMPv6-Forward"
		icmpv6 type . icmpv6 code { packet-too-big . no-route, parameter-problem . no-route, parameter-problem . admin-prohibited } limit rate 1000/second counter packets 0 bytes 0 accept comment "!fw4: Allow-ICMPv6-Forward"
		meta l4proto esp counter packets 0 bytes 0 jump accept_to_lan comment "!fw4: Allow-IPSec-ESP"
		udp dport 500 counter packets 0 bytes 0 jump accept_to_lan comment "!fw4: Allow-ISAKMP"
		jump reject_to_wan
	}

	chain accept_to_wan {
		oifname "wan" counter packets 110 bytes 9908 accept comment "!fw4: accept wan IPv4/IPv6 traffic"
	}

	chain reject_from_wan {
		iifname "wan" counter packets 331 bytes 43190 jump handle_reject comment "!fw4: reject wan IPv4/IPv6 traffic"
	}

	chain reject_to_wan {
		oifname "wan" counter packets 0 bytes 0 jump handle_reject comment "!fw4: reject wan IPv4/IPv6 traffic"
	}

	chain dstnat {
		type nat hook prerouting priority dstnat; policy accept;
	}

	chain srcnat {
		type nat hook postrouting priority srcnat; policy accept;
		oifname "wan" jump srcnat_wan comment "!fw4: Handle wan IPv4/IPv6 srcnat traffic"
	}

	chain srcnat_wan {
		meta nfproto ipv4 masquerade comment "!fw4: Masquerade IPv4 wan traffic"
	}

	chain raw_prerouting {
		type filter hook prerouting priority raw; policy accept;
	}

	chain raw_output {
		type filter hook output priority raw; policy accept;
	}

	chain mangle_prerouting {
		type filter hook prerouting priority mangle; policy accept;
	}

	chain mangle_postrouting {
		type filter hook postrouting priority mangle; policy accept;
	}

	chain mangle_input {
		type filter hook input priority mangle; policy accept;
	}

	chain mangle_output {
		type route hook output priority mangle; policy accept;
	}

	chain mangle_forward {
		type filter hook forward priority mangle; policy accept;
		iifname "wan" tcp flags syn tcp option maxseg size set rt mtu comment "!fw4: Zone wan IPv4/IPv6 ingress MTU fixing"
		oifname "wan" tcp flags syn tcp option maxseg size set rt mtu comment "!fw4: Zone wan IPv4/IPv6 egress MTU fixing"
	}
}
```

rikke-rp, 2023-05-31:
Everything works as expected on ASUS RT-AC51U with version 21.02.3 (based on iptables). I always make the rule from the luci gui, and change the mac address and enable/disable the rule from ssh. I can see my custom rules with `$ cat etc/config/firewall`

```
config rule
	option name 'block mqtt port 1883'
	option src 'lan'
	option dest 'wan'
	option dest_port '1883'
	option target 'DROP'
	option src_mac 'EC:62:60:4D:34:7C'
	option enabled '0'

config rule
	option name 'block internet access'
	option src 'lan'
	option dest 'wan'
	option target 'DROP'
	list proto 'all'
	option src_mac 'EC:62:60:4D:34:7C'
	option enabled '0'
```
And with `$ uci show firewall`
```
firewall.@rule[0]=rule
firewall.@rule[0].name='block mqtt port 1883'
firewall.@rule[0].src='lan'
firewall.@rule[0].dest='wan'
firewall.@rule[0].dest_port='1883'
firewall.@rule[0].target='DROP'
firewall.@rule[0].src_mac='EC:62:60:4D:34:7C'
firewall.@rule[0].enabled='0'
firewall.@rule[1]=rule
firewall.@rule[1].name='block internet access'
firewall.@rule[1].src='lan'
firewall.@rule[1].dest='wan'
firewall.@rule[1].target='DROP'
firewall.@rule[1].proto='all'
firewall.@rule[1].src_mac='EC:62:60:4D:34:7C'
firewall.@rule[1].enabled='0'

```
And I enable rules like this:
```
$ uci set firewall.@rule[0].enabled='1'
$ uci commit firewall
$ service firewall restart
```

The rules only work maybe 3 out of 10 times with nftables. Maybe the openwrt luci gui does not create the rules correctly? nft wasn't even installed on ASUS RT-AC51U with version 22.03.4.

I'm really not on expert on these nftables and iptables, but I'm trying to make it work on both :)

brada4, 2023-05-31:
> rules does not always apply

Now they never applied?

jow-, 2023-05-31:
> The rules only work maybe 3 out of 10 times with nftables

Sounds as if there's still established connections. Do they start to work after a `echo f > /proc/net/nf_conntrack` ?

rikke-rp, 2023-05-31:
> Sounds as if there's still established connections. Do they start to work after a `echo f > /proc/net/nf_conntrack` ?

That worked! After restarting the firewall the device was still online, but after clearing the nf_conntrack it got disconnected. Disabling the rule again worked as expected :D