openwrt/openwrt #12516: NWA50AX

FirmWareVT, 2023-05-01:
### Describe the bug

OpenWrt 22.03.5 r20134-5f15225c1e 

5Ghz channel width selection show up 160Mhz which is not allowed, if you select 160Mhz 5Ghz stop working. Only 80Mhz allowed. Version OpenWrt 22.03.4 correctly limited selection width to 80Mhz. Version OpenWrt 22.03.5 r20134-5f15225c1e restored 160Mhz width which lock up 5Ghz radio

### OpenWrt version

OpenWrt 22.03.5 r20134-5f15225c1e 

### OpenWrt target/subtarget

OpenWrt 22.03.5 r20134-5f15225c1e 

### Device

NWA50AX

### Image kind

Official downloaded image

### Steps to reproduce

Select 160Mhz, with lock up your 5Ghz radio

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-05-01:
Try to look at `logread` for DFS/ACS records. In most of the world half of 160MHz bands is with radar detection and may not work.

Djfe, 2023-05-07:
that's not the issue. the chip doesn't support 160mhz but the latest openwrt release has a regression that allows selecting it anyways.
atleast that's the report. I haven't confirmed this

FirmWareVT, 2023-05-08:
Exactly correct & my report.

Neustradamus, 2024-11-04:
@FirmWareVT: What is the status of this ticket with master code?