openwrt/openwrt #12522: Please add nftables raw table support

brada4, 2023-05-03:
### Describe the bug

nftables raw table support is not present, while legacy iptables support for raw tables is visible in form of kmod. 
Please bring NFT functionality in line with historical iptables

### OpenWrt version

r20134-5f15225c1e

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer C7 v5

### Image kind

Official downloaded image

### Steps to reproduce

```
iptables-nft -t raw -A PREROUTING -i lo -j CT --notrack
nft 'add rule ip raw PREROUTING iifname "lo" counter notrack'
```

### Actual behaviour

```
root@OpenWrt:~# iptables-nft -t raw -A PREROUTING -i lo -j CT --notrack
iptables v1.8.7 (nf_tables): unknown option "--notrack"
Try `iptables -h' or 'iptables --help' for more information.
root@OpenWrt:~# nft 'add rule ip raw PREROUTING iifname "lo" counter notrack'
Error: Could not process rule: No such file or directory
add rule ip raw PREROUTING iifname "lo" counter notrack
            ^^^
```

### Expected behaviour

Rule is added and loopback connections are no longer tracked.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dave14305, 2023-05-03:
```
nft add rule inet fw4 raw_prerouting iifname "lo" counter notrack
```
Does this work for you?

brada4, 2023-05-04:
Thanks a million.
Functional side is fully covered, to update title to reflect merely cosmetic problem.

Permanent integration as follows.

`/etc/nftables.d/20-notrack.nft`
```
chain user_pre_raw_prerouting {
         type filter hook prerouting priority raw - 1; policy accept;
         iifname "lo" counter notrack;
}
```

brada4, 2023-05-04:
Rules are added to nicely named tables, but conntrack state is created ignoring notrack statement.

swestlake, 2023-05-27:
i noticed today when try to add **prerouting** in /etc/config/firewall(an include section), it doesn't get honored. however the include can work with the option chain 'dstnat' which is another priority of prerouting..

added section in /etc/config/firewall:
config include
  option type 'nftables'
  option path '/root/fw4-includes/chainpre_dstnat_1.nft'
\#  option chain 'prerouting' -- is an issue
  option position 'chain-pre' 
  option chain 'dstnat'

(OpenWrt 22.03.5, r20134-5f15225c1e)

brada4, 2023-05-27:
No effect.

```
        chain dstnat {
                type nat hook prerouting priority dstnat; policy accept;
                iifname "lo" counter packets 0 bytes 0 notrack
        }
```
`nc 127.1 12345`
```
    [NEW] tcp      6 120 SYN_SENT src=127.0.0.1 dst=127.0.0.1 sport=39516 dport=12345 [UNREPLIED] src=127.0.0.1 dst=127.0.0.1 sport=12345 dport=39516
[DESTROY] tcp      6 src=127.0.0.1 dst=127.0.0.1 sport=39516 dport=12345 packets=1 bytes=60 [UNREPLIED] src=127.0.0.1 dst=127.0.0.1 sport=12345 dport=39516 packets=1 bytes=40
```

brada4, 2023-11-03:
The support is there, just one has to install and load respextive xt_ kmod alongside nftables.

closed 2023-11-03 by brada4

TrayHard, 2023-11-21:
>          type filter hook prerouting priority raw - 1; policy accept;

How you made that work? It gets me an error when I try to do this:

[image]

brada4, 2023-11-21:
```
chain notrackI {
        type filter hook prerouting priority raw; policy accept;
        iif "lo" notrack
}

chain notrackO {
        type filter hook output priority raw; policy accept;
        oif "lo" notrack
}
```

brada4, 2023-11-21:
save in /etc/nftables.d/99-local.nft