openwrt/openwrt #12524: ksmbd: "force create/directory mode" does not work

jacgl, 2023-05-03:
### Describe the bug

force create (directory) mode should work as bitwise ORed after the bitmask specified with  create  mask (directory mask) is applied; unfortunately it is always affected by default umask 0022, even for root user forced
checked on 22.03 (r22090), and mainline

### OpenWrt version

r22702-cf8d861978

### OpenWrt target/subtarget

kirkwood/generic

### Device

ZyXEL NSA310S

### Image kind

Official downloaded image

### Steps to reproduce

in ksmbd.conf
        create mask = 0666
        directory mask = 0777
        force create mode = 0660            or 0666
        force directory mode = 0770        or 0777
optionally  option force_root '1' 

observed for both ext4 and ntfs3

### Actual behaviour

group/all never gets write permission, w is always cleared for group and all
i.e.  rw- r-- --- /rw- r-- r-- for files, rwx r-x --- /rwx r-x r-x for dirs

### Expected behaviour

the mode shoud be forced by the setting as follows
rw- rw- --- /rw- rw- rw- for files
rwx rwx ---/rwx rwx rwx for dirs 

### Additional info

similar issue described for Alpine
https://github.com/cifsd-team/ksmbd/issues/586

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Neustradamus, 2023-05-18:
@neheb: Have you seen?

neheb, 2023-05-18:
Upstream issue.

Neustradamus, 2024-01-20:
@jacgl, @neheb: The problem is always here with the current version?

If yes, it is possible to create a ticket upstream?
- https://github.com/cifsd-team/ksmbd + - https://github.com/cifsd-team/ksmbd-tools
- https://github.com/namjaejeon/ksmbd + https://github.com/namjaejeon/ksmbd-tools

Neustradamus, 2024-02-09:
@neheb: For information, have you seen the new version?
- https://github.com/cifsd-team/ksmbd/releases